Koenkk/zigbee2mqtt #3807: zigbee2mqtt sets wrong color after update

crackytsi, 2020-06-25:
# Bug Report
## What happened
Upgrade to latest zigbee2mqtt version from 1.11.0 to 1.14.0 breaks color setting for Müller-Licht (e.g. LED-Stripe)

## What did you expect to happen
You set color to #030100 and you expect a dark red color, but it is getting a light, shiny blue color
## How to reproduce it (minimal and precise)
set color using command:
`set ledstripe color 030100`

## Debug Info
Zigbee2mqtt version: 1.14.0
MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.0","commit":"9009de2","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}},"log_level":"debug","permit_join":true}'

Adapter hardware: CC2531
Adapter firmware version:  ZStack30x

Koenkk, 2020-06-27:
Can you provide the herdsman debug logging when setting the color on both 1.11.0 and 1.14.0?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

crackytsi, 2020-06-29:
Here the Start-Log from 1.14.0
```

zigbee2mqtt:info  2020-06-29 09:30:42: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-06-29.09-30-42' filename: log.txt
zigbee2mqtt:debug 2020-06-29 09:30:42: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-06-26.16-11-32'
zigbee2mqtt:debug 2020-06-29 09:30:43: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-06-29 09:30:43: Starting zigbee2mqtt version 1.14.0 (commit #9009de2)
zigbee2mqtt:info  2020-06-29 09:30:43: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-06-29 09:30:43: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE87D4D-if00"},"adapter":{"concurrent":null}}'
  zigbee-herdsman:adapter Path '/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE87D4D-if00' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE87D4D-if00"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE87D4D-if00 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +10ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +1ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,7,28] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,7,28] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,7] - 28 +2ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1913} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +29ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +24ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,2,2,7,2,217,20,52,1,2,0,0,0,0,144] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,2,2,7,2,217,20,52,1,2,0,0,0,0,144] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,2,2,7,2,217,20,52,1,2,0,0,0,0] - 144 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425} +17ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack30x' ({"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +18ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +34ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +38ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +34ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +31ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +27ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +30ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +29ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,8,0,0]}} +29ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,132,0,0,174] +29ms
[... cut ...] mp":424134,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,19,10,5,5,33,237,0]}} +21s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":19,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1285,"dataType":33,"attrData":237}]},"address":33448,"endpoint":1,"linkquality":60,"groupID":0}' +21s
```

crackytsi, 2020-06-29:
Here the Start-Log from 1.11.0
```
zigbee2mqtt:info  2020-06-29 09:43:41: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-06-29.09-43-41' filename: log.txt
zigbee2mqtt:debug 2020-06-29 09:43:41: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-06-17.07-30-29'
zigbee2mqtt:debug 2020-06-29 09:43:42: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-06-29 09:43:42: Starting zigbee2mqtt version 1.11.0 (commit #e140cf5)
zigbee2mqtt:info  2020-06-29 09:43:42: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-06-29 09:43:42: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE87D4D-if00"}}'
  zigbee-herdsman:adapter Path '/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE87D4D-if00' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE87D4D-if00"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE87D4D-if00 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +10ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +1ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [145,114,28,57,254,28,68,129,0,0,4,11,168,130,1,1,0,60,0,3,50,47,0,0,8,24,222,10,5,5,33,239,0,145,114,28,43,254,3,69,196,239,158,0,243,254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,155,205,47,0,0,8,24,209,10,8,5,33,33,0,239,158,29,116,254,3,69,196,168,130,0,168,254,28,68,129,0,0,4,11,168,130,1,1,0,21,0,254,220,47,0,0,8,24,59,10,5,5,33,237,0,168,130,29,62,254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,157,10,48,0,0,8,24,239,10,8,5,33,30,0,239,158,29,171] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [145,114,28,57,254,28,68,129,0,0,4,11,168,130,1,1,0,60,0,3,50,47,0,0,8,24,222,10,5,5,33,239,0,145,114,28,43,254,3,69,196,239,158,0,243,254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,155,205,47,0,0,8,24,209,10,8,5,33,33,0,239,158,29,116,254,3,69,196,168,130,0,168,254,28,68,129,0,0,4,11,168,130,1,1,0,21,0,254,220,47,0,0,8,24,59,10,5,5,33,237,0,168,130,29,62,254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,157,10,48,0,0,8,24,239,10,8,5,33,30,0,239,158,29,171] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,168,130,1,1,0,60,0,3,50,47,0,0,8,24,222,10,5,5,33,239,0,145,114,28] - 43 +2ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":33448,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":60,"securityuse":0,"timestamp":3092995,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,222,10,5,5,33,239,0]}} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,239,158,0,243,254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,155,205,47,0,0,8,24,209,10,8,5,33,33,0,239,158,29,116,254,3,69,196,168,130,0,168,254,28,68,129,0,0,4,11,168,130,1,1,0,21,0,254,220,47,0,0,8,24,59,10,5,5,33,237,0,168,130,29,62,254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,157,10,48,0,0,8,24,239,10,8,5,33,30,0,239,158,29,171] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [239,158,0] - 243 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":40687,"relaycount":0,"relaylist":[]} +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,155,205,47,0,0,8,24,209,10,8,5,33,33,0,239,158,29,116,254,3,69,196,168,130,0,168,254,28,68,129,0,0,4,11,168,130,1,1,0,21,0,254,220,47,0,0,8,24,59,10,5,5,33,237,0,168,130,29,62,254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,157,10,48,0,0,8,24,239,10,8,5,33,30,0,239,158,29,171] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,239,158,1,1,0,0,0,155,205,47,0,0,8,24,209,10,8,5,33,33,0,239,158,29] - 116 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":40687,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3132827,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,209,10,8,5,33,33,0]}} +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,168,130,0,168,254,28,68,129,0,0,4,11,168,130,1,1,0,21,0,254,220,47,0,0,8,24,59,10,5,5,33,237,0,168,130,29,62,254,28,68,129,0,0,4,11,239,158,1,1,0,0,0,157,10,48,0,0,8,24,239,10,8,5,33,30,0,239,158,29,171] +4ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
zigbee2mqtt:info  2020-06-29 09:45:26: MQTT publish: topic 'zigbee2mqtt/ledstripe', payload '{"state":"ON","brightness":255,"color":{},"color_temp":null}'

```

I don't understand it, but it seems still to be wrong.
What I did was:
restoring 1.11 backup of /opt/zigbee2mqtt back and testing it. But also with the backup the color is now light blue (not even a little red)
Was my restore way wrong? Does this help to identify the issue?

Koenkk, 2020-06-29:
The payload is invalid (`{"color":"030100"}`) should be `{"color":"#030100"}` (with **#**). Try with `set ledstripe color #030100`

crackytsi, 2020-06-29:
Thanks :)
Unforunately this also does not work. Strange is, that it was working fine for months until I updated zigbee2mqtt.

Koenkk, 2020-06-29:
I'm not sure what changed, but for sure the send payload is invalid. To make sure this is not a zigbee2mqtt issue use e.g. MQTT.fx and send to the topic `zigbee2mqtt/ledstripe/set`  the following payload `{"color":"#030100"}`

crackytsi, 2020-06-29:
Hmm strange.
This also sets the color to blue.
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#030100"}'
it becomes a red e.g. setting only the red-factor:
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#ff0000"}'

If you compare the color, it is absollutly not blue, and it wasn't in the past...
https://www.hexcolortool.com

I had a wakeup-light using these colors, but they are now starting with light-blue instead of dark red:
`'030100','0B0300','190601','230902','2F0D03','370F03','401204','481405','521806','591007','622108','6E2508','772809','7F2A09','882D0A','90300A','96320A','9D330A','A6360A','B0390A','B0450A','B9480B','C1580B','CB610B','D3650C','D6740C','DC770B','DC810B','E5860B','ED8E12','F2A113','F2B013','F5B51A','F9BA22','FBBD29','FEC232','FFC742','FECC55','FED166','FDD473','FDD880','FEDD8C','FDDF97','FDE4A7','FDEABC','FDEEC8','FDF0D1','FDF3DA','FCF4E2'`

crackytsi, 2020-06-29:
As the color seems to be set for Müller-Licht Tint using brightness, color-x, color-y and color_temp:
Was there a change in the calculation from RGB to color_x/color_y?

crackytsi, 2020-06-29:
Thats really strange.... 
Here some examples:
```
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#040000"}' ==> blue
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#050000"}' ==> blue
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#060000"}' ==> blue
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#070000"}' ==> red
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#080000"}' ==> red
...
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#ff0000"}' ==> red
```

Koenkk, 2020-06-29:
Can you provide the herdsman debug logging of those commands?

crackytsi, 2020-06-29:
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#040000"}'  ==> blue
```
zigbee2mqtt:debug 2020-06-29 20:06:58: Received MQTT message on 'zigbee2mqtt/ledstripe/set' with data '{"color":"#040000"}'
zigbee2mqtt:debug 2020-06-29 20:06:58: Publishing 'set' 'color' to 'ledstripe'
  zigbee-herdsman:controller:endpoint Command 0xccccccfffea2754c/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":0,"colory":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +25s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":29329,"destendpoint":1,"srcendpoint":1,"clusterid":768,"transid":4,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[1,5,7,0,0,0,0,0,0]}} +25s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,145,114,1,1,0,3,4,0,30,9,1,5,7,0,0,0,0,0,0,198] +25s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +25s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,4,194] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,4,194] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,4] - 194 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":4} +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,145,114,0,97] +28ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,145,114,0,97] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [145,114,0] - 97 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":29329,"relaycount":0,"relaylist":[]} +29ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,0,3,145,114,1,1,0,60,0,225,82,46,0,0,5,8,5,11,7,0,145,114,29,103] +64ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,0,3,145,114,1,1,0,60,0,225,82,46,0,0,5,8,5,11,7,0,145,114,29,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,0,3,145,114,1,1,0,60,0,225,82,46,0,0,5,8,5,11,7,0,145,114,29] - 103 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":29329,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":60,"securityuse":0,"timestamp":3035873,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,5,11,7,0]}} +65ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":7,"statusCode":0}},"address":29329,"endpoint":1,"linkquality":60,"groupID":0}' +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
zigbee2mqtt:info  2020-06-29 20:06:58: MQTT publish: topic 'zigbee2mqtt/ledstripe', payload '{"state":"ON","brightness":0,"color":{"x":0,"y":0},"color_temp":14085}'

```

mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/ledstripe/set -m '{"color":"#050000"}' ==> blue
```
zigbee2mqtt:debug 2020-06-29 20:07:30: Received MQTT message on 'zigbee2mqtt/ledstripe/set' with data '{"color":"#050000"}'
zigbee2mqtt:debug 2020-06-29 20:07:30: Publishing 'set' 'color' to 'ledstripe'
  zigbee-herdsman:controller:endpoint Command 0xccccccfffea2754c/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":0,"colory":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +32s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":29329,"destendpoint":1,"srcendpoint":1,"clusterid":768,"transid":5,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[1,6,7,0,0,0,0,0,0]}} +32s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,145,114,1,1,0,3,5,0,30,9,1,6,7,0,0,0,0,0,0,196] +32s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +17s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +32s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,5,195] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,5,195] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,5] - 195 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":5} +17s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,145,114,0,97] +29ms
[... cut ...]
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":10,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":7,"statusCode":0}},"address":29329,"endpoint":1,"linkquality":60,"groupID":0}' +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
zigbee2mqtt:info  2020-06-29 20:09:30: MQTT publish: topic 'zigbee2mqtt/ledstripe', payload '{"state":"ON","brightness":0,"color":{"x":0.7006,"y":0.2993},"color_temp":37}'

```

Koenkk, 2020-06-30:
The colors you are setting are (almost) black. Try e.g. `#fffb00` -> should be yellow.

crackytsi, 2020-06-30:
yes, this is yellow.
But the colors was dark red in the past. If I now set this colors (imitating a sunrise) they are light-blue like the sky. 
That is very strange.
I have also the same issue with my other "Müller Licht" Tint GU10 Lamps and the only thing I was changing was to upgrade zigbee2mqtt.

Koenkk, 2020-07-01:
This is probably related to the gamma correction that has been added: https://en.wikipedia.org/wiki/Gamma_correction 

You probably have to adjust your hex colors slightly.

crackytsi, 2020-07-02:
Thank you @Koenkk 
So to summarize
a) the light-blue is some kind of "error-message" of the bulb, that it can't display the requested color.
b) in the past the # character before the bulb hex-color-code was really optional and is now required
c) gamma correction is now applied to the value.

I have some questions and the hope you can help me to find answers:

- Can I disable the gamma correction? I'm not happy about the changed colors.

-  I think the gamma correction is applied by converting RGB to xyY, right? Can I directly set xyY colors to the "Müller Licht" bulbs using zigbee2mqtt?

-  Is there a way to convert the old color outputs into the new (gamma corrected) ones?

Again, thank you very much for your patience with me and the good help :)

Koenkk, 2020-07-03:
- This cannot be disabled currently
- Yes, you can also set it directly and this is actually preferred (as there is no risk of setting unsupported colors), no gamma correction will be applied in this case. An example to set x and y can be found here: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset
- You can use https://www.easyrgb.com/en/convert.php#inputFORM , pick "sRGB (0-255)" and pick the second and third value of Yxy (e.g. `Yxy          =   44.038  0.33536  0.56881`, `x = 0.33536`, `y= 0.56881`)

crackytsi, 2020-07-06:
@Koenkk: Understand. Thank you very much :)
Sorry, but unforunately I'm still struggeling, even after lot of googling and documentation reading:

Actually I stuck again: FHEM can't set values using x/y values, meanwhile I don't know how to convert x y values into (the new) gamma corrected RGB Format.

Questions:
-How can I convert the xy values to zigbee2mqtt (gamma-corrected) RGB values?
(There are parameters like a  rotation-degree and different values like A,B,C, D65(daylight)...) 
-Another big question: what is the Y value of a x/y color? Is this brightness or color_temp and how can I convert this into "Y"-format? (My plan is to convert original RGB value into xyY and than convert it back to gamma corrected zigbee2mqtt rgb format)

closed 2020-08-16 by github-actions[bot]

CSchlipp, 2020-09-27:
@Koenkk : I basically have the same problem as @crackytsi , any new development here? Unfortunately, this issue has been closed by a bot, but not answered completely...
I'd rather like to have Z2M to set exactly the colors requested instead of performing some unwanted and obviously in some cases wrong conversions. 
Is there really no chance of adding an additional parameter to the color section of the JSON like `"gammacorrection":"false"` or directly make the gamma value editable like `"gamma": 1.0`?

Koenkk, 2020-09-27:
@CSchlipp could you provide me the obviously wrong conversions?

CSchlipp, 2020-10-03:
Sure, e.g. #360603 should show a dark red color:
If I send `{"state": "ON", "brightness": 50, "color": { "hex": "#360603" }, "transition": 0 }` to my Hue bulb, it instead shows a pink color.
Z2M transform this into a state of `'{"brightness":50,"color":{"x":0.5139,"y":0.2318},"color_temp":39,"last_seen":"2020-10-03T12:59:45+02:00","state":"ON","update_available":false}'`

Choosing a different setting for brightness does not have any influence on the color itself, e.g. brightness: 1 would simply dim the pink colored light.

Checking the EasyRGB website mentioned in this ticket, the values of Z2M do not match their calculations either:
HTML **#360603** == Yxy 0.921  **0.59286  0.34074** vs. **0.5139 0.2318**

In the meantime I managed to implement the formulas RGB->XYZ->Yxy given by EasyRGB with an OpenHAB rule, which, for the same Hex value, results in
`{"state": "ON", "brightness": 50, "color": { "x": 0.5929064720796391, "y": 0.3407128376294455 }, "transition": 0 }` 
This is accepted by Z2M as `'{"brightness":50,"color":{"x":0.5929064720796391,"y":0.3407128376294455},"color_temp":1357,"last_seen":"2020-10-03T13:14:00+02:00","state":"ON","update_available":false}'` 
and indeed shows a red color.

Koenkk, 2020-10-05:
Can you check if this has been fixed in the latest dev branch. It turns out this had nothing to do with the gamma correction but with a bug (fixed in https://github.com/Koenkk/zigbee-herdsman-converters/commit/e7427b6e0622cb5e6d032f6d923eab85161f970a). Before this fix I indeed get pink with `#360603`, after the fix red.

CSchlipp, 2020-10-07:
Updated to the latest dev and the bulb is showing a red color for me as well now.

The values calculated by Z2M `'{"brightness":50,"color":{"x":0.7006,"y":0.2993},"color_temp":37,"last_seen":"2020-10-07T16:24:59+02:00","state":"ON","update_available":false}'` still differ from my EasyRGB calculations
 `'{"brightness":50,"color":{"x":0.5929064720796391,"y":0.3407128376294455},"color_temp":1357,"last_seen":"2020-10-07T16:34:42+02:00","state":"ON","update_available":false}'`, but the actual color is pretty close (a bit darker, which is probably the gamma correction).

So far this looks good to me and eliminates my need for the manual conversion! Thanks!