Koenkk/zigbee2mqtt #3808: Correct way to avoid having to reset every bulb after re-image?

GForce2010, 2020-06-25:
I am currently running V1.13.0 with a Texas Instruments LAUNCHXL-CC26X2R1 running 20191106 revision of Firmware on a Raspberry Pi 3+

About a month ago I had some faults that seemed to point to the SD card starting to fail so I installed a new card and set Zigbee2MQTT up afresh using the same config, the same Pan ID, channel and network key etc.

Unfortunately I was unable to start Zigbee2MQTT, it got to the point of trying to start Zigbee Herdsmen but would then fail. If I changed the Pan ID then it started without issue.

To get round this I reset all my devices so they would connect to this new network.

Unfortunately I am now in a similar situation where I need to replace my Pi and wanted to know what the process would be to allow me to setup Zigbee2MQTT on my new device without having to change any config allowing all my devices to connect back up without having to reset everything again.

Thanks.

MPM1107, 2020-06-25:
There is a lot more data stored about the network than just the channel and PAN and all of it is important to be able to properly start the network, that's why your first try failed. It's all stored in the `data` folder, so you should be able to just copy that whole folder over and be good to go.

GForce2010, 2020-06-25:
Will give that a go.

Thanks for your help.

closed 2020-06-26 by GForce2010