Koenkk/zigbee2mqtt #3811: Node Red -> Simple basic question (Turn on/ off with switch node a zigbee device)

get-command, 2020-06-26:
Hello community,

I am new here and need your help. i use the super device CC2531.

If I replace the switch node with an inject not, then I can switch the lamp/bulb on and off. When I change the zigbee-light-node with a RF433 light/bulb then function this very well. This is a sign that the switch node is set correctly. The command is passed on.   how can I start the execute node? I couldn't read it anywhere.

i get an error message:

Error: Command 0x00158d0XXXXXXXX/1 genOnOff.on({“onOff”:0}, {“timeout”:6000,“manufacturerCode”:null,“disableDefaultResponse”:false}) failed (Error: Timeout - 40061 - 1 - 210 - 6 - 11 after 6000ms)

I try to explain that with these screeshots:

[image]

[image]

Thank you in advance.

sjorge, 2020-06-26:
I'm guessing you're using https://github.com/hobbyquaker/node-red-contrib-zigbee, might have better luck asking there for help.

get-command, 2020-06-27:
> 
> 
> I'm guessing you're using https://github.com/hobbyquaker/node-red-contrib-zigbee, might have better luck asking there for help.

thank you for the tip. we have already solved it. here: https://github.com/hobbyquaker/node-red-contrib-zigbee/issues/113

closed 2020-06-27 by Koenkk