Koenkk/zigbee2mqtt #3812: Add support for Blitzwolf BW-SHP13...gets recognised as TuYa TS0121...how to proceed?

SaturnusDJ, 2020-06-26:
Received the new Blitzwolf BW-SHP13 smart plugs today so I was starting up the procedure for adding the support (https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html).

Upon pairing they got recognised as TuYa TS0121. I did not test them further but from https://www.zigbee2mqtt.io/devices/TS0121.html it already becomes clear it does not support electricity monitoring. Please instruct me how to proceed.

sjorge, 2020-06-26:
There is a chance it is actually the rebrand of the TuYa device with some more optioned configured in the firmware...

Perhaps a good start would be to append `fz.metering_power` to the fromZigbee array for the TS0121 and see if you get anything useful.

SaturnusDJ, 2020-06-26:
Thanks. Got that suggestion too on Tweakers forum. Also added `fz.ignore_basic_report`. No MQTT output for `power` yet.

In the `metering_power` code is mentioned:
```
* When using this converter also add the following to the configure method of the device:
* await readMeteringPowerConverterAttributes(endpoint);
```

I have not been able to figure this out. Tried a lot of code that is used for other devices, can't get it to work.

It seems to me that the output as shown in the first post is the cumulative power in decawatt. Needs to go times 10 to make sense for the device I am running on the plug.

sjorge, 2020-06-26:
```
    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.metering_power],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await configureReporting.onOff(endpoint);
            await readMeteringPowerConverterAttributes(endpoint);
            await configureReporting.instantaneousDemand(endpoint);
        },
    },
```

Just taking a stab in the dark here, the endpoint might be wrong. But based on the log entry i think EP1 is the good one.

SaturnusDJ, 2020-06-26:
Thanks, this is the error it gives:
```
zigbee2mqtt:error 2020-06-26 22:53:19: Failed to configure 'ac', attempt 3 (Error: ConfigureReporting 0x..../1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 7015 - 1 - 6 - 6 - 7 after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:341:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

sjorge, 2020-06-26:
```
    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.metering_power],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await readMeteringPowerConverterAttributes(endpoint);
            await configureReporting.instantaneousDemand(endpoint);
        },
    },
```
Lets try without reporting for genOnoff.

SaturnusDJ, 2020-06-26:
Thanks. Next error is:
```
Failed to configure 'ac', attempt 2 (Error: Read 0x..../1 seMetering(["multiplier","divisor"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:192:23)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

sjorge, 2020-06-26:
Interesting, so it doens't support reporting at all probably.

```
    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.metering_power],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await readMeteringPowerConverterAttributes(endpoint);
        },
    },
```

Lets try the absolute minimal then.

SaturnusDJ, 2020-06-26:
The same error comes up.

sjorge, 2020-06-26:
I'm guessing it's not liking the readMeteringPowerConverterAttributes also :(
No idea how to proceed from here. Hopefully Koen will once he's online

SaturnusDJ, 2020-06-26:
Thanks for the time. 👌

Koenkk, 2020-06-27:
When did you receive `Received Zigbee message from 'ac', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,431]}' from endpoint 1 with groupID 0` ? When adding `fz.metering_power` this should already work.

SaturnusDJ, 2020-06-27:
That must have been after adding `fz.metering_power`. So that it looks like:
```

    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.metering_power],
        },
    },
```

Koenkk, 2020-06-28:
But you don't get any power messages now?

SaturnusDJ, 2020-06-28:
Reading with MQTT Explorer: Power field stays at "null".

Koenkk, 2020-06-28:
Can you try with:

```js
    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.metering_power],
        toZigbee: [tz.on_off],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        },
    },
```

attach a load to the plug, stop zigbee2mqtt and share your debug log and `data/dabase.db`?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

SaturnusDJ, 2020-06-28:
Ok thanks.

One of the Blitzwolf BW-SHP13 is called ac, the other has no friendly name yet.

Filtered log:
```
info  2020-06-28 18:25:40: Logging to console and directory: '/app/data/log/2020-06-28.18-25-40' filename: log.txt
debug 2020-06-28 18:25:40: Removing old log directory '/app/data/log/2020-06-26.18-38-47'
debug 2020-06-28 18:25:41: Loaded state from file /app/data/state.json
info  2020-06-28 18:25:41: Starting zigbee2mqtt version 1.14.0 (commit #9009de2)
info  2020-06-28 18:25:41: Starting zigbee-herdsman...
debug 2020-06-28 18:25:41: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
info  2020-06-28 18:25:42: zigbee-herdsman started
info  2020-06-28 18:25:42: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
debug 2020-06-28 18:25:42: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-06-28 18:25:42: Currently 14 devices are joined:
info  2020-06-28 18:25:43: ac (0x....1): TS0121 - TuYa 10A UK or 16A EU smart plug (Router)
info  2020-06-28 18:25:43: 0x....2 (0x....2): TS0121 - TuYa 10A UK or 16A EU smart plug (Router)
info  2020-06-28 18:25:43: Configuring 'ac'
info  2020-06-28 18:25:43: Successfully configured 'ac'
info  2020-06-28 18:25:43: Configuring '0x....2'
info  2020-06-28 18:25:43: Connected to MQTT server
info  2020-06-28 18:25:43: MQTT publish: topic 'zigbee2mqtt/ac', payload '{"state":"ON","linkquality":0,"power":null}'
info  2020-06-28 18:25:43: MQTT publish: topic 'zigbee2mqtt/0x....2', payload '{"state":"ON","linkquality":2}'
info  2020-06-28 18:25:43: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.0","commit":"9009de2","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}'
info  2020-06-28 18:25:44: Successfully configured '0x....2'
debug 2020-06-28 18:26:11: Received MQTT message on 'zigbee2mqtt/ac/set' with data 'OFF'
debug 2020-06-28 18:26:11: Publishing 'set' 'state' to 'ac'
info  2020-06-28 18:26:11: MQTT publish: topic 'zigbee2mqtt/ac', payload '{"state":"OFF","linkquality":0,"power":null}'
debug 2020-06-28 18:26:11: Received Zigbee message from 'ac', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-06-28 18:26:11: MQTT publish: topic 'zigbee2mqtt/ac', payload '{"state":"OFF","linkquality":0,"power":null}'
debug 2020-06-28 18:26:14: Received MQTT message on 'zigbee2mqtt/ac/set' with data 'ON'
debug 2020-06-28 18:26:14: Publishing 'set' 'state' to 'ac'
info  2020-06-28 18:26:14: MQTT publish: topic 'zigbee2mqtt/ac', payload '{"state":"ON","linkquality":0,"power":null}'
debug 2020-06-28 18:26:14: Received Zigbee message from 'ac', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-06-28 18:26:14: MQTT publish: topic 'zigbee2mqtt/ac', payload '{"state":"ON","linkquality":0,"power":null}'
debug 2020-06-28 18:27:16: Received Zigbee message from 'ac', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":64}' from endpoint 1 with groupID 0
debug 2020-06-28 18:27:16: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":30,"appVersion":64}'
debug 2020-06-28 18:27:36: Received Zigbee message from '0x....2', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,100]}' from endpoint 1 with groupID 0
debug 2020-06-28 18:28:00: Received Zigbee message from '0x....2', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":64}' from endpoint 1 with groupID 0
debug 2020-06-28 18:28:00: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":30,"appVersion":64}'
debug 2020-06-28 18:28:54: Received MQTT message on 'zigbee2mqtt/ac/set' with data 'OFF'
debug 2020-06-28 18:28:54: Publishing 'set' 'state' to 'ac'
debug 2020-06-28 18:28:55: Received Zigbee message from 'ac', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-06-28 18:28:55: MQTT publish: topic 'zigbee2mqtt/ac', payload '{"state":"OFF","linkquality":0,"power":null}'
info  2020-06-28 18:28:55: MQTT publish: topic 'zigbee2mqtt/ac', payload '{"state":"OFF","linkquality":0,"power":null}'
debug 2020-06-28 18:30:03: Received Zigbee message from 'ac', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1218]}' from endpoint 1 with groupID 0
debug 2020-06-28 18:30:29: Received Zigbee message from 'ac', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":64}' from endpoint 1 with groupID 0
debug 2020-06-28 18:30:29: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":30,"appVersion":64}'
debug 2020-06-28 18:30:41: Saving state to file /app/data/state.json
debug 2020-06-28 18:30:51: Saving state to file /app/data/state.json
[... cut ...] ,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"currentSummDelivered":[0,100]}},"genOnOff":{"attributes":{"32769":1,"32770":2,"onOff":1,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x....","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x....","endpointID":1}]}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1593361680987}
```

Koenkk, 2020-06-29:
Can you try with:

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK or 16A EU smart plug',
    supports: 'on/off',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 3},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await configureReporting.instantaneousDemand(endpoint);
    },
},
```

attach a load and provide the complete log again? (from the point where you attached the load).

SaturnusDJ, 2020-06-29:
Thanks.

Here it is:

```
zigbee2mqtt:debug 2020-06-29 16:23:19: Received Zigbee message from 'ac', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1224]}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-06-29 16:23:28: Failed to configure 'ac', attempt 3 (Error: ConfigureReporting 0x..../1 seMetering([{"attribute":"instantaneousDemand","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 7015 - 1 - 8 - 1794 - 7 after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:341:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

Koenkk, 2020-06-29:
Can you try with:

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK or 16A EU smart plug',
    supports: 'on/off',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 4},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        endpoint.saveClusterAttributeKeyValue('seMetering', {
            divisor: 100,
            multiplier: 1,
        });
    },
},
```

with this I expect the `energy` attribute to change to a meaningful value after some time. This will indicate the total deliver energy in kWh.

SaturnusDJ, 2020-06-29:
Yes thanks that is working as you describe. 👌 

As I have no experience with other electricity monitoring plugs...what else can it do? Or what can Z2M do with it?
At time of buying I expected that the plug would publish the current watt usage every x seconds but so far the only thing that shows up is a message for each decawatt increase (only) as they happen.

Koenkk, 2020-06-29:
> At time of buying I expected that the plug would publish the current watt usage every x seconds 

We need to check if the plug supports this. When running with the code below, does the device configure successfully? 

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK or 16A EU smart plug',
    supports: 'on/off',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 4},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        endpoint.saveClusterAttributeKeyValue('seMetering', {
            divisor: 100,
            multiplier: 1,
        });
       await endpoint.read('seMetering', ['instantaneousDemand']);
    },
},
```

piotrekoo, 2020-06-29:
Hi,

Actually I got three of them today. Bought hoping they'll work like Innr SP 120, but supporting higher wattage.

The device doesn't configure using this code
```
Failed to configure '0x...', attempt 1 (Error: Read 0x.../1 seMetering(["instantaneousDemand"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
```

SaturnusDJ, 2020-06-29:
Here the added line of code from https://github.com/Koenkk/zigbee2mqtt/issues/3812#issuecomment-651200047 seems to do nothing. No error and no new features. Or is remove and pair again needed?

piotrekoo, 2020-06-29:
@SaturnusDJ I suppose configuration takes place when pairing. I got the error when repairing the plug.

piotrekoo, 2020-06-29:
deCONZ is somehow able to read the _haElectricalMeasurement_ cluster of this device: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2988

I'll try some tinkering on my own, but if you have any suggestions or snippets to test, I'll be happy to test them.

Koenkk, 2020-06-29:
It is also possible to force the configure via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure and watch the logging at this moment.

> deCONZ is somehow able to read the haElectricalMeasurement

Try with:

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK or 16A EU smart plug',
    supports: 'on/off',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 4},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering', 'haElectricalMeasurement']);
        // Return UNSUPPORTED_ATTRIBUTE upon reading divisor/multiplier
        endpoint.saveClusterAttributeKeyValue('seMetering', {
            divisor: 100,
            multiplier: 1,
        });

        await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.activePower(endpoint);
    },
},
```

SaturnusDJ, 2020-06-29:
Gives:

```
Configuring 'ac'
zigbee2mqtt:debug 2020-06-29 19:49:57: Received Zigbee message from 'ac', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-06-29 19:49:57: Failed to configure 'ac', attempt 1 (Error: Read 0x..../1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:192:23)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

Koenkk, 2020-06-29:
And:

```js
{
    zigbeeModel: ['TS0121'],
    model: 'TS0121',
    description: '10A UK or 16A EU smart plug',
    supports: 'on/off',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.metering_power, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 4},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering', 'haElectricalMeasurement']);
        // Return UNSUPPORTED_ATTRIBUTE upon reading divisor/multiplier
        endpoint.saveClusterAttributeKeyValue('seMetering', {
            divisor: 100,
            multiplier: 1,
        });

        await configureReporting.activePower(endpoint);
    },
},
```

SaturnusDJ, 2020-06-29:
Gives:

```
zigbee2mqtt:info  2020-06-29 19:55:38: Configuring 'ac'
zigbee2mqtt:error 2020-06-29 19:56:02: Failed to configure 'ac', attempt 1 (Error: ConfigureReporting 0x..../1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":300,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 7015 - 1 - 3 - 2820 - 7 after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:341:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

Koenkk, 2020-06-29:
do you get any power values when now attaching a load to the plug? 

@piotrekoo do you get the same error?

SaturnusDJ, 2020-06-29:
Looks like it:
```
20:06:54: Received Zigbee message from 'ac', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1234]}' from endpoint 1 with groupID 0```

piotrekoo, 2020-06-29:
@Koenkk Yes, I get the same error. I'm tinkering now a bit with overriding the reporting parameters. Just checking if it configures with the values used by deCONZ.

piotrekoo, 2020-06-29:
Got my old ConBee from the drawer.

At first the device reports only On/Off, Simple Metering and Basic clusters. After clicking Read Simple Descriptors it reported Groups, Scenes, Electrical Measurement, as well as OTAU and Time clusters.

It reports RMS Voltage, RMS Current and Active Power on demand (ZCL: Read Attributes). When I tried setting Reporting from GUI, I got UNSUP_GENERAL_COMMAND.

I have near-zero knowledge of how Zigbee works underneath but it looks like the device doesn't support configuring reporting.

That's how it looks sniffed in Wireshark:

[image]

[image]

Koenkk, 2020-06-30:
@piotrekoo thanks for sniffing this, those findings seems to be consistent with ours. The current summation delivered is exposed as `energy` with the following config: https://github.com/Koenkk/zigbee2mqtt/issues/3812#issuecomment-651171671

Next question is on how to deal with the voltage, current and power (as the device does not support reporting of those). One thing we can do is poll those every X seconds (60?). But this will break certain use cases which require more frequent updates on large changes (e.g. the kettle detection use case).

piotrekoo, 2020-06-30:
As these plugs support 16A and "monitoring", many will buy them to monitor kettles, washing machines.., just as I did. The choice is pretty limited. You get either 16A support or monitoring or.. spend around €50 on a Salus or a Bitron to get both ;) 
The MCLH-03 from LifeControl needs polling too if I understand the code correctly. Possibly a similar solution can be used here.

Just noticed in logs that the BW-SHP13 also sends regular _Basic_ reports:
```No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":30,"appVersion":64}'```
So a `fz.ignore_basic_report` converter will be needed too.

daxy01, 2020-06-30:
@Koenkk Thanks for your help here! In case it would speed-up this, would you like access to my test server so you can just play around? I sent you a DM on Tweakers in case that'll help. Thanks!

daxy01, 2020-06-30:
@piotrekoo That seems to be quite a good call! 😀 I just tested the code from MCLH-03 and added those specifics to the TS0121 and now I do get current usage! Still an error and no clue how to fix this though. This is the config I used:
```
    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement_power],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            await configureReporting.onOff(endpoint);
            await readEletricalMeasurementPowerConverterAttributes(endpoint);
        },
        onEvent: async (type, data, device) => {
            // This device doesn't support reporting correctly.
            // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(store[device.ieeeAddr]);
            } else if (!store[device.ieeeAddr]) {
                store[device.ieeeAddr] = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {
                        // Do nothing
                    }
                }, 10*1000); // Every 10 seconds
            }
        },
    },
```

This is in the logs now, last ones showing power usage:
```
error 2020-06-30 13:46:57: Failed to configure '0x842e14fffee1aad4', attempt 3 (Error: ConfigureReporting 0x842e14fffee1aad4/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 51224 - 1 - 18 - 6 - 7 after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:341:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
    
debug 2020-06-30 13:48:58: Received Zigbee message from '0x842e14fffee1aad4', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":64}' from endpoint 1 with groupID 0

debug 2020-06-30 13:48:58: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":30,"appVersion":64}'

debug 2020-06-30 13:48:58: Received Zigbee message from '0x842e14fffee1aad4', type 'attributeReport', cluster 'genBasic', data '{"65503":"�\u0011t&\u0017�\u0011t&\u0013�\u0011t&\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005"}' from endpoint 1 with groupID 0

debug 2020-06-30 13:48:58: No converter available for 'TS0121' with cluster 'genBasic' and type 'attributeReport' and data '{"65503":"�\u0011t&\u0017�\u0011t&\u0013�\u0011t&\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005"}'

info  2020-06-30 13:49:35: MQTT publish: topic 'zigbee2mqtt-zolder/0x842e14fffee1aad4', payload '{"state":"ON","linkquality":99,"power":41,"current":214,"voltage":226}'

debug 2020-06-30 13:49:45: Received Zigbee message from '0x842e14fffee1aad4', type 'readResponse', cluster 'haElectricalMeasurement', data '{"rmsVoltage":227,"rmsCurrent":202,"activePower":41}' from endpoint 1 with groupID 0
```

piotrekoo, 2020-06-30:
I'm out of home so I won't be able to check it on my own soon.

I got rid of remaining ```configureReporting``` as the device definitely doesn't support this command. Added ```fz.ignore_basic_report```, so we're no longer bothered with mysterious 65503 and 65506 values. Added ```fz.metering_power```, so the ```currentSummDelivered``` is converted, but I'm not sure if we should take it or ignore it.

```javascript
    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement_power, fz.metering_power, fz.ignore_basic_report],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            await readEletricalMeasurementPowerConverterAttributes(endpoint);
        },
        onEvent: async (type, data, device) => {
            // This device doesn't support reporting correctly.
            // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(store[device.ieeeAddr]);
            } else if (!store[device.ieeeAddr]) {
                store[device.ieeeAddr] = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {
                        // Do nothing
                    }
                }, 10*1000); // Every 10 seconds
            }
        },
    },
```

daxy01, 2020-06-30:
I can confirm that @piotrekoo his code is working for me. It does throw the error below 3 times, but then all is fine. I'm not sure if this is a zigbee2mqtt issue or Blitzwolf, but I can live with it as it has no further impact 😀

Although this is working fine for this Blitzwolf, I do not know if this (potentially) has any complications for other (let's say older) TuYa TS0121's out there.

```
zigbee2mqtt:info  2020-06-30 20:32:20: MQTT publish: topic 'zigbee2mqtt-zolder/Wasmachine', payload '{"state":"ON","linkquality":92,"power":0,"current":0,"voltage":226}'
zigbee2mqtt:info  2020-06-30 20:32:20: Configuring 'Wasmachine'
zigbee2mqtt:debug 2020-06-30 20:32:20: Received Zigbee message from 'Wasmachine', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-06-30 20:32:20: Failed to configure 'Wasmachine', attempt 2 (Error: Read 0x842e14fffee1aad4/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:192:23)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

SaturnusDJ, 2020-06-30:
Obviously the same here. Nice work.

I thought that here it should be *voltage * current = power* however; not true.

piotrekoo, 2020-07-01:
I added missing multipliers and divisors that went forgotten after copy-pasting the MCLH-03 code. Hopefully, this will make the _voltage * current = power_ formula valid again ;)
Again, I'm not home and I left the SHP13 without any load, so I can't test whether it works properly.

```javascript
    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement_power, fz.metering_power, fz.ignore_basic_report],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {
                divisor: 100,
                multiplier: 1,
            });
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1,
                acVoltageDivisor: 1,
                acCurrentMultiplier: 1,
                acCurrentDivisor: 1000,
            });
            await readEletricalMeasurementPowerConverterAttributes(endpoint);
        },
        onEvent: async (type, data, device) => {
            // This device doesn't support reporting correctly.
            // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(store[device.ieeeAddr]);
            } else if (!store[device.ieeeAddr]) {
                store[device.ieeeAddr] = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {
                        // Do nothing
                    }
                }, 10*1000); // Every 10 seconds
            }
        },
    },
```

SaturnusDJ, 2020-07-01:
Hmm here the current is still doubtful.

``
  "power": 114,
  "current": 633,
  "voltage": 227
````

Power and voltage make sense, verified before as well at least in the big scheme, therefore the current should be about 114W / 227V = 0.5022A = 502mA.
502 vs 633...A bit too much off.  

I will have a check with a standalone watt meter to see if the watts/power is accurate.

SaturnusDJ, 2020-07-01:
Auch...
[ Wall socket ] - [ Blitzwolf BW-SHP13 ] - [ Watt meter ] - [ Freezer ]
Blitzwolf reports 127W
Watt meter reports 111W

swapping the meters:
Blitzwolf reports 122W
Watt meter reports 108W

The watt meter could be 10 years old and is just a basic one. 
I guess one of them is not doing so well. 🤣

piotrekoo, 2020-07-01:
[ Socket ] - [ Watt-o-meter ] - [ BW-SHP13 ] - [ Laptop charger ]

|| Power  | Current |
-------------|--------|---------|
Watt-o-meter | 32.8 W | 0.241 A |
BW-SHP13 | 36 W | 0.272 A |

When readings are about 10-15% off, I think I get why these sockets are so cheap 🙄
I'll try to get an incandescent light bulb and have a test with a known load.

Koenkk, 2020-07-01:
@daxy01 thanks for the offer, but based on the good progress here I don't think that is needed.

The configuration from https://github.com/Koenkk/zigbee2mqtt/issues/3812#issuecomment-652266693 looks good to me.

Regarding the off power measurements, please figure out if it's a static deviation (e.g. always 10 watts) or a percentage deviation (e.g. always 15%). If it's a percentage one we should be able to easily fix this by changing:

```js
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1,
                acVoltageDivisor: 1,
                acCurrentMultiplier: 1,
                acCurrentDivisor: 1000,
            });
```

to (for e.g. 0.85 correction)

```js
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1,
                acVoltageDivisor: 1,
                acCurrentMultiplier: 1,
                acCurrentDivisor: 1000,
                acPowerMultiplier: 0.85, // Power pulished to MQTT = reported power * 0.85
                acPowerDivisor: 1,
            });
```

piotrekoo, 2020-07-01:
Well, we've got four observations. That's not enough to tell the final correction value, but I'd assume a correction by 0.88 for current and power.
I'm wondering whether it's correct for higher power/current values.

SaturnusDJ, 2020-07-01:
I will check 1000W on two plugs and one normal meter tomorrow.

Worse case the correction is variable per load and per plug, that would be something...

dimaiv7, 2020-07-02:
Hello everybody. I also followed last days and nights here and tried everything.
This works perfectly for me:
`    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement_power, fz.metering_power, fz.ignore_basic_report],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {
                divisor: 100,
                multiplier: 1,
            });
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1,
                acVoltageDivisor: 1,
                acCurrentMultiplier: 1,
                acCurrentDivisor: 1000,
            });
//            await readEletricalMeasurementPowerConverterAttributes(endpoint);
        },
        onEvent: async (type, data, device) => {
            // This device doesn't support reporting correctly.
            // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(store[device.ieeeAddr]);
            } else if (!store[device.ieeeAddr]) {
                store[device.ieeeAddr] = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {
                        // Do nothing
                    }
                }, 10*1000); // Every 10 seconds
            }
        },
    },`

dimaiv7, 2020-07-02:
Hello everybody. I also followed last days and nights here and tried everything.
This works perfectly for me:
 ```js
    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement_power, fz.metering_power, fz.ignore_basic_report],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {
                divisor: 100,
                multiplier: 1,
            });
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1,
                acVoltageDivisor: 1,
                acCurrentMultiplier: 1,
                acCurrentDivisor: 1000,
            });
//            await readEletricalMeasurementPowerConverterAttributes(endpoint);
        },
        onEvent: async (type, data, device) => {
            // This device doesn't support reporting correctly.
            // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(store[device.ieeeAddr]);
            } else if (!store[device.ieeeAddr]) {
                store[device.ieeeAddr] = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {
                        // Do nothing
                    }
                }, 10*1000); // Every 10 seconds
            }
        },
    },
 ```

DBa2016, 2020-07-03:
I tried the configs as suggested by @dimaiv7 and by @piotrekoo ... None of these worked for me - "readEletricalMeasurementPowerConverterAttributes" errors out with an exception about UNSUPPORTED_ATTRIBUTE:
````
Failed to configure '0x842e14fffe51c655', attempt 1 (Error: Read 0x842e14fffe51c655/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:192:23)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
````

I modified readEletricalMeasurementPowerConverterAttributes to try the attributes one by one in a loop and report the failing ones - all six of them failed.

Also, after configuration, I get periodically following messages (I put some logging into the "catch" block of the "onEvent" handler):
````
Read 0x842e14fffe51c655/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (TypeError: Cannot read property 'frame' of null)
````

mabosch74, 2020-07-03:
> owerConverterAttributes

> "readEletricalMeasurementPowerConverterAttributes" errors out with an exception about UNSUPPORTED_ATTRIBUTE:

look in the typo:
readEle**c**tricalMeasurementPowerConverterAttributes

dimaiv7, 2020-07-03:
If you look closely, the line is by me commented out. Works perfectly with 6 plugs.
Can anyone say what is the line for?

DBa2016, 2020-07-03:
> 
> 
> If you look closely, the line is by me commented out. Works perfectly with 6 plugs.
> Can anyone say what is the line for?

Yeah, I tried with it commented out - all the same, no metering, only on/off and linkquality.

DBa2016, 2020-07-03:
> readEle**c**tricalMeasurementPowerConverterAttributes

While you are certainly right, the typo is in the entire file, so the correct function is picked.

SaturnusDJ, 2020-07-03:
Bad news...
Here are my supposed 1000W (quick) test results.

The code used is:
```
    {
        model: 'TS0121',
        description: '10A UK or 16A EU smart plug',
        supports: 'on/off',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement_power, fz.metering_power, fz.ignore_basic_report],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {
                divisor: 100,
                multiplier: 1,
            });
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1,
                acVoltageDivisor: 1,
                acCurrentMultiplier: 1,
                acCurrentDivisor: 1000,
            });
            await readEletricalMeasurementPowerConverterAttributes(endpoint);
        },
        onEvent: async (type, data, device) => {
            // This device doesn't support reporting correctly.
            // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(store[device.ieeeAddr]);
            } else if (!store[device.ieeeAddr]) {
                store[device.ieeeAddr] = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {
                        // Do nothing
                    }
                }, 10*1000); // Every 10 seconds
            }
        },
    },

```

In the below results mind the axis scale of the graphs, they are set specifically to see the difference.

[ Wall socket ] - [ Non-pro watt meter ] - [ Plug 1 ] - [ Plug 2 ] - [ AC ]

Power:
[image]
**Here the non-pro watt meter towards the end was giving values around 726-734W.**

Current:
[image]

Voltage:
[image]

Swapping the plugs. Note that the AC is reaching its target power.

[ Wall socket ] - [ Non-pro watt meter ] - [ Plug 2 ] - [ Plug 1 ] - [ AC ]

Power:
[image]
**Here the non-pro watt meter towards the end was giving values around 774-779W.**

[... 88 lines cut ...]
13:01:57 | plug2 | 850 | 3825 | 231
13:02:01 | plug1 | 815 | 3660 | 226
13:02:07 | plug1 | 819 | 3672 | 226
13:02:07 | plug2 | 855 | 3851 | 231
13:02:17 | plug1 | 821 | 3684 | 225
13:02:17 | plug2 | 855 | 3851 | 230
13:02:27 | plug1 | 822 | 3696 | 226
13:02:27 | plug2 | 859 | 3863 | 231
13:02:37 | plug1 | 826 | 3708 | 226
13:02:37 | plug2 | 864 | 3876 | 231
13:02:38 | plug2 | 864 | 3876 | 231
13:02:44 | plug1 | 826 | 3708 | 226
13:02:47 | plug1 | 829 | 3720 | 225
13:02:47 | plug2 | 866 | 3888 | 231
13:02:57 | plug1 | 831 | 3732 | 225
13:02:57 | plug2 | 866 | 3888 | 230
13:03:07 | plug1 | 833 | 3732 | 225
13:03:07 | plug2 | 870 | 3926 | 230
13:03:17 | plug1 | 836 | 3744 | 225
13:03:17 | plug2 | 872 | 3926 | 230
13:03:20 | plug2 | 872 | 3926 | 230
13:03:21 | plug2 | 872 | 3926 | 230
13:03:27 | plug1 | 838 | 3755 | 225
13:03:27 | plug2 | 875 | 3939 | 230
13:03:28 | plug1 | 838 | 3755 | 225

DBa2016, 2020-07-03:
@SaturnusDJ how did your code work? It is missing the "zigbeeModel" attribute...

SaturnusDJ, 2020-07-03:
@DBa2016 
It is code from here, not mine. Nano was being annoying with copy past, let me fix it now then.

daxy01, 2020-07-03:
I've used this one which seems to work flawlessly: https://github.com/Koenkk/zigbee2mqtt/issues/3812#issuecomment-652928634
I do like the 30sec interval better than the 10sec one. No errors anymore. I would vote to use this version 😀 

There is still the outstanding question of what impact this might have to the original TS0121's and how they would react on this change. I believe this is quite important to understand prior to committing this fix. Perhaps @Koenkk can comment on this based on his experience?

DBa2016, 2020-07-03:
I found something now... Lookslike the metering mesages are received over ZB and posted to MQTT:
````
zigbee2mqtt:debug 2020-07-03 16:59:22: Received Zigbee message from '0x842e14fffe51c655', type 'readResponse', cluster 'haElectricalMeasurement', data '{"rmsVoltage":225,"rmsCurrent":0,"activePower":0}' from endpoint 1 with groupID null
zigbee2mqtt:info  2020-07-03 16:59:22: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe51c655', payload '{"power":0,"current":0,"voltage":225,"linkquality":200,"energy":0}'
````
However, none of the metering attributes are posted to discovery, and also Z2MA does not show anything in that context.

daxy01, 2020-07-03:
> I found something now... Lookslike the metering mesages are received over ZB and posted to MQTT:
>
> ```
> zigbee2mqtt:debug 2020-07-03 16:59:22: Received Zigbee message from '0x842e14fffe51c655', type [...]

If you see those in your logs, it seems to work!
As for "posted to discovery", what do you mean? Is that Home-assistant by any chance?
If that's the case you'll need to edit the file "homeassistant.js" in your z2m directories and copy the MCLH-03 line to TS0121. This way HA will know about these values coming in.

[69 more replies]

closed 2020-07-20 by Koenkk