Koenkk/zigbee2mqtt #3813: Gledopto GL-C-007 thinks it's not supported

matstrange, 2020-06-26:
# Bug Report
## What happened
I purchased a new Gledopto RGWB controller (GL-C-007) after checking compibility on the supported device list. However, after eventually getting it paired, the z2m logs state it is not supported:

I can see the device announce when I power it up
`zigbee2mqtt:info  2020-06-26 12:38:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ada3420"}}'`

And it does register on the zigbee network map in HA.
[image]

But when I issue an instruction I get this:
`zigbee2mqtt:warn  2020-06-26 12:01:23: Device with modelID 'GL-C-007' is not supported.
zigbee2mqtt:warn  2020-06-26 12:01:23: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

The database.db implies that the interview completed successfully.
`{"id":20,"type":"Router","ieeeAddr":"0x00124b001ada3420","nwkAddr":45179,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,12,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{},"binds":[]},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-007","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2"}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1593167547132}`

Not sure where to go from here, but am happy to help test a fix

## What did you expect to happen
Device was expected to be supported

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: Tried both 1.14.0 & 1.14.0-dev - hassio plug ins. Lots above from dev branch.
Adapter hardware: CC2531
Adapter firmware version: {"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}

nanobra1n, 2020-06-27:
As it happens I paired just that device today and got this:
```
warn  2020-06-26 14:43:18: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:18: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-06-26 14:43:19: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:19: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-06-26 14:43:19: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:19: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-06-26 14:43:19: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:19: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-06-26 14:43:19: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:19: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-06-26 14:43:19: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:19: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-06-26 14:43:19: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:19: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-06-26 14:43:19: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:19: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-06-26 14:43:19: Received message from unsupported device with Zigbee model 'GL-C-007'
warn  2020-06-26 14:43:19: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2020-06-26 14:43:19: Successfully interviewed '0x00124b001f7ab6a4', device has successfully been paired
info  2020-06-26 14:43:19: Device '0x00124b001f7ab6a4' is supported, identified as: Gledopto Zigbee LED controller RGBW (1 ID) (GL-C-007-1ID)
info  2020-06-26 14:43:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001f7ab6a4","model":"GL-C-007-1ID","vendor":"Gledopto","description":"Zigbee LED controller RGBW (1 ID)","supported":true}}'
info  2020-06-26 14:43:19: MQTT publish: topic 'homeassistant/light/0x00124b001f7ab6a4/light/config', payload '{"brightness":true,"color_temp":true,"xy":true,"schema":"json","command_topic":"zigbee2mqtt/0x00124b001f7ab6a4/set","brightness_scale":254,"state_topic":"zigbee2mqtt/0x00124b001f7ab6a4","json_attributes_topic":"zigbee2mqtt/0x00124b001f7ab6a4","name":"0x00124b001f7ab6a4_light","unique_id":"0x00124b001f7ab6a4_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001f7ab6a4"],"name":"0x00124b001f7ab6a4","sw_version":"Zigbee2mqtt 1.14.0","model":"Zigbee LED controller RGBW (1 ID) (GL-C-007-1ID)","manufacturer":"Gledopto"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-06-26 14:43:19: MQTT publish: topic 'homeassistant/sensor/0x00124b001f7ab6a4/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00124b001f7ab6a4","json_attributes_topic":"zigbee2mqtt/0x00124b001f7ab6a4","name":"0x00124b001f7ab6a4_linkquality","unique_id":"0x00124b001f7ab6a4_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001f7ab6a4"],"name":"0x00124b001f7ab6a4","sw_version":"Zigbee2mqtt 1.14.0","model":"Zigbee LED controller RGBW (1 ID) (GL-C-007-1ID)","manufacturer":"Gledopto"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```
So despite all the warnings it paired fine. Trey controlling the device and if it works, don't worry about it.

matstrange, 2020-06-27:
So I tried again but still no joy. It seemed to take ~6 mins to complete pairing, but then the interview successful payload also contains _supported: false_ but in your example it shows as _true_. Thus it does then not annouce itself to HA.

```
zigbee2mqtt:info  2020-06-27 09:47:56: Device '0x00124b001ada3420' joined
zigbee2mqtt:info  2020-06-27 09:47:57: Starting interview of '0x00124b001ada3420'
zigbee2mqtt:info  2020-06-27 09:47:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001ada3420"}}'
zigbee2mqtt:info  2020-06-27 09:47:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001ada3420"}}'
zigbee2mqtt:info  2020-06-27 09:47:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ada3420"}}'
zigbee2mqtt:info  2020-06-27 09:47:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ada3420"}}'
zigbee2mqtt:warn  2020-06-27 09:53:00: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:00: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-27 09:53:00: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:00: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-27 09:53:00: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:00: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-27 09:53:00: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:00: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-27 09:53:00: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:00: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-27 09:53:00: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:00: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-27 09:53:00: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:00: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-27 09:53:01: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:01: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-27 09:53:01: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-27 09:53:01: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:info  2020-06-27 09:53:01: Successfully interviewed '0x00124b001ada3420', device has successfully been paired
zigbee2mqtt:warn  2020-06-27 09:53:01: Device '0x00124b001ada3420' with Zigbee model 'GL-C-007' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-06-27 09:53:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001ada3420","supported":false}}'
```

When I tried to manually issue a command such as (I think the syntax is correct)
[image]

I get this:
```
zigbee2mqtt:error 2020-06-27 10:04:46: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'white' is given but device is unsupported
    at Zigbee.resolveEntity (/app/lib/zigbee.js:199:25)
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:59:44)
    at Controller.callExtensionMethod (/app/lib/controller.js:356:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:250:14)
    at MQTT.emit (events.js:315:20)
    at MQTT.onMessage (/app/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:315:20)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1269:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:409:12)
    at work (/app/node_modules/mqtt/lib/client.js:320:12))
```

Koenkk, 2020-06-27:
@nanobra1n can you also share your `database.db` entry? And is yours a RGB + White or RGB + color temperature controller?

nanobra1n, 2020-06-27:
```
{"id":26,"type":"Router","ieeeAddr":"0x00124b001f7ab6a4","nwkAddr":53644,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-007","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0"}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1593196999675}
```
It's [this](https://www.amazon.com/gp/product/B07X8V8TBR) RGBW model.

Koenkk, 2020-06-28:
@nanobra1n yours should already be supported in the latest dev branch.

@matstrange I've added support for your device, should be supported in the latest dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

matstrange, 2020-06-29:
Thanks for your help on this so far, I upgraded to the latest dev branch and here is what I found.

The device now pairs eventually and shows as supported in the db:

```
zigbee2mqtt:info  2020-06-29 12:24:48: Device '0x00124b001ada3420' joined
zigbee2mqtt:info  2020-06-29 12:24:48: Starting interview of '0x00124b001ada3420'
zigbee2mqtt:info  2020-06-29 12:24:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001ada3420"}}'
zigbee2mqtt:info  2020-06-29 12:24:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001ada3420"}}'
zigbee2mqtt:info  2020-06-29 12:24:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ada3420"}}'
zigbee2mqtt:info  2020-06-29 12:24:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ada3420"}}'
zigbee2mqtt:warn  2020-06-29 12:28:31: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:31: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-29 12:28:31: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:31: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-29 12:28:31: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:31: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-29 12:28:32: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-29 12:28:32: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-29 12:28:32: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-29 12:28:32: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-29 12:28:32: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-06-29 12:28:32: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-06-29 12:28:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:info  2020-06-29 12:28:32: Successfully interviewed '0x00124b001ada3420', device has successfully been paired
zigbee2mqtt:info  2020-06-29 12:28:32: Device '0x00124b001ada3420' is supported, identified as: Gledopto Zigbee LED controller RGBW (2 ID) (GL-C-007-2ID)
zigbee2mqtt:info  2020-06-29 12:28:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001ada3420","model":"GL-C-007-2ID","vendor":"Gledopto","description":"Zigbee LED controller RGBW (2 ID)","supported":true}}'
```

However I am still having issues controlling it. The device does not seem to announce to HA (well it did once but wouldn't send commands back, and after removing and repairing multiple times it hasn't shown up again) 
When I fire commands via mqtt I have success for the first 1 or 2 brightness changes, then the lights go off and it stops responding until I pull the plug. 

```
zigbee2mqtt:error 2020-06-29 14:38:49: Publish 'set' 'brightness' to '0x00124b001ada3420' failed: 'Error: Command 0x00124b001ada3420/11 genLevelCtrl.moveToLevelWithOnOff({"level":25,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 45179 - 11 - 190 - 8 - 11 after 10000ms)'
zigbee2mqtt:info  2020-06-29 14:38:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to '0x00124b001ada3420' failed: 'Error: Command 0x00124b001ada3420/11 genLevelCtrl.moveToLevelWithOnOff({\"level\":25,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 45179 - 11 - 190 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"0x00124b001ada3420"}}'
```

Koenkk, 2020-06-29:
Can you try changing the definition in `devices.js` to below and see if that fixes the issue?

```js
{
    fingerprint: [
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
            {ID: 15, profileID: 49246, deviceID: 256, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
        ]},
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GLEDOPTO', endpoints: [
            {ID: 10, profileID: 49246, deviceID: 256, inputClusters: [0, 3, 4, 5, 6, 8], outputClusters: []},
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
        ]},
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 12, profileID: 260, deviceID: 258, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
        ]},
    ],
    model: 'GL-C-007-2ID', // 2 ID controls white and color separate
    vendor: 'Gledopto',
    description: 'Zigbee LED controller RGBW (2 ID)',
    meta: {disableDefaultResponse: true},
    extend: gledopto.light_onoff_brightness_colortemp_colorxy,
    endpoint: (device) => {
        if (device.getEndpoint(10) && device.getEndpoint(11) && device.getEndpoint(13)) {
            return {rgb: 11, white: 10};
        } else if (device.getEndpoint(11) && device.getEndpoint(12) && device.getEndpoint(13)) {
            return {rgb: 11, white: 12};
        } else {
            return {rgb: 11, white: 15};
        }
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

matstrange, 2020-06-29:
Thanks, that certainly got it to stop erroring - can new see in HA and control (kinda)

The colours seem confused (see video below). Its wired as it should be but R/G seem to render as opposite. Not a big deal, I could re-wire and swap but hopefully something easier to fix in code? 

Finally, you can see that the 'white' entity actually controls the colours, not the whites - the colour temp in the rgb entity seems to change the white, but discards the colour settings (perhaps this is a quirk of HA though). Even in node red I'm struggling to get a colour AND the WW leds on at the same time. 

Video link: https://we.tl/t-lNVFVgmHVV

Koenkk, 2020-06-29:
> The colours seem confused (see video below). Its wired as it should be but R/G seem to render as opposite. Not a big deal, I could re-wire and swap but hopefully something easier to fix in code?

This cannot be fixed from zigbee2mqtt, zigbee2mqtt has no knowledge about the channels, it just send the color in x/y values and the controller is possible for controlling the channels. So either a bug in the software in the controller or the markings are just wrong. To confirm this you can try pairing it to a hue bridge (in case you own one) and see if it shows the same issue. 

Regarding the swapped rgb/white, should be fixed with the following definition:

```js
{
    fingerprint: [
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
            {ID: 15, profileID: 49246, deviceID: 256, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
        ]},
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GLEDOPTO', endpoints: [
            {ID: 10, profileID: 49246, deviceID: 256, inputClusters: [0, 3, 4, 5, 6, 8], outputClusters: []},
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
        ]},
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 12, profileID: 260, deviceID: 258, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
        ]},
    ],
    model: 'GL-C-007-2ID', // 2 ID controls white and color separate
    vendor: 'Gledopto',
    description: 'Zigbee LED controller RGBW (2 ID)',
    meta: {disableDefaultResponse: true},
    extend: gledopto.light_onoff_brightness_colortemp_colorxy,
    endpoint: (device) => {
        if (device.getEndpoint(10) && device.getEndpoint(11) && device.getEndpoint(13)) {
            return {rgb: 11, white: 10};
        } else if (device.getEndpoint(11) && device.getEndpoint(12) && device.getEndpoint(13)) {
            return {white: 11, rgb: 12};
        } else {
            return {rgb: 11, white: 15};
        }
    },
},
```

matstrange, 2020-06-29:
I swapped the wires since I didn't have an alternative hub to test with.

It seems like perhaps the controller cant handle the white + rgb doing seperate things - both chips only illuminate in cold white mode - any other colour over-rides the white. Not a big deal since it will handle the scenes I have in mind.

I really appreciate your help with this and the great project in general!

Koenkk, 2020-06-30:
@matstrange so if I understand correctly the separate control doesn't work?

I've fixed a similar issue (#3828) and maybe it will also fix this. Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

matstrange, 2020-07-01:
Sadly, that didnt help. I upgraded the dev version and re-edited devices.js

Now I am seeing this error when I try to publish a 'white' change:

`zigbee2mqtt:error 2020-07-01 17:48:19: Publish 'set' 'brightness' to '0x00124b001ada3420' failed: 'Error: Command 0x00124b001ada3420/11 genLevelCtrl.moveToLevelWithOnOff({"level":236,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 45179 - 11 - 52 - 8 - 11 after 10000ms)'`

In addition the device is now much slower to respond in general to changes in the period after I have submitted the white request.

Koenkk, 2020-07-03:
To be sure we didn't mess up rgb and white, can you first turn off both and then turn on the brightness off the white entity and visa versa for the RGB, do the correct leds light up?

matstrange, 2020-07-05:
Still no difference sadly, if anything it's less reliable now - I can't even get HA to acknowledge when rgb is turned on (and thus no colour picker / temperature changes are possible)

I removed the device from HA, and the entries from the db and json files to give it a clean break but it didn't make a differece. Only the rgb leds light up with both entities.

`zigbee2mqtt:error 2020-07-05 08:01:10: Publish 'set' 'state' to '0x00124b001ada3420' failed: 'Error: Command 0x00124b001ada3420/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 36355 - 11 - 29 - 6 - 11 after 10000ms)'`

`zigbee2mqtt:error 2020-07-05 08:09:08: Publish 'set' 'brightness' to '0x00124b001ada3420' failed: 'Error: Command 0x00124b001ada3420/11 genLevelCtrl.moveToLevelWithOnOff({"level":202,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 36355 - 11 - 42 - 8 - 11 after 10000ms)'`

Eventually some 'positive' messages, but still only the rgb is changing:

```
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420', payload '{"state_white":"ON","brightness_white":157}'
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/rgb', payload '{}'
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/white', payload '{"state":"ON","brightness":157}'
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420', payload '{"state_white":"ON","brightness_white":254}'
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/rgb', payload '{}'
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/white', payload '{"state":"ON","brightness":254}'
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420', payload '{"state_white":"ON","brightness_white":102}'
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/rgb', payload '{}'
zigbee2mqtt:info  2020-07-05 08:10:19: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/white', payload '{"state":"ON","brightness":102}'
```

Koenkk, 2020-07-06:
Something strange is going on, as can be seen in the log `disableDefaultResponse` is not `true` anymore (while it should be like in https://github.com/Koenkk/zigbee2mqtt/issues/3813#issuecomment-651240627). This causes the big delays between commands. Are you sure you are using the exact definition from: https://github.com/Koenkk/zigbee2mqtt/issues/3813#issuecomment-651263115 ?

matstrange, 2020-07-21:
Sorry for the **very** delayed response, I've just not had time to sit and play with it.

I've recreated the settings and its a bit happier - no lag on changing colours or related errors. 

The issue still remains where I cannot independantly control rgb & white. What ever I do to one overrides the other. Only the colour temperature control in the rgb option seems to accurately manage the W leds. : https://we.tl/t-QrBK8eye0E

Snip of log from turning the white command on and off:
 
Turn white off

```
zigbee2mqtt:info  2020-07-21 17:18:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420', payload '{"state_rgb":"OFF","brightness_rgb":0,"state_white":"OFF","brightness_white":0,"color_rgb":{"x":0.175,"y":0.417},"color_temp_rgb":82}'
zigbee2mqtt:info  2020-07-21 17:18:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/rgb', payload '{"state":"OFF","brightness":0,"color":{"x":0.175,"y":0.417},"color_temp":82}'
zigbee2mqtt:info  2020-07-21 17:18:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/white', payload '{"state":"OFF","brightness":0}'
```

Turn white on

```
zigbee2mqtt:info  2020-07-21 17:18:50: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420', payload '{"state_rgb":"OFF","brightness_rgb":0,"state_white":"ON","brightness_white":254,"color_rgb":{"x":0.175,"y":0.417},"color_temp_rgb":82}'
zigbee2mqtt:info  2020-07-21 17:18:50: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/rgb', payload '{"state":"OFF","brightness":0,"color":{"x":0.175,"y":0.417},"color_temp":82}'
zigbee2mqtt:info  2020-07-21 17:18:50: MQTT publish: topic 'zigbee2mqtt/0x00124b001ada3420/white', payload '{"state":"ON","brightness":254}'
```

Koenkk, 2020-07-21:
Do things improve with:

```js
{
    fingerprint: [
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
            {ID: 15, profileID: 49246, deviceID: 256, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
        ]},
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GLEDOPTO', endpoints: [
            {ID: 10, profileID: 49246, deviceID: 256, inputClusters: [0, 3, 4, 5, 6, 8], outputClusters: []},
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
        ]},
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 12, profileID: 260, deviceID: 258, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
        ]},
    ],
    model: 'GL-C-007-2ID', // 2 ID controls white and color separate
    vendor: 'Gledopto',
    description: 'Zigbee LED controller RGBW (2 ID)',
    meta: {disableDefaultResponse: true},
    extend: gledopto.light_onoff_brightness_colortemp_colorxy,
    endpoint: (device) => {
        if (device.getEndpoint(10) && device.getEndpoint(11) && device.getEndpoint(13)) {
            return {rgb: 11, white: 10};
        } else if (device.getEndpoint(11) && device.getEndpoint(12) && device.getEndpoint(13)) {
            return {rgb: 11, white: 12};
        } else {
            return {rgb: 11, white: 15};
        }
    },
},
```

matstrange, 2020-07-22:
No, I'm afraid that still behaves the same. 

Since so long elapsed since I last had time to look at it, I even wiped all changes and went through it again incase I missed/forgot something.

Koenkk, 2020-07-23:
I'm not sure how to continue with the separate control as this used to work for all other geldopto 2-id controllers. It could also be due to your device having old software? (`"dateCode":"20131206","swBuildId":"1.0.2"`, unfortunately not upgradable as gledopto doesn't provides updates). To make sure you can try testing how it behaves with a Hue bridge (in case you have) as Gledopto advertises full compatibility with this hub.

Anyway we can also mark it as a `GL-C-007-1ID`, for this undo your current changes and change the entry of `GL-C-007-1ID` to 

```js
{
    fingerprint: [
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 528, inputClusters: [4096], outputClusters: [4096]},
        ]},
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 12, profileID: 260, deviceID: 258, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
        ]},
    ],
    model: 'GL-C-007-1ID', // 1 ID controls white and color together
    vendor: 'Gledopto',
    description: 'Zigbee LED controller RGBW (1 ID)',
    extend: gledopto.light_onoff_brightness_colortemp_colorxy,
},
```

k3erg, 2020-08-29:
Maybe a regression?

I have two GL-C-008 which where working for over a year. I just updated zigbee2mqtt (master) two days ago, and now I get a "Device with modelID 'GLEDOPTO' is not supported" for both, and I can't use them no longer.

Will try to remove and reconnect them later.

Koenkk, 2020-08-30:
@k3erg if this is still the case in the latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) please provide the `data/database.db` entry of this device and if this is a RGBW or RGB + CCT device.

k3erg, 2020-08-30:
@Koenkk Txs. I switched to dev. Now the devices are properly recognised ("Device '0x00124b0019xxxxxx' is supported, identified as: Gledopto Zigbee LED controller RGB + CCT (1 ID) (GL-C-008-1ID)). Just got to fiddle with new hue and transitions now :).

Koenkk, 2020-08-31:
@k3erg could you provide the herdsman debug logging of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

matstrange, 2020-09-18:
> I'm not sure how to continue with the separate control as this used to work for all other geldopto 2-id controllers. It could also be due to your device having old software? [...]

I have updated to use it like this (one control instead of two) and it suits me fine. Thanks for your help and patience!

Koenkk, 2020-09-18:
Added it, will be supported in next release.

closed 2020-09-18 by Koenkk

matstrange, 2020-09-18:
Sorry, in my haste I forgot to mention that I had to add in

` meta: {disableDefaultResponse: true},`

It was slow/unstable without it

Koenkk, 2020-09-19:
Done!