Koenkk/zigbee2mqtt #3818: MiJia Honeywell smoke detector

ardenking, 2020-06-27:
Hi All 
new install so latest firmware and software 
installed using site instructions  opnhabian pi 

Have added a few devices working fine except MiJia Honeywell smoke detector
can see the status of the device but cant self test  mosquitto_pub -t zigbee2mqtt/smoke_alarm/set -m  '{"selftest": ""}'
gives this message in the logs 

{"type":"zigbee_publish_error","message":"Publish 'set' 'selftest' to 'smoke_alarm' failed: 'Error: Write 0x00158d00052bad9d/1 ssIasZone({\"65521\":{\"value\":50397184,\"type\":35}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 45941 - 1 - 80 - 1280 - 4 after 35000ms)'","meta":{"friendly_name":"smoke_alarm"}}

anyhelp  would be appreciated

ardenking, 2020-06-27:
ok after setting the sensitivity the self test command appears to be working but there is a delay of about 5 mins between issuing the payload and the beep

McGiverGim, 2020-07-01:
I've suffered a similar problem. After restarting zigbee2mqtt it works better and self test works, but it seems not reliable in time neither.

closed 2020-08-08 by stale[bot]