Koenkk/zigbee2mqtt #3819: Very bad signal zig-a-zig-ah! (CC26X2R1)

giejay, 2020-06-27:
# Bug Report
## What happened
Just got my zig-a-zig-ah usb stick from Electrolama. I was expecting a signal boost comparing it to my cc2531, which never had a good signal to begin with. Unfortunately, the linkquality of for example my Xiaomi Temp Sensor, or my Opple Switch, decreases quite a lot. 3 meters and 1 wall, the linkquality is about 55. When I move 1 sensor upstairs, about 6 meters and 2 walls, the linkquality is 0 and the device very unresponsive. 

Because the shipped antenna is quite small, I ordered this one: https://www.aliexpress.com/item/32913680322.html?spm=a2g0s.9042311.0.0.6d1f4c4dAHw4jQ

Would that help in my setup? Or is there some other issue I'm overlooking? 

My Wifi is setup on channel 6 and 12. The zigbee coordinator is on channel 11 (that's the default right?), which I think overlaps with channel 1 of WiFi. So that should not be a problem.  

I was using the Xiaomi Bridge, that has a very nice range so I was hoping to at least equal that. 

I have tried both plugging it directly in the raspberry pi, or using an usb extender cable, didn't really matter signal wise. 

Btw, I'm not looking to buy all kinds of devices which should act as a router, had some bad experiences as well with devices just ignoring those routers and seeing as my Xiaomi Bridge is working fine without them, I'd like the CC26X2R1 to work without them as well. 

## What did you expect to happen
Having a link quality for devices two floors up of at least 20. Having a link quality of 90 for devices 3 meters away from the coordinator.

## How to reproduce it (minimal and precise)
Pair devices, (temp sensors, buttons), move up 1 floor, link quality drops to 0. 

## Debug Info
Zigbee2mqtt version: 1.14
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200417

omerk, 2020-06-27:
Hi @giejay,

zzh (CC2652R) does not have a PA/LNA (something that you would get with CC1352**P**, CC2652**P** or CC2530/1/8 paired with CC2592) so if you had signal issues before it is likely that you will see similar behaviour with zzh as well. Getting a stronger antenna and mounting zzh somewhere a bit more central using a USB extension lead (which you've tried already) might help but realistically speaking adding routers is always the best way to extend range in a Zigbee network in my experience.

Having said that, `3 meters and 1 wall, the linkquality is about 55` certainly doesn't sound right. Is this with just 1 or two devices or are you seeing similar behaviour with others as well? Was this any better or comparable when you were using your CC2531 stick?

Side note: Not sure if @Koenkk is happy with zzh issues opened in z2m repo? I don't monitor z2m issues so if you (and readers) have issues you will get a quicker response emailing support@electrolama.com or opening an issue on the [zzh repo](https://github.com/electrolama/zig-a-zig-ah).

giejay, 2020-06-27:
Hi @omerk 

Im seeing the same behaviour for all devices that I coupled, two temperature sensors, 1 opple switch and 1 motion sensor. I think this was comparable to the CC2531. When I remove the antenna, it get's even worse so that one does improve it a bit. 

I hope the new external antenna will give me some better results, otherwise I would have to stick with the Xiaomi Bridge (which I also don't like because the newer versions are quite hard to "unlock" and make compatible with OpenHab)

ArkadiuszNiemiec, 2020-06-27:
I also have just received my ZZH. My experience so far is pretty good. 
One thing I had to change was pairing Xiaomi sensors close to coordinator (max. few meters) and I believe that on CC2531 I could pair it 2 walls and 10 meters away (maybe they used routers before and now they can't?).
The sensor that is 2 walls and 10 meters away reports link quality in the range of 90-96.

giejay, 2020-06-28:
> I also have just received my ZZH. My experience so far is pretty good.
> One thing I had to change was pairing Xiaomi sensors close to coordinator (max. few meters) and I believe that on CC2531 I could pair it 2 walls and 10 meters away (maybe they used routers before and now they can't?).
> 
> The sensor that is 2 walls and 10 meters away reports link quality in the range of 90-96. Edit: that probably means, that there is a router in-between, right?

Can you draw the network map to check if that's the case? 90 is very good, I can imagine there's a router helping... What kind of routers do you have?

ArkadiuszNiemiec, 2020-06-28:
@giejay I can but Xiaomi Sensors are never connected. Do you know how to check their routing?

[image]

giejay, 2020-06-28:
Thanks for sharing. Quite odd that they are not connected. You retrieved the network map by publishing a message to the zigbee2mqtt topic?

What type of product is the bedroom_wall_switch? It seems to work as router so maybe I will get that if the antenna won't improve the signal.

ArkadiuszNiemiec, 2020-06-28:
@giejay I have coupled https://www.zigbee2mqtt.io/devices/QBKG12LM.html and one Gledopto Zigbee LED controller. From the sniffer I know that the sensor is connecting via Gledopto, which is almost as far as the coordinator itself. There are at least 2 Aqara Switches in-between that are skipped.

Regarding the network: as far as I know I cannot get the full map because sensors are sleeping during the network scan.

ArkadiuszNiemiec, 2020-06-28:
The `livingroom_sensor` is 4 meters from the coordinator in LOS. In logs It shows "linkquality" to be around 80, in Z2M Assistant it shows: `Link quality is 43/255.`.

giejay, 2020-06-28:
Weird that there's a big difference between your home assistant and the logs... I'm using OpenHab, will create an item for the link quality so I can create a nice graph in Grafana. 

Regarding the Aqara switches, unfortunately those won't fit my EU socket. I read that Aqara is working on an EU version so hopefully that will be released soon.

ArkadiuszNiemiec, 2020-06-28:
@giejay I swapped my EU sockets to Chinese in my new flat 😄 You can try to add one or two Ikea Tradfri Repeaters, they work out of the box.

giejay, 2020-06-28:
Very good tip, they even have a USB connection so I can use them to charge for example my Arlo Babycam:) Will get one next time I visit ikea:)

kennymc-c, 2020-07-02:
I also noticed that none of my devices report a linkquality higher than 60 even with a 13dbi antenna connected and some of them are in the same room as the coordinator.

Just received this mail from Electrolama/Tindie and I think my unit could also be affected:

> Hello,
> 
> I am investigating what appears to be a manufacturing issue with some of the zzh boards shipped in the last batch (sold on Tindie 20/06/2020) with users reporting bad RF performance.
> 
> Enquiries have been made with the manufacturing company, the component vendor and the component manufacturer and once I hear from all parties involved I will send out another update with the next steps.
> 
> Please bear with me while we get to the bottom of this. It certainly is frustrating but we will fix this one way or another, as soon as humanly possible.
> 
> You can hit "reply" if you want to get in touch and as always, thanks for your continued support!
> 
> Omer @ Electrolama

giejay, 2020-07-02:
Thanks for sharing @kennymc-c. That's the same batch as mine, so it might be related.

omerk, 2020-07-02:
@kennymc-c @giejay It is likely you are affected by this. Please keep an eye on your emails in the next few days as I will be sending an update.

giejay, 2020-07-02:
> @kennymc-c @giejay It is likely you are affected by this. Please keep an eye on your emails in the next few days as I will be sending an update.
Just checked my mail and also got it. Thanks for investigating!

alekslyse, 2020-07-03:
Would it be pointless to pair devices to the dongle as of now? (I have same batch). If it needs to be replaced is it possible to just change the dongle without re-pair?

apmosss, 2020-07-10:
Any news on that? I am also affected (same batch)

giejay, 2020-07-10:
> Would it be pointless to pair devices to the dongle as of now? (I have same batch). If it needs to be replaced is it possible to just change the dongle without re-pair?

I would hold off pairing for now. I'm also just waiting for a fix.

> Any news on that? I am also affected (same batch)

Latest news:
Hello again,

Just a quick update on the zzh signal quality problems: I have narrowed down the problem to a particular component on the board and also the batch it was used in. Just to stress that not all boards sold on 20/06 have this issue.

With help from one of the community members who has access to the necessary RF kit in their lab, I am putting together a detailed account of this issue and will be publishing this, along with the next steps in getting you replacement parts and boards within the next few days. This article will contain all the details you need to identify boards that are affected.

Electrolama is very much a weekends/evenings kind of a project so please give me a few more days to get things sorted, I will provide you with an update on or before Friday (10th). I can assure you that I will replace all affected boards.

Thank you for your patience.

JLFN, 2020-07-10:
Hi when are you selling zzh-p version?

JLFN, 2020-07-10:
@omerk 
Hi when are you selling zzh-p version?

kennymc-c, 2020-07-15:
Did anyone receive a confirmation email after submitting the form from the mail?

apmosss, 2020-07-15:
No. There was no option for email in the google form. I hope somehow Omer will find the email from the order id and notify us.

kennymc-c, 2020-07-15:
At the top there is a field for a email address, which is also a required field for submitting the form.

omerk, 2020-07-16:
@alekslyse Moving from one zzh to another, running the same firmware, is absolutely fine and you would not need to re-pair or change any settings with your existing network if you swap boards.

@JLFN No ETA unfortunately. Soon.

@kennymc-c and others: No immediate feedback after filling the form is normal, collecting all input in a Sheet and will process when replacement boards are available. In the meantime, email support@ if you have any queries.

@apmosss @kennymc-c Forgot the email field in the first version of the form so if you've opened it straight after I sent the email then it wouldn't have been there. It is, now.

kennymc-c, 2020-07-28:
Could the ELIFECYCLE antenna startup problem (https://github.com/Koenkk/zigbee2mqtt/issues/3664) also be caused by the faulty component or was this already reported in previous batches? The error usually occurs when I restart my server, but not always.

omerk, 2020-07-28:
@kennymc-c There is no "antenna startup problem" and removing the antenna before starting does not "minimize interference" as that linked issue suggests. Removing the antenna simply results in your coordinator having worse signal tx/rx capability (as there is no antenna plugged in) and as a result mask any underlying problems you might have in your network. So it's perhaps a workaround to another issue but not something that is inherently a problem by itself. FL1 issue reported causes RF performance degradation in the affected units.

Every time I've seen `panID is already in use`, it was because I had another coordinator running that I've forgotten about. Not sure what other causes there might be.

sjorge, 2020-07-28:
> Every time I've seen `panID is already in use`, it was because I had another coordinator running that I've forgotten about. Not sure what other causes there might be.

I’ve seen this too when having the network in permit_join = true when unplugging the coordinator. Other routers might then respond if the coordinator comes back and it will back off.

giejay, 2020-08-28:
So I received my new zigazigah! I notice a minor increase in signal quality but the main thing that I'm really pleased with is that pairing is now a breeze. 

I paired it with my Ikea Tradri Repeater, which was impossible with the old stick, that one is one floor up, has a quality of 50, all devices another floor up are connected to the repeater with a link quality of > 100. So, I can now finally get rid off my Xiaomi Bridge and only use local connections.

kisseler, 2020-09-15:
I experienced a decrease (almost total failure) of network range after switch to zzh coming from CC1352P-2 (see [my post in zzh-issues](https://github.com/electrolama/zig-a-zig-ah/issues/17)). I'm currently in contact with the developer (astonishing good support!) to exclude hardware issues.

For now I switched back to CC1352P-2, so my network stabilized again. But in general I'm facing quite bad network quality und unresponsive devices all over the house. I built up a few CC2530 and CC2530-CC2591 routers, but reliability is still low. I also tried replacing the antenna and using a usb extension cable.

- Has anyone had similar problems and was able to solve them?

- How important is it to try different channels?

- What about building stock influences? We're living in an old timber framing house. But that shouldn't shield the signal from very next room, right?

Thanks for sharing your experiences! I very keen on [improving reliability](https://github.com/Koenkk/zigbee2mqtt/issues/2705#issuecomment-691515791) to reduce frustration and increase overall acceptance of smart home/zigbee/zigbee2mqtt/mqtt/openhab systems.

yoution, 2020-09-19:
I also have the signal problem with cc2531.I use cc2531 as coordinator with a antenna
when put my motion sensor with cc2531 together, the linkquality is about 140
when the distance is about 10cm, the linkquality is about 100
when the distance is about 3m, the linkquality is only 60
when the distance is about 5m, the linkquality is almost 0,
when remove the antenna and put them almost together, the linkquality is only 50
the problem puzzle me a lot of time, I also buy a 1m use extend cable, but it didn't work

magicdude4eva, 2020-10-16:
I just got a zzh and can also report poor signal-quality and unable to pair. ConbeeII has superior connection.

peterjseymour, 2020-10-20:
I just received my zzh and have been having no end of trouble with signal quality. I'm using a raspberry pi 4b rev 1.2. 
I find that with the included antenna and plugged straight into the USB port, I was getting no signal whatsoever. Strangely when I removed the external antenna I got a better connection. I also get a better connection by using an antenna with a long cable, and also when using a USB extension cable (a 10cm cable doesn't work but a 1m cable does).

magicdude4eva, 2020-10-20:
FWIW: No feedback from seller. I have reverted back to Conbee2 (same setup via Zigbee2mqtt) and have had zero issues. I could not pair any device (even new ones or factory reset ones) with the ZZH unless I was literally next to the device. With the Conbee2 and a 3m extension cable (placed in the basement now) I reach devices on the 1st floor. With the ZZH I could not reach a device on the groundfloor.

omerk, 2020-10-20:
@magicdude4eva As requested on https://github.com/electrolama/zig-a-zig-ah/issues/18, please send an email to support@electrolama.com so we can figure out a solution.

@peterjseymour That doesn't sound right, could you also drop me an email with your Tindie order number to support@electrolama.com as well please?

For future reference, as many people seem to find this issue with Google searches: **If you have a problem with your zzh, please send an email to support@electrolama.com**. I do not actively monitor Github issues, especially issues on the zigbee2mqtt project. 

Development happens on Github, customer support happens on support@electrolama.com. Thank you for your cooperation.

timrosede, 2020-11-05:
I received my zig-a-zig-ah a few days ago and today I reconnected my devices. I connected the Raspberry via Powerbank and walked around the house with it, the pairing worked out wonderfully - only an Ikea Tradfri Remote was a bit bitchy. 

But now I have problems reaching the lamps in the kitchen:

```
zigbee2mqtt | Zigbee2MQTT:info 2020-11-05 14:43:41: MQTT publish: topic 'zigbee2mqtt/0x588e81fffe44bce1', payload '{"brightness":254, "linkquality":135, "state": "ON", "update_available":true}'
    | Zigbee2MQTT:info 2020-11-05 14:42:55: MQTT publish: topic 'zigbee2mqtt/0xccccccfffebeb43d', payload '{"brightness":254, "linkquality":9, "state": "ON", "update_available":true}'
zigbee2mqtt | Zigbee2MQTT:info 2020-11-05 14:42:59: MQTT publish: topic 'zigbee2mqtt/0xccccccfffebeb43d', payload '{"brightness":254, "linkquality":0, "state": "OFF", "update_available":true}'
zigbee2mqtt | Zigbee2MQTT:info 2020-11-05 14:43:22: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe44ede1', payload '{"brightness":254, "linkquality":0, "state": "OFF", "update_available":false}'
zigbee2mqtt | Zigbee2MQTT:info 2020-11-05 14:43:26: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe44ede1', payload '{"brightness":254, "linkquality":3, "state": "ON", "update_available":false}'
zigbee2mqtt | Zigbee2MQTT:info 2020-11-05 14:43:33: MQTT publish: topic 'zigbee2mqtt/0x588e81fffe44bce1', payload '{"brightness":254, "linkquality":0, "state": "OFF", "update_available":true}' 
```

I had previously a CC2531 with an external Antenna(from aliexpress) and had absolutely no problems with reception.

closed 2020-12-13 by github-actions[bot]

omerk, 2021-01-18:
A lot of folks are referencing this issue when sending support requests, so just a quick note to say that the component issue referenced above does not affect any new zzh units sold. 

Range and signal issues are very difficult to diagnose, here are some good starting points for troubleshooting such issues: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html Adding to that list, there are increasing reports of SSDs used next to coordinator sticks (mainly on Raspberry Pi boards) causing power and/or interference problems, you might want to temporarily remove the SSD out of the equation when troubleshooting.