Koenkk/zigbee2mqtt #3820: Re-pairing problems after moving to zzh! CC2652R

kennymc-c, 2020-06-27:
I tried to replace my CC2531 coordinator yesterday with a zzh! CC2652R. After a few start problems (ELIFECYLE), which apparently can be caused by the antenna, I finally got Z2M running with the stick attached to a USB extension cable.
I was able to reset and repair the first devices without problems. They also appeared in Zigbee2MQTTAssistant. But partly with a question mark as symbol. But I didn't think anything of it yet, because according to the display, the devices still showed regular updated values. After some time I noticed that the MQTT messages did not contain any values like linkquality etc. After a restart these devices disappeared in the list. I could reset some of them and they were displayed correctly. With others this failed.
After some research, it seemed to be the database.db, which still contained all devices paired with the old CC2531. I then deleted the file and the state.json, stopped Z2M, removed the stick for a few seconds and then restarted. The first thing that happened was the ELIFECYLE error again, which disappeared after a few reboots and replugging the stick.
I had to repair all devices again. But now there are problems with even more devices. I also can't get my CC2530 router added. It even appeared twice in the database.db with both InterviewCompleted:false but a different NetworkAddress. I could remove one entry with force remove the device but the other is still in the database. Despite several power cycle reset attempts Z2M doesn't seem to recognize the router anymore. 
The remaining temperature sensors (3x Xiaomi Mijia) seem to react to the reset (3x flashing) but don't show up either.
I have attached a herdsman debug log where I tried to pair one of the Xiaomi sensors (DB_Sensor):

https://controlc.com/0b7845df

## Debug Info
Zigbee2mqtt version: 1.14.0 (docker)
Adapter hardware: CC2652R (zzh!)
Adapter firmware version: 20200417

kennymc-c, 2020-06-28:
I‘ve managed to pair the CC2530 router again today. It now regularly sends messages but only with its linkquality and led state. The online/offline state is still missing. In database.db the device is now listed with InterviewCompleted:true and more metadata, so also z2MA recognizes it completely. I noticed that the router is still twice in the database with the same ieeeAdress but the other has no metadata and interviewCompleted:false. Could that be the problem? Manually deleting the line doesn’t work until a restart. 
Still no luck with the Mijia sensors. Even when I try to pair it just a few cm from the coordinator.

Koenkk, 2020-06-29:
> The online/offline state is still missing 

This has been removed, as it represented the led state, not the online/offline state of the router.

For the Mija sensors, try with a different battery/removing the battery for a few seconds. Also make sure that your old network is completely down (so both coordinator and all routers)

kennymc-c, 2020-06-29:
Ok, I was just wondering, because according to my home automation server (Fhem) the CC2530 router still shows an additional online/offline state next to the led status, which is updated at Z2M startup. Since I did not update Z2M, the router FW or Fhem, I found it surprising that the state is no longer updated after several resets of the router for re-pairing. In state. json the CC2530 router has only a led and a linkquality entry.

I have all my Mijia sensors laying here for several days without batteries and only insert them when trying to pair them again. I also ordered some new batteries, if that should help. The LEDs are not as bright as at the beginning but still seem to response.
Of course the CC2531 is not in use anymore and all other devices are paired with the zzh stick. At first i was also able to pair all sensors. But with the Mijia there were problems with the pairing interview process. I suspect that they are still present somewhere, although they do not appear in the database.db.
Shouldn't you be able to see this in the debug log when a pairing attempt is rejected?

Koenkk, 2020-06-29:
@kennymc-c do you see anything in the log when trying to pair the device?

kennymc-c, 2020-06-29:
Nothing, even with debug logging enabled.

Koenkk, 2020-06-29:
In that case the device doesn't even attempt to join. I've seen this multiple times with my own MiJa sensors, after disconnecting the battery for a minute or with a new battery the device pairs fine.

kennymc-c, 2020-06-30:
It was definitely the batteries. When I put in new ones, the sensors were detected the first time the button was pressed. I assume that the voltage of the old batteries was not high enough for pairing despite the led feedback. But for regular sending the values it was probably still sufficient.

Koenkk, 2020-06-30:
Good that you've managed to fix this, can this issue be closed?

kennymc-c, 2020-06-30:
yes

closed 2020-06-30 by kennymc-c