Koenkk/zigbee2mqtt #3821: Tuya thermostatic valve

Romioyar, 2020-06-27:
# Bug Report
## What happened
New Tuya thermostatic valve was auto discovered in Zigbee2Mqtt 1.14.0 as Tuya curtain motor. Valve reports model as TS0601 which is occupied already in devices.js by Tuya curtain motor.
https://aliexpress.ru/item/4001043738901.html?spm=a2g0s.9042311.0.0.1b6033edHXNB4c&_ga=2.27185918.2138683156.1593276537-279917533.1590225196

```
info  2020-06-27 17:39:11: Device '0xbc33acfffe6d821c' joined
info  2020-06-27 17:39:11: Starting interview of '0xbc33acfffe6d821c'
info  2020-06-27 17:39:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xbc33acfffe6d821c"}}'
info  2020-06-27 17:39:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xbc33acfffe6d821c"}}'
debug 2020-06-27 17:39:11: Device '0xbc33acfffe6d821c' announced itself
info  2020-06-27 17:39:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xbc33acfffe6d821c"}}'
debug 2020-06-27 17:39:11: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0601"}' from endpoint 1 with groupID 0
info  2020-06-27 17:39:12: MQTT publish: topic 'homeassistant/cover/0xbc33acfffe6d821c/cover/config', payload '{"command_topic":"zigbee2mqtt/0xbc33acfffe6d821c/set","position_topic":"zigbee2mqtt/0xbc33acfffe6d821c","set_position_topic":"zigbee2mqtt/0xbc33acfffe6d821c/set","set_position_template":"{ \"position\": {{ position }} }","value_template":"{{ value_json.position }}","json_attributes_topic":"zigbee2mqtt/0xbc33acfffe6d821c","name":"0xbc33acfffe6d821c_cover","unique_id":"0xbc33acfffe6d821c_cover_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xbc33acfffe6d821c"],"name":"0xbc33acfffe6d821c","sw_version":"Zigbee2mqtt 1.14.0","model":"Curtain motor (TS0601)","manufacturer":"TuYa"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-06-27 17:39:12: MQTT publish: topic 'homeassistant/sensor/0xbc33acfffe6d821c/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0xbc33acfffe6d821c","json_attributes_topic":"zigbee2mqtt/0xbc33acfffe6d821c","name":"0xbc33acfffe6d821c_linkquality","unique_id":"0xbc33acfffe6d821c_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xbc33acfffe6d821c"],"name":"0xbc33acfffe6d821c","sw_version":"Zigbee2mqtt 1.14.0","model":"Curtain motor (TS0601)","manufacturer":"TuYa"},"availability_topic":"zigbee2mqtt/bridge/state"}'
debug 2020-06-27 17:39:12: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_ckud7u2l"}' from endpoint 1 with groupID 0
debug 2020-06-27 17:39:12: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2020-06-27 17:39:12: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2020-06-27 17:39:12: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{"appVersion":83}' from endpoint 1 with groupID 0
debug 2020-06-27 17:39:12: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2020-06-27 17:39:12: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2020-06-27 17:39:13: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2020-06-27 17:39:13: Received Zigbee message from '0xbc33acfffe6d821c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
info  2020-06-27 17:39:13: Successfully interviewed '0xbc33acfffe6d821c', device has successfully been paired
info  2020-06-27 17:39:13: Device '0xbc33acfffe6d821c' is supported, identified as: TuYa Curtain motor (TS0601)
```

I deleted TS0601 device in devices.js and repaired and got this:
```
warn  2020-06-27 19:05:57: Received message from unsupported device with Zigbee model 'TS0601'
warn  2020-06-27 19:05:57: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
Then I changed devices.js and added TS0601 as Tuya thermostatic valve and deleted curtain part, also fixed homeassistant.js for autodiscovery :

```
    {
        zigbeeModel: ['TS0601'],
        model: 'TS0601',
        vendor: 'Moes',
        description: 'Radiator valve with thermostat',
        supports: 'thermostat, temperature',
        fromZigbee: [
            fz.tuya_thermostat,
            fz.tuya_thermostat_on_set_data,
            fz.ignore_basic_report,
        ],
        toZigbee: [
            tz.tuya_thermostat_child_lock,
            tz.tuya_thermostat_window_detection,
            tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint,
            tz.tuya_thermostat_system_mode,
            tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration,
            tz.tuya_thermostat_min_temp,
            tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time,
            tz.tuya_thermostat_comfort_temp,
            tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force,
        ],

    },
```
And thermostatic valve works correctly after this.
I bought 4 of them so could you please fix so it is correctly recognized out of the box.

Thanks!

## What did you expect to happen

Valve autodiscovered as Tuya thermostatic valve which is actually similar to Moes.

## How to reproduce it (minimal and precise)
Start pairing thermostatic valve.
## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2538
Adapter firmware version: zStack30x

Koenkk, 2020-06-28:
Can you share the `data/database.db` entry of this device?

Romioyar, 2020-06-28:
Hi, sure here are entries for 4 different devices
```
{"id":48,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6d807b","nwkAddr":2412,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593341216187}
{"id":49,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6d821c","nwkAddr":40345,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593274610200}
{"id":50,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6557d3","nwkAddr":55948,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593342591143}
{"id":51,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6d8231","nwkAddr":51668,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593342645599}
```

Koenkk, 2020-06-28:
Thanks, asked for some help in https://github.com/Koenkk/zigbee2mqtt/issues/2778#issuecomment-650788198

fr34kyn01535, 2020-07-04:
Same issue here:

`{"id":5,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6eb997","nwkAddr":8048,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593875683098}
{"id":6,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6d7fef","nwkAddr":2904,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593875621151}
{"id":7,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6d80cf","nwkAddr":29801,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593876100210}
{"id":8,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6ec1cc","nwkAddr":38004,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593876848311}`

On a side note, this device seems to identical to HY369RT from Moes.

tux0n, 2020-07-11:
Thanks for solution. How did you fix auto Discovery?

Koenkk, 2020-07-12:
Added support for this device out-of-the-box now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

damianpasek, 2020-07-22:
@Koenkk It there any ETA when this will be in the master branch?

Koenkk, 2020-07-22:
@damianpasek it's in the just release zigbee2mqtt 1.14.2, please let me know if it works there.

Sthopeless, 2020-07-23:
@Koenkk they are now correctly added to Zigbee2MQTT but at least on Hassio with the addon they are not working properly.

on printscreen is what's added from it to my HA, `Valve detection` and `Window detection` don't seen to do anything
[image]

And here on the operation mode I can only set correctly `Auto` and whenever I try to set `Heat` or `Off` the device goes into `away mode` 
[image]

I'm currently using a CC2531 usb stick with 1.2 stack

Koenkk, 2020-07-23:
@Sthopeless good that it is discovered correctly now, I'm not sure if this has been implemented/susposed to work, hopefully others from this thread can help (I don't have the device).

cadavre, 2020-07-24:
Confirmed that valve was discovered correctly and integrated into HA with no (or maybe little?) problems.

Right after pairing it is discovered as "cover" (browsed via z2mqtthelper), but after restarting z2m – it shows valve correctly and maked HA integration proper.

I can confirm what @Sthopeless said – modes are not working best, but I guess it's more to do with converters than z2m itself.

Tontze, 2020-07-25:
https://www.moeshouse.com/collections/tuya-zigbee-switch/products/tuya-zigbee-push-light-switch-zigbee-hub-required-smart-home-automation-wireless-remote-control-work-with-tuya-zigbee-hub-only

This switch is also reported as TS0601

galzil, 2020-07-27:
Same for a [single light switch](https://www.zigbee2mqtt.io/devices/GDKES-01TZXD.html) now that was identified as: TuYa Curtain motor (TS0601)

dedors, 2020-07-29:
I got the same device from aliexpress. Mine has a Moes branded box with model number HY368. Maybe add that on the wiki page too?

I had to troubles joining the network. Joining the network is explained to work after "short press home (turn on), long press home (enter settings), go to setting 5 (wifi logo), press home (only wifi now showing), long press home (wifi now blinking). I then pressed home every few seconds as it seems to timeout pretty fast. But it never joined the network.

Unscrew/rescrew the battery cover suddenly did the trick, it just joined, even I tried that a few times before.

Tontze, 2020-07-30:
this issue is for thermostat, for switch go to https://github.com/Koenkk/zigbee2mqtt/issues/3989

RaveGun, 2020-08-03:
Hi,

I also got the two MOES and installed them a few days apart.
Both where after the pairing detected as Covers at the beginning and a Z2M restart updated the configuration and I had to delete the cover RETAINED topic. 

The one issue I have is that the devices an "AUTO", a "MANUAL" and a "HOLIDAY" mode and if I setup the temperature manually it reports the system mode as "manual" and this generates a log into the HA. 

I guess this also have something to do with the available modes in the HA for the Climate/MQTT component.

The following configuration mode is sent:
`homeassistant/climate/0x842e14fffef6d77c/climate/config`
```
{
    "min_temp": "5",
    "max_temp": "30",
    "modes": [
        "off",
        "auto",
        "heat"
    ],
    "mode_state_topic": "zigbee2mqtt/0x842e14fffef15510",
    "mode_state_template": "{{ value_json.system_mode }}",
    "mode_command_topic": "zigbee2mqtt/0x842e14fffef15510/set/system_mode",
    "current_temperature_topic": "zigbee2mqtt/0x842e14fffef15510",
    "current_temperature_template": "{{ value_json.local_temperature }}",
    "temperature_state_topic": "zigbee2mqtt/0x842e14fffef15510",
    "temperature_state_template": "{{ value_json.current_heating_setpoint }}",
    "temperature_command_topic": "zigbee2mqtt/0x842e14fffef15510/set/current_heating_setpoint",
    "temp_step": 0.5,
    "action_topic": "zigbee2mqtt/0x842e14fffef15510",
    "action_template": "{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}",
    "json_attributes_topic": "zigbee2mqtt/0x842e14fffef15510",
    "name": "0x842e14fffef15510_climate",
    "unique_id": "0x842e14fffef15510_climate_zigbee2mqtt",
    "device": {
        "identifiers": [
            "zigbee2mqtt_0x842e14fffef15510"
        ],
        "name": "0x842e14fffef15510",
        "sw_version": "Zigbee2mqtt 1.14.2",
        "model": "Radiator valve with thermostat (TS0601_thermostat)",
        "manufacturer": "TuYa"
    },
    "availability_topic": "zigbee2mqtt/bridge/state"
}

```
There seems to be also a firmware difference between the two because one sends also the "auto" configuration for the weekdays and holidays through the main topic and the other one doesn't.

Bart-1992, 2020-08-08:
Hi guys,

I also have 6 moes branded TRV (TS0601 thermostat ), they all first appeared as curtain motor, but eventually got them to show up as TRV in HA.

I have some issues with the mode settings as well.

I found out that it's not possible to turn the valves into 'manual' mode unfortunately. 
Auto mode = weekly schedule mode on TRV itself, off = holiday mode, and heating mode immediately turns into off mode in HA?

To put/hold the TRV in manual mode you have to adjust it on the TRV itself, and only change the temperature trough HA ( so don't do anything with the modes).

Battery status is also not working for all 6 off the valves. 
Valve detection is not doing anything? Open window mode is working for me.

mgrom, 2020-08-10:
Hey, I'm currently working on making some changes to this thermostat implementation. One of them is using Presets instead of Modes. List of modes is fixed set in home assistant so it's not good to use them. Presets are open so you can put there whatever you want. 
There will be presets:
- schedule
- manual
- boost
- away
- complex (this one means that thermostat is working in schedule mode, but temperature was changed manually)
[image]
Mode will be only one - it's not possible to remove this property from climate sensor. 
I think I will create some PR in next 2 days.

Bart-1992, 2020-08-10:
@mgrom Nice work!!

This would be a perfect solution to put the TRV remotely into manual mode.

mgrom, 2020-08-11:
Guys,

Does current configuration let you to change open window detection parameter through mqtt/homeassistant?

Bart-1992, 2020-08-11:
Hi,
@mgrom answer to your questions:

1) There is an option in the current config to turn on/off window detection, however im not able to test the functionality at the moment.

2) all of my 6 TRVS are updating at ireggular frequency unfortunately.
How did you fixxed it? With an automation/script sendint this mqtt message?

Curious when your changes will be implemented in the stable version of zigbee2mqtt.

Thanks,

mgrom, 2020-08-11:
@Bart-1992 
Thanks for info. Window detection will work in next release.

I've made simple automation:
`
trv_automations:
  automation:
    - id: bedroom_trv_get_temp
      alias: Bedroom TRV get current temp
      trigger: 
        - platform: time_pattern
          minutes: "/2"
      action:
        - service: mqtt.publish
          data_template:
            topic: 'zigbee2mqtt/bedroom_moes_trv/set/local_temperature_calibration'
            payload: '0'
`

Just change minutes to anything fits you. The same with payload. Finally change topic and make action to run for all your trvs 

No idea when my changes will be implemented :) sorry, but I'm just a gust here :D

Bart-1992, 2020-08-12:
Hi @mgrom,

I have not tested the schedule mode on the TRV itself, i'm also not planning to use it. There probably is a cheap bad functioning RTC module inside.... if there even is a RTC module allready.... 

I've implemented your automation, it's working perfectly ;)!

However, I have some issues, check the log :

`
zigbee2mqtt:info  2020-08-12 11:59:43: MQTT publish: topic 'zigbee2mqtt/0x842e14fffef6ea3e', payload '{"boost_time":300,"linkquality":36,"force":"normal","comfort_temperature":20,"eco_temperature":15,"position":0,"week":"5+2","workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"preset_temperature":16,"auto_lock":"MANUAL","preset":1,"current_heating_setpoint":"20.0","system_mode":"manual","local_temperature":"28.0","min_temperature":5,"max_temperature":35,"child_lock":"UNLOCKED","local_temperature_calibration":"0.0","window_detection_params":{"valve":"OFF","temperature":5,"minutes":10}}'
zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #366 with data {"type":"Buffer","data":[1]}
`

It gives a strange error about siterwell GS361 conversion? I'm not that into Zigbee2Mqtt's internal functioning, but it looks like it's not recognizing it properly? Do you know what it means? 

Thanks,
BR,
Bart

mgrom, 2020-08-12:
@Bart-1992 can you give me some previous messages? I didn't get dp #366 so far. Is it often?

Bart-1992, 2020-08-12:
@mgrom 

Check the log :

It appears pretty often as you can see. In this log it's only appearing on the same TRV(0x842e14fffefd9bbe), but in my previous post it was another TRV (0x842e14fffef6ea3e)

Indeed maybe 2 minutes is a bit too often, don't know if the battery will drain faster? 

`
zigbee2mqtt:info  2020-08-12 12:54:08: MQTT publish: topic 'zigbee2mqtt/0x842e14fffefd9bbe', payload '{"window_detection_params":{"valve":"OFF","temperature":5,"minutes":10},"linkquality":39,"boost_time":300,"force":"normal","comfort_temperature":20,"eco_temperature":15,"position":0,"week":"5+2","workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":136,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"preset_temperature":15,"auto_lock":"MANUAL","preset":1,"current_heating_setpoint":"17.5","system_mode":"manual","local_temperature":"29.0","min_temperature":5,"max_temperature":35,"child_lock":"UNLOCKED","local_temperature_calibration":"0.0"}'
zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #366 with data {"type":"Buffer","data":[1]}
zigbee2mqtt:info  2020-08-12 12:54:12: MQTT publish: topic 'zigbee2mqtt/0x842e14fffeef7023', payload '{"window_detection_params":{"valve":"OFF","temperature":5,"minutes":10},"linkquality":39,"boost_time":300,"force":"normal","comfort_temperature":20,"eco_temperature":15,"position":0,"week":"5+2","workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":150,"minute":0,"temperature":15}],"preset_temperature":15,"auto_lock":"MANUAL","preset":1,"min_temperature":5,"max_temperature":35,"current_heating_setpoint":"19.5","local_temperature":"29.5","system_mode":"manual","child_lock":"UNLOCKED","local_temperature_calibration":"0.0"}'
zigbee2mqtt:info  2020-08-12 12:54:22: MQTT publish: topic 'zigbee2mqtt/0x00158d00044c76b6', payload '{"battery":100,"voltage":3045,"illuminance":21,"illuminance_lux":21,"linkquality":39,"occupancy":false}'
zigbee2mqtt:info  2020-08-12 12:55:33: MQTT publish: topic 'zigbee2mqtt/0x00158d000313fefd', payload '{"battery":100,"voltage":3025,"illuminance":1000,"illuminance_lux":1000,"linkquality":39,"occupancy":false}'
zigbee2mqtt:info  2020-08-12 12:55:49: MQTT publish: topic 'zigbee2mqtt/0x00158d000313fefd', payload '{"battery":100,"voltage":3025,"illuminance":1000,"illuminance_lux":1000,"linkquality":39,"occupancy":false}'
zigbee2mqtt:info  2020-08-12 12:55:49: MQTT publish: topic 'zigbee2mqtt/0x00158d000313fefd', payload '{"battery":100,"voltage":3025,"illuminance":1000,"illuminance_lux":1000,"linkquality":39,"occupancy":true}'
zigbee2mqtt:info  2020-08-12 12:55:57: MQTT publish: topic 'zigbee2mqtt/0x00158d0004201077', payload '{"battery":100,"voltage":3055,"illuminance":625,"illuminance_lux":625,"linkquality":18,"occupancy":false}'
zigbee2mqtt:info  2020-08-12 12:55:57: MQTT publish: topic 'zigbee2mqtt/0x00158d0004201077', payload '{"battery":100,"voltage":3055,"illuminance":625,"illuminance_lux":625,"linkquality":21,"occupancy":true}'
zigbee2mqtt:info  2020-08-12 12:56:00: MQTT publish: topic 'zigbee2mqtt/0x842e14fffef30ccf', payload '{"min_temperature":5,"linkquality":63,"max_temperature":35,"current_heating_setpoint":"20.0","local_temperature":"30.5","system_mode":"auto","child_lock":"UNLOCKED","local_temperature_calibration":"0.0","window_detection_params":{"valve":"OFF","temperature":5,"minutes":10},"boost_time":300,"force":"normal","comfort_temperature":20,"eco_temperature":15,"position":0,"week":"5+2","workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"preset_temperature":15,"auto_lock":"MANUAL","preset":1}'
zigbee2mqtt:info  2020-08-12 12:56:00: MQTT publish: topic 'zigbee2mqtt/0x842e14fffef30ccf', payload '{"min_temperature":5,"linkquality":63,"max_temperature":35,"current_heating_setpoint":"20.0","local_temperature":"30.5","system_mode":"auto","child_lock":"UNLOCKED","local_temperature_calibration":"0.0","window_detection_params":{"valve":"OFF","temperature":5,"minutes":10},"boost_time":300,"force":"normal","comfort_temperature":20,"eco_temperature":15,"position":0,"week":"5+2","workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20}, [... cut ...] holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":150,"minute":0,"temperature":15}],"preset_temperature":15,"auto_lock":"MANUAL","preset":1,"min_temperature":5,"max_temperature":35,"current_heating_setpoint":"19.5","local_temperature":"29.5","system_mode":"manual","child_lock":"UNLOCKED","local_temperature_calibration":"0.0"}'
`

vyzaq, 2020-08-17:
> I got the same thermostat valve from aliexpress. Mine has a Moes branded box with model number HY368. Maybe add that on the wiki page too?
> 
> I had to troubles joining the network. Joining the network is explained to work after "short press home (turn on), long press home (enter settings), go to setting 5 (wifi logo), press home (only wifi now showing), long press home (wifi now blinking). I then pressed home every few seconds as it seems to timeout pretty fast. But it never joined the network.
> 
> Unscrew/rescrew the battery cover suddenly did the trick, it just joined, even I tried that a few times before.

Got the same device but can't pair it. Debug log is silent also. Do you know any details how to pair it properly?

mgrom, 2020-08-17:
What have you tried already?

mgrom, 2020-08-17:
@Bart-1992 I've tried to monitor mine messages, but never go the one you have... strange

Bart-1992, 2020-08-18:
@mgrom  Got the TRV"s running for a week or so now, the messages disappeared, everything is working stable at the moment. Just waiting for your changes to be updated in the stable release :)!

insipiens, 2020-08-19:
Pairing the TRV to zigbee: I've attached the instruction that came with mine

[image]

dedors, 2020-08-19:
I was only able to pair after reconnecting battery as [commented above](https://github.com/Koenkk/zigbee2mqtt/issues/3821#issuecomment-665960350).
I tried quite long with the instructions and that did not work. Just repower the Tuya while Zigbee joining enabled was successful.

insipiens, 2020-08-20:
Looking forward to seeing the presets version released....being new to zigbee2mqtt (4 weeks) seems a really great project. Looking forward to contributing once I've examined all the functionality....although seems a little intrepid making an update without peer review.

insipiens, 2020-08-25:
> 
> 
> Not sure if this is new info, I can pair the HY368 to Z2M but the device is deactivated - I can change the values on the device by publishing but nothing physically occurs (the motor doesn't respond) . Works fine on tuya hub...
> 
> see payload **running: false** `

Recalling this, the device delays by 4 or 5 minutes before activating to the published current_heating_setpoint command, presumably so any part adjustments aren't acted upon too quickly.

publishing the current_heating_setpoint works fine.

Bartekn86, 2020-08-26:
strange think, domoticz cant set anything  on ts601  so it will be problem in implementation in domoticz ?

Stoatwblr, 2020-08-28:
> I had to troubles joining the network

I've just acquired 4 of these too. The solution (as usual) was to park them right next to a repeater whilst pairing (I used an Ikea Tradfri) and the interview process was near-instant

Once paired, power cycling will demand a clock reset and recalibration routine but it stays zigbee paired.  
This allows pairing then repositioning into final location when it will look for the best repeater  (unlike some other battery devices which stay locked to the same repeater...)

Make sure you run the valve motor calibration routine (AD 1 & 2) AFTER attaching it to the TRV base. This is started by hand after setting the clock but my experience is that it's very easy to inadvertently do that that whilst mounting to the base.

Stoatwblr, 2020-08-28:
comment: with 1.14.3-stable I was getting "not very much" reported.
moving to dev branch is showing a lot more. Thanks @Koenkk 

Now to try and make it work...

Stoatwblr, 2020-08-29:
I can see a bunch of topics (more showing up all the time), but not all are programmable even if you might think they should be (such as putting the TRV into 5+2,. 6+1 or 7 day mode)

Is there any way to get a mqtt device to report all its topics?

Stoatwblr, 2020-08-31:
It looks like "valve_detection" is bogus - in https://github.com/Koenkk/zigbee-herdsman-converters/commit/4f96c1e8792c0565105b2819d69f3ab3990efba1

window_detection is ON or OFF but doesn't show up until toggled on the device or remotely

'window_detection_params': {'valve': 'OFF', 'temperature': 5, 'minutes': 10}  
   I'm still trying to figure out exactly what sent to be sent to it to set these parameters

  'mosquitto_pub -t zigbee2mqtt/radiator_kitchen/window_detection_params-minutes -m 10'
shows nothing logged or sent - presumably I'm doing it wrong.
  
week = 7                           # reports '5+2', '6+1', '7', but not settable using those values
 what's logged is: error 2020-08-31 20:40:29: No converter available for 'week' (5+2)

Is there any way to set herdsman into debug for just this device?

Stoatwblr, 2020-08-31:
@Koenkk 
If it's of any help: this thing appears to have a serial debugging port available and is using a STM8L052

insipiens, 2020-09-01:
@Stoatwblr , In your log the "holidays program" consist of 6 sets of 3 bytes(?), very similar to the analysis I have of the Moes room thermostat program message - although that has 5+1+1 programmable only. See #4185 

This device is using the Tuya conversions currently, it is a Tuya based device but the commands are a different set although I have yet to check the Moes radiator valve.

Have you set up a zigbee sniffer and connected it to their hub to see what is being sent?

Stoatwblr, 2020-09-01:
Mine is a Moes HY368 

Digging around on Tuya's website shows that the actual maker is: 
Xiamen Hysen Control Technology Co., Ltd, 
Product Model：HY368 Zigbee

(https://expo.tuya.com/product?id=543210)

This explains the variations away from Tuya "standard"

I've managed to snap my sniffer off it's plug, so no dumps (yet)

Stoatwblr, 2020-09-01:
In away mode (the airplane icon, and described in the manual as "holiday mode")

Adjusting the preset temperature and days causes these to pop up in mqtt returns

away_preset_temperature = 5             # R away mode (the airplane icon)
away_preset_days = 1                    # R away days min=1, max=30

These can't be written back using set/ (mosquitto pub), but zigbee2mqtt isn't logging an error _or_ a sent message

Stoatwblr, 2020-09-01:
need to find out:
 how to read/set clock 
 how to read/set hysteresis

Sthopeless, 2020-09-01:
Well I don't know till what extend it helps but this are all the dpID's/functions of the Moes TRV's, I also include what should represent on Home Assistant

[image]

I have also opened one of my units but no luck sniffing the pins either wrong configuration options or is not a serial port at all

Stoatwblr, 2020-09-01:
Thanks. I've heavily edited the table I posted yesterday whilst you were posting this too

Check my notes on "R G S V" pads on the front corner of the board. There may be something usable there with a TTL serial port

Stoatwblr, 2020-09-01:
how are the DPs created? I'm seeing numbers that aren't listed above:

debug 2020-08-31 21:02:56: Received Zigbee message from 'radiator_kitchen', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"data":[25,30,36,0,58],"type":"Buffer"}' from endpoint 1 with groupID 0

debug 2020-08-31 21:04:14: Received Zigbee message from 'radiator_kitchen', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,180],"type":"Buffer"},"dp":514,"fn":0,"status":4,"transid":117}' from endpoint 1 with groupID 0

debug 2020-08-31 21:04:14: Received Zigbee message from 'radiator_kitchen', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,1,4],"type":"Buffer"},"dp":515,"fn":0,"status":4,"transid":118}' from endpoint 1 with groupID 0

debug 2020-08-31 21:04:19: Received Zigbee message from 'radiator_kitchen', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":1028,"fn":0,"status":4,"transid":119}' from endpoint 1 with groupID 0

debug 2020-09-01 13:10:18: Received Zigbee message from 'radiator_kitchen', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,20],"type":"Buffer"},"dp":621,"fn":0,"status":4,"transid":121}' from endpoint 1 with groupID 0

debug 2020-09-01 13:17:34: Received Zigbee message from 'radiator_kitchen', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,10],"type":"Buffer"},"dp":556,"fn":0,"status":0,"transid":88}' from endpoint 1 with groupID 0

DP values reported in debugging so far: 

"dp":1028
"dp":104
"dp":112
"dp":113
"dp":1130
"dp":1135
"dp":1293
"dp":263
"dp":366
"dp":371
"dp":372
"dp":514
"dp":515
"dp":556
"dp":614
"dp":615
"dp":617
"dp":619
"dp":620
"dp":621
"dp":626
"dp":629

Stoatwblr, 2020-09-01:
https://community.home-assistant.io/t/i-am-totally-noob-with-zigbee-and-tuya-and-i-need-advice-for-smart-thermostatic-radiator-valves/163607/25 looks promising as a resource 

I'm using openhab, not HA, so I want to do this with mqtt messages

https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-1-b20227251d46

https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d

The second article includes a table of DP codes which should be useful
https://miro.medium.com/max/875/1*iqpMebw-A2us1_oMN20cQg.png

Stoatwblr, 2020-09-01:
I'd been tootling around on developer.tuya.com looking for that stuff - what i did find is this:

https://developer.tuya.com/en/docs/iot/device-development/embedded-license/embedded-license

It looks like the devices are generally running:
 freertos https://www.freertos.org/
 libemqtt https://github.com/menudoproblema/libemqtt
 lwip http://savannah.nongnu.org/projects/lwip/
 mbed  https://os.mbed.com/

Sthopeless, 2020-09-01:
I sacrificed one of my CC2531 dongles and I'm running zboss on it but first time traveler might take awhile to start understanding it all.

Stoatwblr, 2020-09-01:
You and me both on this journey :) Why do I always end up learning more than I ever want to know about the nuts and bolts?

https://github.com/TuyaInc might have useful content too

Sthopeless, 2020-09-01:
So I'm running Zboss, initially I looked for my Tuya Gateway id which is 0x14 (20).
Then I started wireshark and recorded only me changing the temp from Tuya app to 17 and 16.5 and stopped wireshark and looked for those values 
Here it's 17, `AA` 
[image]

And here the 16.6 `A5`
[image]

I also noticed the `data` is pretty much basic Tuya protocol we use on ESP8266's.

insipiens, 2020-09-01:
A5 = 165 so 16.5 degrees multiplied by 10

0202 (big endian same in normal hex) = 514 decimal which was the DP mentioned above as change temperature target.

In the converters file fromZigbee.js this message is programmed as follows

`    tuya_thermostat_current_heating_setpoint: {
        key: ['current_heating_setpoint'],
        convertSet: async (entity, key, value, meta) => {
            const temp = Math.round(value * 10);
            const payloadValue = utils.convertDecimalValueTo2ByteHexArray(temp);
            sendTuyaCommand(entity, 514, 0, [4, 0, 0, ...payloadValue]);
        },
    },
`

Unfortunately my look at the Moes Room Thermostat has completely different DP values for similar commands - was hoping for some consistency....no luck there....:|

Sthopeless, 2020-09-02:
thank you @insipiens.
I been doing something like this for the device, i'll sort and fix it tomorrow
[image]

insipiens, 2020-09-02:
@Sthopeless  Nice work! 👍 

A comment on the calibration results data you have there, and apologies if I'm explaining the bleeding obvious: for negative values e.g -10  it uses the binary compliment so -10 = 0b1111110110 = 0xfffffff6  (as it's dealing in 1/10 of a degree) . 

I'm not fully up to speed on the programming of zigbee2mqtt yet so, like you finding my way....

mgrom, 2020-09-03:
Guys you're inventing the wheel again
https://github.com/Koenkk/zigbee-herdsman-converters/blob/8102114d8bde669342de577202e8ccefdb9fd281/converters/fromZigbee.js#L221

insipiens, 2020-09-04:
mgrom. understood given all your hard work! trying to make more functionality available....this should be encouraged, if for nothing else but greater participation and understanding.

For me, the program setting aspect is still not fnctional across [any/all?] thermostats - I'm keen to engage further on that given my work on the Moes room thermostat (nearing minimal completion!)

mgrom, 2020-09-04:
@insipiens it wasn't me who made all this hard work. I've just added few fixes.

Anyway, my advise: check the code I've linked, then try to create new methods in toZigbee file in converters that will for example change week type (7, 6+1, 5+2). Command structure that have to be send to device is exaxtly the same as its report, it's quite easy. More difficult will be setting weekly schedule in HA, than coding it back to zigbee format :)

If you have any questions, I'll try to help.

shirou93, 2020-09-04:
I have error on mqtt when change mode
[image]

Showing windows sensor etc as switch is a bug? Why not binary sensor?

shirou93, 2020-09-06:
On Google home (home assistant expose thermostat) only show temperature inside. I can't change temperature on Google home (only text "other" in circle).
Tap mode no working.
[image]

Bartekn86, 2020-09-06:
Sep 06 18:01:19 raspberrypi npm[6918]: Zigbee2MQTT:error 2020-09-06 18:01:19: No converter available for 'mode' (auto)
Sep 06 18:02:17 raspberrypi npm[6918]: Zigbee2MQTT:error 2020-09-06 18:02:17: No converter available for 'mode' (eco)

Sep 06 18:11:48 raspberrypi npm[6918]: Zigbee2MQTT:info 2020-09-06 18:11:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'preset' to 'bn duzy glowica' failed: 'Error: Command 0xec1bbdfffe91c050/1 manuSpecificTuyaDimmer.setData({"status":0,"transid":205,"dp":1028,"fn":0,"data":[1,0]}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'Timeout' (9999))'","meta":{"friendly_name":"bn duzy glowica"},"type":"zigbee_publish_error"}'

[772 more replies]

closed 2021-08-11 by github-actions[bot]