Koenkk/zigbee2mqtt #3822: Konke door sensor problem

Hydci, 2020-06-28:
Hello I have a problem with two konke door sensors I am under chanel 15 and I can't synchronize them here are the logs thanks for the help

This is what creates it in the database.db
```
{"id":28,"type":"EndDevice","ieeeAddr":"0x14b457fffe0978cc","nwkAddr":63144,"manufId":4712,"manufName":"Konke","powerSource":"Battery","modelId":"3AFE280104020015","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,64704],"outClusterList":[3,64704],"clusters":{"genBasic":{"attributes":{"modelId":"3AFE280104020015","manufacturerName":"Konke","powerSource":3,"zclVersion":3,"appVersion":40,"stackVersion":0,"hwVersion":1,"dateCode":"190829"}},"aqaraOpple":{"attributes":{"16":"\u0013\u0004\u0002\u0000\u0015111011730000608","17":"\u0018\u0004\u0002\u0000\u0015332R1P000209BD11003E"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x0000000000000000","zoneState":1}}},"binds":[]}},"appVersion":40,"stackVersion":0,"hwVersion":1,"dateCode":"190829","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1593340004086}
```
Here are the log
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4842077/log.txt)

## Debug Info
Zigbee2mqtt version: 1.14 master
Adapter hardware: cc2538+cc2592
Adapter firmware version:

Hydci, 2020-06-30:
Up ?

Koenkk, 2020-07-01:
What do you mean with cant synchronize them? It paired correctly right?

Hydci, 2020-07-01:
Hello,
No as you can see in the logs I get this message
```
juin 28 12:19:53 raspberrypi sudo[981]: zigbee2mqtt:debug 2020-06-28 12:19:53: Received Zigbee message from '0x14b457fffe0978cc', type 'readResponse', cluster 'genBasic', data '{"modelId":"3AFE280104020015"}' from endpoint 1 with groupID 0
juin 28 12:19:53 raspberrypi sudo[981]: zigbee2mqtt:warn  2020-06-28 12:19:53: Received message from unsupported device with Zigbee model '3AFE280104020015'
```
I receive the same message when I open or close the sensor I try to synchronize it a lot of times but that but always make the same message

Koenkk, 2020-07-01:
Fixed the issue, fix will be available in the dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Hydci, 2020-07-02:
I confirm the sensor works perfectly on the DEV branch thanks for the fix I close the problem

closed 2020-07-02 by Hydci