Koenkk/zigbee2mqtt #3823: Howto get rid of a not present device?

SargonofAssyria, 2020-06-28:
# Bug Report
There is an error at every start of zigbee2mqtt
`error 2020-06-28 13:09:42: Failed to setup reporting for '0x00158d0002b7d689' - Error: Bind 0x00158d0002b7d689/1 genOnOff from '0x00124b001ca1b0bc/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)`
## What did you expect to happen
No error
0x00158d0002b7d689 is a Xiaomi light bulb
0x00124b001ca1b0bc was a CC2531 router device that I do not have anymore in the network.
How do I get rid of this binding error to a device that does not exists.
## How to reproduce it (minimal and precise)
Every start of zigbee2mqtt

## Debug Info
Zigbee2mqtt version:  1.14.0-dev (commit #869d6a5)
Adapter hardware: CC1352P-2
Adapter firmware version: 20200417

Koenkk, 2020-06-28:
You have to `force_remove` the device (docs: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove)

closed 2020-06-28 by Koenkk

SargonofAssyria, 2020-06-28:
I tried both remove and force_remove I get this response.
`error 2020-06-28 20:23:18: Cannot remove, device '0x00124b001ca1b0bc' does not exist
error 2020-06-28 20:28:26: Cannot force remove, device '0x00124b001ca1b0bc' does not exist
`

Koenkk, 2020-06-29:
Is `0x00124b001ca1b0bc` still in `database.db`? If so, stop zigbee2mqtt, remove that line from `database.db` and start zigbee2qmtt

SargonofAssyria, 2020-06-29:
I am/was mistaken. I did not look good enough. Sorry. :-( That device 0x00124b001ca1b0bc is the Coordinator CC1352P-2
I already tried to unbind the devices both ways, but that did not help.

to4ko, 2020-08-15:
same here - exactly same error - like my plug were directly bind to coordinator (i'm using CC2538)

kennymc-c, 2023-01-12:
I see the same error message for three Innr SP120 plugs which are not connected to a power outlet at the moment. I tried to use the disable function for these devices since I don't use them regularly but the messages still appear.