Koenkk/zigbee2mqtt #3825: Blitzwolf BW-IS3 false positive - broken?

dkurzelewski, 2020-06-28:
# Bug Report
## What happened
I bay 3x Blitzwolf BW-IS3. After successful pair and 1/2 days, 2 of them start report false positive motion alert. Did this 2 is broken? Last one device work perfect, so I can't find any relationship or other behavior why that 2 starts detect false positive alert.
 
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2531

## Logs
https://pastebin.com/y6sYJNau

CarpeDiemRo, 2020-07-03:
Same problem here. A lot of false positive messages

closed 2020-08-09 by stale[bot]

tibehrius, 2020-10-28:
I have exactly the same problem here. During the night, in a completely empty room, the sensor is activated all the time.

heggink, 2020-12-12:
Same. I ordered a couple and all have the same issue: lot's of false positives. I opened one up to see if there was a way to set the sensitivity. It's built around the tuya TYZS3 module. Wonder if this is 'just' a sensitivilty issue which can be configured in the device... So far no luck finding a solution :-(.

Rubines, 2021-03-02:
Same problem this pir sensor is unuseable over night I hat 300 false detections

Diamond2, 2021-03-17:
I am having the same issue with zigbee2mqtt 1.17.1-4
Adapter hardware: CC2530