Koenkk/zigbee2mqtt #3826: "report: false" does not stop reporting

blaxer, 2020-06-28:
# Bug Report
## What happened

I enabled reporting to test out the feature, I then disabled reporting (report: false) and everything is still reporting quite regularly which eventually caused major problems. I have 150+ devices, mostly Osram and Ikea, I have power cycled some of the Osram devices but they are still reporting. Groups are still reporting. How can I make this stop? it eventually seems to crash / overload Z2M and or the CC26X2R1.

## What did you expect to happen

I expect that when you set reporting to false, all devices would stop reporting

## How to reproduce it (minimal and precise)

Set reporting to true, then false, reporting still occurs

## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200417

Koenkk, 2020-06-28:
This is indeed a bug, will try to fix this soon.

For the meantime, to stop Ikea devices from reporting just re-power them, other devices have to be factory reset.

CSchlipp, 2020-07-16:
I have exactly the same issue - switched it on for testing, now have extensive traffic in the network as the report cannot be disabled. This seems to be even worse with groups, as the group state is also always updated.

Power-cycling the Ikea spots did not help, probably both lines need to be disconnected for a full "reset".

I would highly appreciate a fix for this!

Koenkk, 2020-08-04:
Fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-08-04 by Koenkk