Koenkk/zigbee2mqtt #3827: Philips hue dimmer, failed to configure

wneessen, 2020-06-28:
# Bug Report
## What happened
Only one of my philips hue dimmer switches will correctly configure with zigbee2mqtt. While all of them are paired w/o problems, I was only able to configure one correctly. Once I try to add another switch, it will pair, but fail during the configuration. Corresponding, they do not submit any button-actions to zigbee2mqtt/hassio

## What did you expect to happen
All dimmer switches should be able to configure correctly.

## How to reproduce it (minimal and precise)
* Remove philips hue dimmer switch from zigbee2mqtt using ```zigbee2mqtt/bridge/config/remove```
* Switch zigbee2mqtt into pairing mode
* Reset dimmer switch using a small button a the back until it's in pairing mode
* Zigbee2mqtt will discover the switch and successfully interview/pair, then it will start the configuration and fail with:
```
Failed to configure '0x0017880102c0fc85', attempt 5 (Error: Bind 0x0017880102c0fc85/1 genOnOff from '0x00212effff059569/1' failed (undefined)
     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
     at Generator.throw (<anonymous>)
     at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

## Debug Info
Zigbee2mqtt version: 1.14.0 (also tested 1.14.0-dev)
Adapter hardware: Conbee II
Adapter firmware version: 0x26580700

## Logs
https://hastebin.com/ruhesebiqa.coffeescript

sjorge, 2020-06-28:
Are both the same firmware version? Or is one updated and the other not?

wneessen, 2020-06-28:
The connected one is v5.45.1.17846. I had two more switches paired with the same firmware version, which had the above mentioned issue. For testing purpose I connected one of those to my Hue Hub and updated it to v6.1.1.28573, but the issue still occurs.

So in summary:
- switch 1: v5.45.1.17846 (works)
- switch 2: v5.45.1.17846 (does not work)
- switch 3: v6.1.1.28573 (does not work)

wneessen, 2020-07-02:
Any idea/help to troubleshoot/fix this?

Koenkk, 2020-07-03:
This is a Conbee specific issue. First try with zigbee2mqtt 1.14.1-dev (as it contains some conbee fixes). If that doesn't work provide the herdsman debug logging to the conbee team here: https://github.com/Koenkk/zigbee-herdsman/issues/72

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

closed 2020-07-03 by Koenkk