Koenkk/zigbee2mqtt #3830: E18-MS1PA1-PCB attached to Shelly to extend Zigbee network

matejzero, 2020-06-30:
Hello,

I'm just doing a double check if it's possible to use [this board](https://www.aliexpress.com/item/4000834166422.html) as a router with Zigbee2mqtt? I found it on https://www.zigbee2mqtt.io/information/supported_adapters.html, but just want to be sure if this still works.

I use Shelly's for all my relays and zigbee for my sensors. Since I don't have many powered zigbee devices, my zigbee range is not very good and I want to extend it. 

My idea is to but this board and cut it, so I only have a Zigbee chip and headers (as I don't need USB when used as router) and connect it to Shelly 3,3V and GND pins for power and hide it in flush box.

Will this work?
Are headers on the board enough to flash it with CC-tool or would I need a full stick plugged into USB to be able to flash it?

Koenkk, 2020-06-30:
I don't know this board so hard to judge, but I expect it will. Do you have experience with this chip? @ptvoinfo

matejzero, 2020-06-30:
This is suppose to be CC2530 + RF CC2592 chip.

I found https://github.com/Koenkk/zigbee2mqtt/issues/3772, which is (i think) the same board with more headers.

closed 2020-06-30 by matejzero