Koenkk/zigbee2mqtt #3831: Cannot bind devices from same brands that should interoperate

vdkeybus, 2020-06-30:
## What happened

Sent:
mosquitto_pub -t 'zigbee2mqtt/bridge/bind/0x000b57fffe11b778' -m '0xd0cf5efffec853c9'
mosquitto_pub -t 'zigbee2mqtt/bridge/bind/0x0017880104ad2466' -m '0x00178801083b7407'

## What did you expect to happen

Ikea dimmer binds to Ikea light
Philips Hue dimmer binds to Philips Hue light
(in both cases, logs indicated that they could not bind 2 (Philips: 3) clusters.

## How to reproduce it (minimal and precise)

As under 'What happened'.

## Debug Info

Jun 30 13:03:48 zacate npm[591271]: zigbee2mqtt:error 2020-06-30 13:03:48: Failed to bind cluster 'genScenes' from '0x0017880104ad2466' to '0x00178801083b7407' (Error: Bind 0x0017880104ad2466/1 genScenes from '0x00178801083b7407/11' failed (Error: AREQ - ZDO - bindRsp after 10000ms))
Jun 30 13:03:48 zacate npm[591271]: zigbee2mqtt:info  2020-06-30 13:03:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind_failed","message":{"from":"0x0017880104ad2466","to":"0x00178801083b7407","cluster":"genScenes"}}'
Jun 30 13:03:58 zacate npm[591271]: zigbee2mqtt:error 2020-06-30 13:03:58: Failed to bind cluster 'genOnOff' from '0x0017880104ad2466' to '0x00178801083b7407' (Error: Bind 0x0017880104ad2466/1 genOnOff from '0x00178801083b7407/11' failed (Error: AREQ - ZDO - bindRsp after 10000ms))
Jun 30 13:03:58 zacate npm[591271]: zigbee2mqtt:info  2020-06-30 13:03:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind_failed","message":{"from":"0x0017880104ad2466","to":"0x00178801083b7407","cluster":"genOnOff"}}'
Jun 30 13:04:08 zacate npm[591271]: zigbee2mqtt:error 2020-06-30 13:04:08: Failed to bind cluster 'genLevelCtrl' from '0x0017880104ad2466' to '0x00178801083b7407' (Error: Bind 0x0017880104ad2466/1 genLevelCtrl from '0x00178801083b7407/11' failed (Error: AREQ - ZDO - bindRsp after 10000ms))
Jun 30 13:04:08 zacate npm[591271]: zigbee2mqtt:info  2020-06-30 13:04:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind_failed","message":{"from":"0x0017880104ad2466","to":"0x00178801083b7407","cluster":"genLevelCtrl"}}'

Zigbee2mqtt version: 1.14.0-dev
Adapter hardware: CC2531
Adapter firmware version: {"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}

Koenkk, 2020-06-30:
This error usually happens when the device is sleeping while the command is executed. Make sure to wakeup the device right before executing it (by clicking a button on it).

vdkeybus, 2020-07-03:
I was able to bind the Philips Hue devices by issuing the bind command right after having pressed a button. It does not operate though; the light does not respond to button presses.

Koenkk, 2020-07-03:
Did you first unbind it from the coordinator? (https://www.zigbee2mqtt.io/devices/324131092621.html#binding)

vdkeybus, 2020-07-03:
Missed that. Thanks. But:

- it appears you must unbind the dimmer from the coordinator before you bind it to the lamp
First: mosquitto_pub -t 'zigbee2mqtt/bridge/unbind/0x0017880104ad2466' -m 'coordinator'
Then: mosquitto_pub -t 'zigbee2mqtt/bridge/bind/0x0017880104ad2466' -m '0x00178801083b7407'

- the lamp can only be turned on or off. It is not possible to dim it ?

Koenkk, 2020-07-04:
> the lamp can only be turned on or off. It is not possible to dim it ?

Should also be possible, what do you see in the debug logging when executing those MQTT commands?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

vdkeybus, 2020-07-04:
For now I keep the config that is failing, i.e. unable to dim with the Hue
remote.

Pressing 'I' (on) results in (lamp turns on):

Jul 04 22:15:57 zacate npm[1001592]: zigbee2mqtt:debug 2020-07-04 22:15:57:
Received Zigbee message from '0x0017880104ad2466', type
'commandHueNotification', cluster 'manuSpecificPhilips', data
'{"button":1,"unknown1":3145728,"type":0,"unknown2":0,"time":0}' from
endpoint 2 with groupID 0
Jul 04 22:15:57 zacate npm[1001592]: zigbee2mqtt:debug 2020-07-04 22:15:57:
Received Zigbee message from '0x0017880104ad2466', type
'commandHueNotification', cluster 'manuSpecificPhilips', data
'{"button":1,"unknown1":3145728,"type":2,"unknown2":0,"time":0}' from
endpoint 2 with groupID 0
Jul 04 22:15:57 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:15:57:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload
'{"counter":1,"linkquality":99,"brightness":193,"battery":100,"update_available":false,"action":"on-press","duration":0}'
Jul 04 22:15:57 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:15:57:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload
'{"counter":1,"linkquality":99,"brightness":193,"battery":100,"update_available":false,"action":""}'
Jul 04 22:15:57 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:15:57:
MQTT publish: topic
'homeassistant/device_automation/0x0017880104ad2466/action_on-press/config',
payload
'{"automation_type":"trigger","type":"action","subtype":"on-press","payload":"on-press","topic":"zigbee2mqtt/0x0017880104ad2466/action","device":{"identifiers":["zigbee2mqtt_0x0017880104ad2466"],"name":"0x0017880104ad2466","sw_version":"Zigbee2mqtt
1.14.0-dev","model":"Hue dimmer switch
(324131092621)","manufacturer":"Philips"}}'
Jul 04 22:15:57 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:15:57:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466/action', payload
'on-press'

Pressing '*' (dim down) results in (lamp does not dim however):

Jul 04 22:19:01 zacate npm[1001592]: zigbee2mqtt:debug 2020-07-04 22:19:01:
Received Zigbee message from '0x0017880104ad2466', type 'commandStep',
cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":30,"transtime":9}'
from endpoint 1 with groupID 0
Jul 04 22:19:01 zacate npm[1001592]: zigbee2mqtt:debug 2020-07-04 22:19:01:
Received Zigbee message from '0x0017880104ad2466', type
'commandHueNotification', cluster 'manuSpecificPhilips', data
'{"button":3,"unknown1":3145728,"type":0,"unknown2":0,"time":0}' from
endpoint 2 with groupID 0
Jul 04 22:19:01 zacate npm[1001592]: zigbee2mqtt:debug 2020-07-04 22:19:01:
Received Zigbee message from '0x0017880104ad2466', type
'commandHueNotification', cluster 'manuSpecificPhilips', data
'{"button":3,"unknown1":3145728,"type":2,"unknown2":0,"time":0}' from
endpoint 2 with groupID 0
Jul 04 22:19:02 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:19:02:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload
'{"counter":1,"linkquality":99,"brightness":223,"battery":100,"update_available":false,"action":"down-press","duration":0}'
Jul 04 22:19:02 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:19:02:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload
'{"counter":1,"linkquality":99,"brightness":223,"battery":100,"update_available":false,"action":""}'
Jul 04 22:19:02 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:19:02:
MQTT publish: topic
'homeassistant/device_automation/0x0017880104ad2466/action_down-press/config',
payload
'{"automation_type":"trigger","type":"action","subtype":"down-press","payload":"down-press","topic":"zigbee2mqtt/0x0017880104ad2466/action","device":{"identifiers":["zigbee2mqtt_0x0017880104ad2466"],"name":"0x0017880104ad2466","sw_version":"Zigbee2mqtt
1.14.0-dev","model":"Hue dimmer switch
(324131092621)","manufacturer":"Philips"}}'
Jul 04 22:19:02 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:19:02:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466/action', payload
'down-press'

Then, I unbound the remote control from the lamp again (I pressed 'O' (turn
off) to ensure the remote is listening, I highlighted the unbind events):
mosquitto_pub -t 'zigbee2mqtt/bridge/unbind/0x0017880104ad2466' -m
'0x00178801083b7407'

[... 263 lines cut ...]
'commandHueNotification', cluster 'manuSpecificPhilips', data
'{"button":2,"unknown1":3145728,"type":3,"unknown2":0,"time":17}' from
endpoint 2 with groupID 0
Jul 04 22:27:55 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:27:55:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload
'{"counter":1,"linkquality":99,"brightness":255,"battery":100,"update_available":false,"action":"up-hold-release","duration":1.794}'
Jul 04 22:27:55 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:27:55:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload
'{"counter":1,"linkquality":99,"brightness":255,"battery":100,"update_available":false,"action":""}'
Jul 04 22:27:55 zacate npm[1001592]: zigbee2mqtt:info  2020-07-04 22:27:55:
MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466/action', payload
'up-hold-release'

The pairing / binding happens away from the coordinator, i.e. there are
devices in between to complete the path. The Hue remote control and Hue
lamp are within 2 m from each other.

I have a Zigbee Wireshark configured and working in the same room where the
binding is attempted. If you think it's useful, I can log that as well, but
please let me know whether (and where) I should provide the raw output
rather than the Wireshark output.

Thanks,

J.

Koenkk, 2020-07-06:
I don't understand why it's not working yet, the log shows that `genLevelCtrl` is bound succesfully. Something you can also try is to first `force_remove` the device (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove) and factory reset it. Then unbind it from the coordinator and bind it to the bulb again.

If that doesn't work, please provide the sniff when doing the binding and sending the dim commands to the bulb. You can share the pcapng file + network key privately on telegram (@koenkk).

vdkeybus, 2020-07-06:
After I remove / force_remove and factory reset, I should still have it
(re-)join first before unbind from coordinator and bind to lamp, right ?

vdkeybus, 2020-07-06:
BTW I have no idea how to contact you for the pcapng file. I see no contact
information link on your github page.

jeroen.vandenkeybus@gmail.com>:

vdkeybus, 2020-07-06:
I performed a full reset. The device re-joined automatically. I did unbind
(coord) and bind (lamp) and now it is working fine. I have the impression
that the pairing sequence is different than last time. The dimmer got an
OTA update though.

Thanks for your help.

I will now try the Ikea ones...

Koenkk, 2020-07-07:
Good, I assume this can be closed now.

closed 2020-07-07 by Koenkk