Koenkk/zigbee2mqtt #3832: Problem reading device specific configuration values.  e.g. occupancy_timeout, motion_sensitivity

trankin, 2020-06-30:
I'm super new to zigbee2mqtt, and am in the process of exploring this solution.  First impressions are very positive, and Zigbee in general seems pretty interesting.

One of the things that we do often in our environments is that we set custom values on the sensor devices.  This seems to be working very well in the zigbee2mqtt environment, and in fact is smoother and more seamless than what we've been doing with zWave.  However I'm having a hard time confirming that custom values have been set.

Examples. 
My controller is one of the CC2531 USB development boards (seems to be working as expected.)

I'm exploring the Hue Motion detector: https://www.zigbee2mqtt.io/devices/9290012607.html

I can successfully set motion_sensitivity and occupancy_timeout using the "zigbee2mqtt/[FRIENDLY_NAME]/set" topic, I've also confirmed that I can set the values with topics such as "zigbee2mqtt/[FRIENDLY_NAME]/set/occupancy_timeout"

I can tell that the values are being set as they appear to be working as expected after changing the values.  Additionally if I turn debugging on I can see that the debug messages indicate that the value has been sent to the device.

The Problem
What I need to be able to do is to query the device to see what the current settings are or to confirm that they have been set properly.  In the documentation I can see that there is the following topic "zigbee2mqtt/[FRIENDLY_NAME]/get/occupancy_timeout", and if I publish a blank message to that topic, in the debug messages I can see that a response indicates what the value is, however I'm not seeing any value being published in the mqtt topics with that response.  

I very well could be overlooking something simple, but my expectation is that I would send a "get" request and would be able to monitor some topic to see the response, or I would expect that new value to be sent back in the next data packet for device state @ "zigbee2mqtt/[FRIENDLY_NAME]" but it's not there.

See below for an example of the debug output I'm getting from a set and a get flow.

`
zigbee2mqtt:debug 2020-06-30 16:36:15: Received MQTT message on 'zigbee2mqtt/0x0017880106f57f73/set/occupancy_timeout' with data '120'
zigbee2mqtt:debug 2020-06-30 16:36:15: Publishing 'set' 'occupancy_timeout' to '0x0017880106f57f73'

zigbee2mqtt:debug 2020-06-30 16:36:44: Publishing get 'get' 'occupancy_timeout' to '0x0017880106f57f73'
zigbee2mqtt:debug 2020-06-30 16:36:45: Received Zigbee message from '0x0017880106f57f73', type 'readResponse', cluster 'msOccupancySensing', data '{"pirOToUDelay":120}' from endpoint 2 with groupID 0
`
All seems good, but I don't see how to access that get response in my application code.

Many thanks,

Thomas

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

trankin, 2020-06-30:
Apologies, I realize that I've destroyed the template provided.. I will reformat the original message momentarily.

Koenkk, 2020-07-01:
Thanks for reporting, just fixed this. Now it will confirm to MQTT on both set and get. Changes will be available in the dev branch in a few hours: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

trankin, 2020-07-01:
@Koenkk, thanks very much for the quick response.. I was reviewing the source code just a few minutes ago because I believed that I was understanding where it needed to go.. e.g. in the converters.. and I happened to notice that you had just put in the response: https://sharex.tomrankin.net/ShareX/2020/07/chrome_FaPxCgaQxH.png   I'm getting a better feel for the source, I'll be able to contribute soon.

trankin, 2020-07-01:
@Koenkk, I added in a PR to add in motion_sensitivity as well.  Thanks for all of your work on this.

Koenkk, 2020-07-03:
Cool, and congratz with your first pr, thanks!

I assume this can be closed now.

closed 2020-07-03 by Koenkk