Koenkk/zigbee2mqtt #3834: [docker] Docker images "latest" and "latest-dev" no longer start

tavalin, 2020-07-01:
# Bug Report
## What happened

I pulled the latest images and tried to start the container.  The container doesn't start and I receive the following error:

`container_linux.go:349: starting container process caused "exec: \"./run.sh\": stat ./run.sh: no such file or directory"`
## What did you expect to happen

The container starts up successfully as with previous images.

## How to reproduce it (minimal and precise)

Running on a RPi 4, using the official docker images.  Pull docker image, create container with the relevant environment variables and try and start the container.

## Debug Info
Zigbee2mqtt version: `latest `and `latest-dev` tags
Adapter hardware: CC2531
Adapter firmware version:

trankin, 2020-07-01:
Looks like the entrypoint was renamed from run.sh to docker-entrypoint.sh  I imagine if you change your last two lines in your custom Dockerfile to 
```
ENTRYPOINT ["docker-entrypoint.sh"]
CMD ["npm", "start"]
``` 
it should get you going again.

Here's a recent diff: in source: https://github.com/Koenkk/zigbee2mqtt/pull/3722/files#diff-ebacf6f6ae4ee68078bb16454b23247d
And confirmed that "latest" appears to be using this new syntax.

https://hub.docker.com/layers/koenkk/zigbee2mqtt/latest/images/sha256-3ab549b1e9d4a7faf3af4d8f242afc6a34a6160c2dd730d85f93b9cc0daaa981?context=explore

trankin, 2020-07-01:
@tavalin what are your specific command line parameters.  I tested this with the documented example settings and it seems to work as expected.

`docker run -it -v $(pwd)/data:/app/data --device=/dev/ttyACM0 -e TZ=Europe/Amsterdam -v /run/udev:/run/udev:ro --privileged=true koenkk/zigbee2mqtt`

I picked that command up from here: https://www.zigbee2mqtt.io/information/docker.html

tavalin, 2020-07-01:
@trankin Thanks for your post, it's helped me resolve the issue.  I didn't specify the entrypoint but when `ouroboros` pulled the new image, the new container was created with the old entrypoint, which obviously no longer exists in the new image.  I've updated those settings and it appears to be working again. 👍

closed 2020-07-01 by tavalin

unholyhimura, 2020-07-12:
@trankin I'm having a similar issue due to the modification of the entry point but I'm using Docker from Synology DSM. Should I open another issue or this one is fine?

1.14.0 version is working fine. 1.14.1.

I'm unable to start the latest version:
/usr/local/bin/docker-entrypoint.sh: exec: line 17: illegal option --

Docker in Synology is a little bit weird, I'm not able to change the entry point in the configuration itself. Without the entry point I'm not able to specify the USB port.

Thanks