Koenkk/zigbee2mqtt #3836: HY369RT.md

GuillaumeP01, 2020-07-01:
Hello,
First of all thank you for your development on the Zigbee.
On the page https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/HY369RT.md
it is indicated the reference HY369RT unfortunately I cannot find this model on Aliexpress.
However, we find this:
https://fr.aliexpress.com/item/4001161658482.html
https://fr.aliexpress.com/item/4001065414044.html
https://fr.aliexpress.com/item/4001065111037.html
The reference on Aliexpress is HY368 and not HY369. Could you confirm the model compatible with Zigbee2mqtt?
cordially
Guillaume

Koenkk, 2020-07-01:
@sjorge do you maybe know something about this?

sjorge, 2020-07-01:
Those look to also be TuYa, I got one of the essential branded ones from amazon.de before sending it back, the quality was poor (and no valve reporting).

Given the TuYa code we have seem to work for all Tuya TRV's so far, I would assume it will also work for the ones linked above. Maybe with a small change to devices.js to add them to the existing list.

GuillaumeP01, 2020-07-02:
ok thank you very much

GuillaumeP01, 2020-07-02:
I found on Aliexpress a version which seems compatible zigbee2mqtt according to a customer.

https://fr.aliexpress.com/item/4001043738901.html?spm=a2g0o.cart.0.0.56843c00fKcU6X&mp=1

shirou93, 2020-07-04:
Hy369RT have the same zigbeemodel as this termostat.

https://zigbee.blakadder.com/MoesHouse_HY369-ZB-RVT.html

Can you confirm this compatible?

shirou93, 2020-07-22:
> Hello,
> First of all thank you for your development on the Zigbee.
> On the page https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/HY369RT.md
> it is indicated the reference HY369RT [...]

On aliexpress someone write  comment "in zigbee2mqtt this device is blind" hy368

cadavre, 2020-07-24:
With 1.14.2 this valve (HY368) integrates well! There was a bug where valve was discovered as cover, but now it is fixed! I just paired mine.

closed 2020-08-17 by GuillaumeP01

shirou93, 2020-09-04:
i have mqtt errors when i choice mode like eco etc.

[image]

shirou93, 2020-09-04:
i have mqtt errors when i choice mode like eco etc.

[image]

Why entity like window open etc show as switch no binary sensor?

@GuillaumeP01  can you reopen issue?

GuillaumeP01, 2020-09-04:
Hello, the subject was to check the reference in the documentation and not the integration of the valve to see the following subject:
https://github.com/Koenkk/zigbee2mqtt/issues/3821