Koenkk/zigbee2mqtt #3838: OTA seems broken for some Trådfri bulbs?

sjorge, 2020-07-01:
# Bug Report
## What happened
I noticed there was a new firmware for some trådfri bulbs, for most it got picked up but for some... debug logging indicate we have an update but update_available remains false and we can't update them.

## What did you expect to happen
Logging and actual situation to be in agreement.

```
zigbee2mqtt:debug 2020-07-01 19:42:20: Is new image available for '0x000b3cfffef575ce', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":8449,"fileVersion":304170354}', latest meta '{"fileVersion":304170354,"fileSize":168318,"url":"https://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_06_24_133052/bin/159696-TRADFRI-bulb-w-1000lm-1.2.214.ota.ota.signed"}'
zigbee2mqtt:debug 2020-07-01 19:42:21: Is new image available for '0x000b3cfffef43b06', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":8449,"fileVersion":304170354}', latest meta '{"fileVersion":304170354,"fileSize":168318,"url":"https://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_06_24_133052/bin/159696-TRADFRI-bulb-w-1000lm-1.2.214.ota.ota.signed"}'
```

These are not E27 1000lm W bulbs, they are E14 400lm WS bulbs.

```
{"id":19,"type":"Router","ieeeAddr":"0x000b3cfffef575ce","nwkAddr":10436,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E14 W op/ch 400lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E14 W op/ch 400lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1592949580545}
{"id":22,"type":"Group","groupID":14243,"members":[{"deviceIeeeAddr":"0x000b3cfffef575ce","endpointID":1},{"deviceIeeeAddr":"0x000b3cfffef43b06","endpointID":1}],"meta":{}}
{"id":20,"type":"Router","ieeeAddr":"0x000b3cfffef43b06","nwkAddr":29809,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E14 W op/ch 400lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E14 W op/ch 400lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1593224813065}
{"id":22,"type":"Group","groupID":14243,"members":[{"deviceIeeeAddr":"0x000b3cfffef575ce","endpointID":1},{"deviceIeeeAddr":"0x000b3cfffef43b06","endpointID":1}],"meta":{}}
```

From the release notes:
```
Release Version: 1.13.0
29th June 2020
LEPTITER, GUNNARP_NA EU, LIGHT-ENGINE, DECO, WS2.0 CLR, WW2.0 (V-2.3.050).
WS2.0 OPAL (V-2.0.029).
CWS1.0 (1.3.013).
Fixed On Off State after OTA Upgrade.
```

So I believe that they should indeed not have an update, so I am not sure why the debug logging says there is?
When using ota_update/check or ota_update/update it correctly (I think) mentions that there is no update.

## How to reproduce it (minimal and precise)
Not sure, I keep getting the above debug logging when these bulbs do the query themselves every few hours.
Coincidentally the test bulb I paired today was a 1000lm W one, so maybe there is some state getting stuck somewhere in the ota code?

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

Koenkk, 2020-07-03:
I'm not sure if I fully got it, but I think there is some misunderstanding of what `Is new image available for..` means. What it means is that it is checking for a new update, not that a new update is available.

sjorge, 2020-07-03:
The image 159696-TRADFRI-bulb-w-1000lm-1.2.214.ota.ota.signed seems odd though as those are definately not 1000lm.
Although the might share a firmware and it's just badly named, as I do not see any other that would be a correct match though. I'll close it for now.

closed 2020-07-03 by sjorge