Koenkk/zigbee2mqtt #3839: Issue with some Bosch Motion Sensors

joe540, 2020-07-02:
Hello! 

# Bug Report
## What happened
One of my Bosch Motion Sensors (model ISW-ZPR1-WP13) does trigger correctly but doesn't set itself back to "occupancy:false"
## What did you expect to happen
The sensor should set "occupancy:false" after 5 seconds

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531 and Conbee2
Adapter firmware version:

## More info

  I'm having issues with some Bosch Motion Sensors model ISW-ZPR1-WP13. It started 2-3 weeks ago when I did some reno and moved a lot of stuff arround in my basement (near the coordinator) but everything is now the way it was and the path is clear between my coordinator and the routers near it.

So what happens is that the sensor does trigger the "occupancy:true" correctly, but it never goes back to false. I have 3 routers near it (1 is 6' away in line-of-sight, one is ~8' away with 1 floor in the way and one is ~10' away with one drywall in the way)

What I tried:
- Restarted Zigbee2MQTT multiple times, and rebooted the device it runs on also
- Ensured Zigbee2MQTT and the CC2531 firmware were up to date
- I replaced the motion sensor with a brand new one (of the same model) and it worked correctly for a few hours then started doing the same thing
- I added another router about 4' apart, without any change
- I wiped my network and paired them from scratch, it worked OK for a few hours, then the  issue came back
- I replaced my CC2531 (which has a BAD signal) with a Conbee II which as a WAY better signal, it worked OK for a few hours, then the issue came back

I have 2 of them currently installed but I have 2 others laying around.

One of them works, but it's near the coordinator (around 6' through one wall), while the other one that has the issue is ~ 16' away from the coordinator with 1 floor in the way. There's 4 routers (2x In-wall switches and 2x IKEA TRADFRI signal repeater)  between the two for it to works, and it worked for 6 months without issue...

When the sensor trigger, I see:
zigbee2mqtt:info  2020-07-01 21:14:03: MQTT publish: topic 'zigbee2mqtt/0x00155f00b44cf9b4', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":14,"temperature":24.16,"battery":100,"voltage":3100}'

The sensor is supposed to reset the occupancy after 5 seconds like this one that works:
zigbee2mqtt:info  2020-07-01 21:36:17: MQTT publish: topic 'zigbee2mqtt/0x00155f00b44cf9e5', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":15,"temperature":24.51,"battery":100,"voltage":3000}'
zigbee2mqtt:info  2020-07-01 21:36:21: MQTT publish: topic 'zigbee2mqtt/0x00155f00b44cf9e5', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":15,"temperature":24.51,"battery":100,"voltage":3000}'

Thanks for your time and help!

Koenkk, 2020-07-03:
Since you have a spare CC2531, can you provide a sniff of both the working and broken one (at the time your trigger occupancy). https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

joe540, 2020-07-22:
I think my issue was similar to #3842 as I had a weak direct connection from the motion sensor to the coordinator. The sensor didn't sent the message using the routers near it.

 I since moved my coordinator to a less central location but nearer those Bosch sensors (as my other devices seems to use routers correctly) and it resolved the issue

closed 2020-07-22 by joe540