Koenkk/zigbee2mqtt #3840: Error in logs since version 1.14.1

McGiverGim, 2020-07-02:
# Bug Report
## What happened
Each time I restart the zigbee2mqtt addon in Home Assistant, I see some errors in the logs. It only happens since the upgrade to 1.14.1. It seems it is affecting to the Xiaomi GZCGQ01LM light intensity sensor, and I have seen that it has been a change about it in the latest update https://github.com/Koenkk/zigbee-herdsman-converters/pull/1293 but I don't know if it is related.

```txt
zigbee2mqtt:error 2020-07-02 09:14:43: Failed to configure 'luminosidad_comedor', attempt 1 (Error: Bind 0x04cf8cdf3c78c831/1 genPowerCfg from '0x00124b0018e86d21/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.14.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

The sensor seems to be working, so is not a fatal problem.

## What did you expect to happen
A clean start without errors in the logs.

## How to reproduce it (minimal and precise)
In my case, I simply need to restart the zigbee2mqtt addon in Home Assistant.

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531
Adapter firmware version: zStack 12-20190608

nanobra1n, 2020-07-02:
I see the same errors starting after the update to 1.14.1

Zigbee2mqtt version: 1.14.1
Adapter hardware: CC1352P-2
Adapter firmware version:{'type': 'zStack3x0', 'meta': {'transportrev': 2, 'product': 1, 'majorrel': 2, 'minorrel': 7, 'maintrel': 1, 'revision': 20200417}}

Koenkk, 2020-07-03:
In https://github.com/Koenkk/zigbee-herdsman-converters/pull/1293 a fix for the spamming behavior of this device was pushed. In order to apply it the device has to be reconfigured **once** succesfully.

As the device is normally sleeping this usually fails. To workaround it use https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure to force it and right before it wake the device up by clicking a button on it.

FedeBZ, 2020-07-03:
> In [Koenkk/zigbee-herdsman-converters#1293](https://github.com/Koenkk/zigbee-herdsman-converters/pull/1293) a fix for the spamming behavior of this device was pushed. In order to apply it the device has to be reconfigured **once** succesfully.
> 
> As the device is normally sleeping this usually fails. To workaround it use https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure to force it and right before it wake the device up by clicking a button on it.

Thanks, it worked

D4NH, 2020-07-31:
Are these fixed in 1.14.2?

closed 2020-09-08 by github-actions[bot]