Koenkk/zigbee2mqtt #3842: Osram Lightbulb prefers weak direct connection to coordinator instead of strong router connection and thus fails

JsBergbau, 2020-07-02:
# Bug Report
## What happened
Osram Lightify Bulb sometimes works and sometimes not. It seems that it prefers the very weak direct connection to the coordinator instead of the strong router connection. Coordinator is behind a thick stone wall. As soon as I hold it 30 cm more left in front of the door the lightbulb reacts immediately to my commands. So I'm very sure the issue arises because the weak direct connection is prefered. I did this 3 times when the lightbulb didn't react. All times then it worked. 
As you can see I have a lot other devices in the network. There are no problems at all with these other devices. Perhaps you can tell why somedevices aren't connected at all? I have these many routers only since yesterday, before I had only "Router1" which worked flawlessly.
2 Meters away from the bulb is a Aqara Motion sensor. This one works very reliable.head l
## What did you expect to happen
I expect the connection to use the good connection via "Router2".
## How to reproduce it (minimal and precise)
Hard to say how to rebuild this setup, but I would use a Lightify bulb together with a Router which is in between the coordinator and the bulb. Then place the bulb this way that there is only a very strong signal between bulb and coordinator, but a very good connection between bulb and router and router and coordinator.
## Debug Info
Zigbee2mqtt version:   1.14.0 (commit #9009de2)
Adapter hardware: CC2531 
Adapter firmware version: CC2531_20190425.zip and for Router  CC2531_router_2019_02.zip 

[image]

Koenkk, 2020-07-03:
Zigbee2mqtt cannot influence how devices connect to each other (we cannot force a device to use a certain router). Putting a good router very close to it may fix your issue.

Regarding the disconnected devices, as these are all battery powered ones this is normal (as they are sleeping a lot the router can forget it is the parent of those, leading in them to be not connected on the networkmap as the parent is unknown at the time of the scan).

JsBergbau, 2020-07-04:
> Putting a good router very close to it may fix your issue.

So what is a good signal strength? On the map it had 50 link quality over the router and still prefered the very weak direct connection.

JsBergbau, 2020-07-04:
> Regarding the disconnected devices, as these are all battery powered ones this is normal (as they are sleeping a lot the router can forget it is the parent of those, leading in them to be not connected on the networkmap as the parent is unknown at the time of the scan).

However thats very strange. All devices send their battery status about once every hour, so they should be connected in the map. I've just tried with one AqaraButton not showing any connection by pressing the button a few times, but still not showing any connection. In the log the button press is recognised. Another device didn't respond for more than 10 days already (I've removed the battery) and is still showing connected.

JsBergbau, 2020-07-09:
I've made a kind of ping, sending every second a message to `zigbee2mqtt/OsramBirne/get` to see when the device is not respoding. Plus I pulled the CC2531 repeater and the Philips Living Colors lamp, and setup a IKEA Tradfri Repeater. This device has a much better coverage/link quality then the CC2531 Router with external antenna

[image]
[image]

At about 6:00 only Tradfri Repeater was in the setup. You can see a much better link quality. Interesting point is, also if I "ping" only once per second, sometimes there are over 6000 replies. But even when stable I head about 3700 replies instead of 3600 (or less) replies as expected. Since there is an Aqara light sensor connected to the Tradfri Repeater and 
it was constantly sending light conditions I know link between coordindator and Repeater was fine. 

So whats going on? Why is the extremly weak direct link sometimes prefered and thus the bulb isn't responding at all?

JsBergbau, 2020-07-10:
There was this issue https://github.com/Koenkk/zigbee2mqtt/issues/1408 with source routing firmware of ZStack 1.2
Could it be possible that there is a simular issue in the Z-Stack_3.0.x version?

Koenkk, 2020-07-10:
#1408 only related to end devices in combination with a source routing firmware (in your case its a router + non source routing firmware). 

Seeing your networkmap, if I understand your correctly, the commands are send directly instead of via the TRADFRI repeater?

JsBergbau, 2020-07-12:
> Seeing your networkmap, if I understand your correctly, the commands are send directly instead of via the TRADFRI repeater?

Yes or the other way round. Commands are sent via Repeater but Osram bulb expects it directly from Coordinator. As you can see there is another device in the map connected to the osram bulb. This device is also inacessible when the bulb is inaccessible. This should be obvious because of how Zigbee works, but just wanted to mention it.

keezzzz, 2020-07-21:
Same problem here with Osram GardenPole. Whatever I do (delete and pair, re-pair) and from what location (inside close to routers, outside close to routers) de lighting set will only connect directly to the coordinator. An OTA did not change the behaviour.

JsBergbau, 2020-08-21:
Issue still exists....

JsBergbau, 2020-09-22:
Issue still exists

Koenkk, 2020-09-22:
@JsBergbau to get more insight what is happening, could you sniff the traffic of this issue? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2020-09-30 by github-actions[bot]