Koenkk/zigbee2mqtt #3844: RTCGQ11LM illuminance calculated right?

Ulli2k, 2020-07-03:
I sucessfully use the RTCQ... since month.
The issue why I created that ticket is that the illumination value was never been plausible.
That means it looks like to be calibrated at very low values.
e.g at daylight I have very low values <50 and therefor I reach pretty early during the day the value 0.
That behavior does not allow to control my light based on illumination.

I recognized an other ticket which is about the new sensor GZCGQ01LM and how to calculate the illumination value.
https://github.com/Koenkk/zigbee2mqtt/issues/2702

I am thinking about if the calculation also need to be added for the noted motion sensor?
{ (10**(states('sensor.kitchen_illuminance_raw') | int / 10000) - 1) | round }}

How are your experience with the motion sensor and the illumination value?

Koenkk, 2020-07-03:
- What zigbee2mqtt version are you using?
- Can you provide the debug logging when receiving illuminance values from the device in full day light?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Ulli2k, 2020-07-05:
in Debug mode I get the following messages:
`
zigbee2mqtt:debug 2020-07-05 10:23:20: Received Zigbee message from '0x00158d000305965b', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":92}' from endpoint 1 with groupID 0

zigbee2mqtt:info  2020-07-05 10:23:20: MQTT publish: topic 'zigbee2mqtt/0x00158d000305965b', payload '{"battery":100,"voltage":3075,"illuminance":92,"linkquality":31,"occupancy":true,"illuminance_lux":92}'

zigbee2mqtt:debug 2020-07-05 10:23:20: Received Zigbee message from '0x00158d000305965b', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0

zigbee2mqtt:info  2020-07-05 10:23:20: MQTT publish: topic 'zigbee2mqtt/0x00158d000305965b', payload '{"battery":100,"voltage":3075,"illuminance":92,"linkquality":31,"occupancy":true,"illuminance_lux":92}'
`

The zigbee2mqtt verison is 1.14.1
firmware 20190608 & zStack12

closed 2020-08-17 by github-actions[bot]