Koenkk/zigbee2mqtt #3845: Z2M service not starting on RPi 4b but works good in a terminal

helsdingen, 2020-07-03:
I'm new with Z2M and after using some installation instructions I got it running on my Raspberry PI 4b. The weird thing is that when I start Z2M in a terminal (using pi user) with "npm start" it works great and my devices can be added in Domoticz. However, when I start it as a service of after a reboot it won't start.
This is the npm log:
`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.5
3 info using node@v12.18.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.14.1~prestart: zigbee2mqtt@1.14.1
6 info lifecycle zigbee2mqtt@1.14.1~start: zigbee2mqtt@1.14.1
7 verbose lifecycle zigbee2mqtt@1.14.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.14.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.14.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.14.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.14.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.14.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.14.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.14.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.118-v7l+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.18.2
19 verbose npm  v6.14.5
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.14.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.14.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

And this is the Z2M log where you can see it was working until 13:40 when I stopped the terminal session and started it as a service using "sudo systemctl start zigbee2mqtt.service"

`info  2020-07-03 13:40:37: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-07-03.13-40-37' filename: log.txt
debug 2020-07-03 13:40:37: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-07-02.18-32-31'
debug 2020-07-03 13:40:38: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2020-07-03 13:40:38: Starting zigbee2mqtt version 1.14.1 (commit #7e46435)
info  2020-07-03 13:40:38: Starting zigbee-herdsman...
debug 2020-07-03 13:40:38: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM-Zigbee"},"adapter":{"concurrent":null}}'
info  2020-07-03 13:40:40: zigbee-herdsman started
info  2020-07-03 13:40:40: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20181024}}'
debug 2020-07-03 13:40:40: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-07-03 13:40:40: Currently 2 devices are joined:
info  2020-07-03 13:40:40: Monitor Switch (0x00158d000205a02c): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
info  2020-07-03 13:40:40: Temp Badkamer (0x00158d0002b51c78): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
warn  2020-07-03 13:40:40: `permit_join` set to  `true` in configuration.yaml.
warn  2020-07-03 13:40:40: Allowing new devices to join.
warn  2020-07-03 13:40:40: Set `permit_join` to `false` once you joined all devices.
info  2020-07-03 13:40:40: Zigbee: allowing new devices to join.
info  2020-07-03 13:40:40: Connecting to MQTT server at mqtt://localhost:1883
info  2020-07-03 13:40:40: Connected to MQTT server
info  2020-07-03 13:40:40: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-07-03 13:40:40: MQTT publish: topic 'zigbee2mqtt/Monitor Switch', payload '{"battery":100,"voltage":3015,"linkquality":13}'
info  2020-07-03 13:40:40: MQTT publish: topic 'zigbee2mqtt/Temp Badkamer', payload '{"battery":100,"voltage":3195,"temperature":22.42,"humidity":58.87,"pressure":1013,"linkquality":42}'
info  2020-07-03 13:40:40: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.1","commit":"7e46435","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20181024}},"log_level":"debug","permit_join":true}'
debug 2020-07-03 13:45:38: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2020-07-03 13:50:38: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2020-07-03 13:55:38: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2020-07-03 14:00:38: Saving state to file /opt/zigbee2mqtt/data/state.json`

So what am I missing?
Logs and configuration added as attachments for easy reading.
[_debug.log](https://github.com/Koenkk/zigbee2mqtt/files/4869592/_debug.log)
[_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4869593/_log.txt)
[_configuration.yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/4869598/_configuration.yaml.txt)

rynr, 2020-07-03:
I think I have the same problem.
Better to read:
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.14.5
3 info using node@v12.18.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.14.1~prestart: zigbee2mqtt@1.14.1
6 info lifecycle zigbee2mqtt@1.14.1~start: zigbee2mqtt@1.14.1
7 verbose lifecycle zigbee2mqtt@1.14.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.14.1~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/app/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.14.1~start: CWD: /app
10 silly lifecycle zigbee2mqtt@1.14.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.14.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.14.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.14.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.14.1
15 verbose cwd /app
16 verbose Linux 5.4.0-1012-raspi
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v12.18.1
19 verbose npm  v6.14.5
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.14.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.14.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

helsdingen, 2020-07-06:
I found the solution here: [https://zigbee2mqtt.discourse.group/t/zigbee2mqtt-not-starting/713/12](https://zigbee2mqtt.discourse.group/t/zigbee2mqtt-not-starting/713/12)

Somehow you need to make a change in this file: /etc/systemd/system/zigbee2mqtt.service

Change:
`ExecStart=/usr/bin/npm start`
To:
`ExecStart=/usr/local/bin/npm start`

I still don't know why because both directories refer to the same source: ../lib/node_modules/npm/bin/npm-cli.js
In both source directories (/usr/lib/node_modules/npm/bin & /usr/local/lib/node_modules/npm/bin) there are exactly the same files but somehow the /usr/lib isn't working and the /user/local/lib is.

closed 2020-08-16 by github-actions[bot]