Koenkk/zigbee2mqtt #3846: LifeControl MCLH-08 pairing

velom, 2020-07-03:
Hello!
Can you explain please, how should I pair MCLH-08 and z2m?
I click/hold the only one button, but there are no logs, `zigbee2mqtt/bridge/log` is empty too.
Other devices work correctly.

Koenkk, 2020-07-03:
@kirovilya do you know how to pair this device? Are there any channel restrictions? (e.g. 15, 20, etc.)

kirovilya, 2020-07-04:
No restrictions. Try to press button for 5-7 sec until blink, then release and press short again.

velom, 2020-07-04:
> No restrictions. Try to press button for 5-7 sec until blink, then release and press short again.

Still nothing happens =(
Any suggestions?

kirovilya, 2020-07-04:
Check battery.  There is some led indication/blink?

velom, 2020-07-04:
Finally I did some hits from check list https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair.
This helped me: "Stop Zigbee2mqtt, unplug the coordinator, wait 10 seconds, plug the coordinator, start Zigbee2mqtt and try to pair the device again."

My apologize for disturbing =(

closed 2020-07-04 by velom