Koenkk/zigbee2mqtt #3847: CC2530 + CC2591 USB support

Westcott1, 2020-07-03:
I have one of these that has a USB plug  -
https://www.aliexpress.com/item/4000060995530.html?spm=a2g0s.9042311.0.0.40694c4dsHFJuu

It's recognised correctly as a cc2530 by the TI programmer, and flashes OK with "CC2530_CC2591_router_2019_02\router-cc2530-cc2591-std.hex"

The RPi sees it as -
usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0

Z2M sees it as -

Device '0x00124b001fa0f3f4' is supported, identified as: Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)
zigbee2mqtt:info  2020-07-03 17:56:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001fa0f3f4","model":"CC2530.ROUTER","vendor":"Custom devices (DiY)","description":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/)","supported":true}}'

Every minute I see -
zigbee2mqtt/0x00124b001fa0f3f4', payload '{"led":false,"linkquality":118}

However, if I plug it into AMC1 instead of my trusty cc2531, Z2M does not see it.
Please, how do I get Z2M to use it as a coordinator?
Thanks!

MPM1107, 2020-07-03:
Well you'll have to decide what you want to do with it. You said you flashed it with router firmware, now it's behaving like a simple router. If you want to use it as a coordinator you will have to flash it with coordinator firmware. It can't do both at the same time.

Westcott1, 2020-07-03:
Thanks, I was wondering that as well.
Which firmware do you recommend?

MPM1107, 2020-07-03:
Go with source routing if you have 30+ devices and plenty of routers, otherwise default [https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin)

Westcott1, 2020-07-03:
Thanks.
Flashed with CC2530_CC2591_SOURCE_ROUTING_20190619\CC2530ZNP-Prod.hex
The RPi thinks it is on USB1 (not ACM1) so I put that in the Config
Z2M -
Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

Westcott1, 2020-07-03:
I think the problem is that it is not an ACM-type device, so Z2M can't talk to it.

MPM1107, 2020-07-03:
That is just the name of the serial port and has no effect on zigbee2mqtt. Did you set rtscts to false in the zigbee2mqtt config?

Westcott1, 2020-07-04:
Yes, that did it!
Thanks for all your help.

closed 2020-07-04 by Westcott1