Koenkk/zigbee2mqtt #3849: Support for Niko Connected Socket

TheMClaeys, 2020-07-04:
Hi,

I am trying to support the [Niko Connected Socket](https://webshop.electro-colli.be/product/niko-170-33505-geconnecteerde-schakelbare-wandcontactdoos-niko-home-control-penaarde-en-kinderveiligheid-zigbee-16-a250-vac-insteekklemmen/). It looks like a normal socket but supports on/off and power measuring. This is currently an unsupported device.

Thanks to the BlitzWolf topic I think I came a long way already, I'm just not there yet.

I added the following (with some trial and error) to devices.js
```
// Niko Connected Socket
    {
        zigbeeModel: ['Connected socket outlet'],
        model: '170-33505',
        vendor: 'Niko',
        description: 'Niko Connected Socket',
        supports: 'on/off, powermeasurement',
        fromZigbee: [fz.on_off, fz.Niko_Power],
        toZigbee: [tz.on_off],
        meta: {configureKey: 4},
        configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            await readEletricalMeasurementPowerConverterAttributes(endpoint);
        },
        onEvent: async (type, data, device) => {
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(store[device.ieeeAddr]);
            } else if (!store[device.ieeeAddr]) {
                store[device.ieeeAddr] = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {
                        // Do nothing
                    }
                }, 10*1000); // Every 10 seconds
            }
        },
    },

```
I first had the error that no parser was available and I fixed that with the fz.Niko_Power.

In Niko_Power I added the following (I made a new one instead of reusing one because the numbers needed to be divided by 10).
```
    Niko_Power: {
        cluster: 'haElectricalMeasurement',
        type: 'readResponse',
        convert: (model, msg, publish, options) => {
            const voltage = parseFloat(msg.data['rmsVoltage']) / 10.0;
            const current = parseFloat(msg.data['rmsCurrent']) / 10.0;
            const power = parseFloat(msg.data['activePower']) / 10.0;
            return {Voltage: voltage, Current: current, Power: power};
        },
    },

```

I added the following to homeassistant.js
`    '170-33505': [cfg.switch, cfg.sensor_voltage, cfg.sensor_current, cfg.sensor_power],`

I seem to get the correct values out of the device this way. The measurements are also visible in Home Assistant, just not as a state. That used to work at some point, but after messing around some more it didn't anymore.
[image]

So I am currently stuck at a few points.
- As long as the device is not in states.json, it does not report it's initial state. [As soon as the state changes it does](https://hastebin.com/araxazijid.js). If it is in that file, it does show the default state, just the one from the file, not the device. I think in the beginning when I still thought the device didn't support power measuring over Zigbee [it did report its initial state](https://hastebin.com/gujeqaleti.js).
- Apparently all data is sent correctly to Home Assistant, but doesn't appear in the State field anymore what it did before. I don't know what I broke.
- I only get a bit of the basics of coding, so it was a lot of copy pasting and therefore I think it can probably be a lot cleaner or shorter.
- In the beginning I got some "[failed to configure](https://hastebin.com/cifarumanu.js)" errors. I don't get these anymore. I don't know if it's because it's solved or because it gave up. (I found another topic with a link on how to force config again, but I wasn't able to figure out how to do it).

Can someone assist with this?

What does work without issues:
- turning the socket on and off
- measuring the consumption of the devices (I'm not sure yet if 10 seconds is a good polling interval)
It didn't report anything in the beginning, so I am happy I got this far. I'm currently still messing around, so if I happen to find it out on my own I'll update this post also.

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531
Adapter firmware version: `Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'`

MPM1107, 2020-07-04:
For starters, maybe try changing the object keys in your fromZigbee definition to all lowercase: `return {Voltage: voltage, Current: current, Power: power};`, I think that may be messing with your homeassistant states.
So it should be: `return {voltage: voltage, current: current, power: power};`

TheMClaeys, 2020-07-04:
Wel, that fixed the Home Assistant thing apparently. Thanks!
I thought it was only important for the variables itself. I remember changing that yesterday, didn't link it to the numbers no longer showing up.

Koenkk, 2020-07-04:
- With the definition below the `onEvent` shouldn't be needed anymore because the device is probably able to report the values themselves (so we don't have to poll).
- Strange that the default power divisors don't work, can you provide me the `data/database.db` entry, a debug log when receiving power messages and the number of watts of the attached load? (in combination with the definition below ofcourse)

```js
{
    zigbeeModel: ['Connected socket outlet'],
    model: '170-33505',
    vendor: 'Niko',
    description: 'Niko Connected Socket',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 5},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.activePower(endpoint);
        await configureReporting.rmsCurrent(endpoint);
        await configureReporting.rmsVoltage(endpoint);
    },
},
```

TheMClaeys, 2020-07-04:
Well, that made it less complex. Changed it to the code above and apparently it still updates (everything except the initial on/off state).

[database.db](https://hastebin.com/nudikozucu.json)-entry
[debug logs](https://hastebin.com/xoyonikobo.js)

Initially I got the data below, that's why I tried the /10 to get the coma right. That was with fz.metering_power though,  and even the fz.ignore_basic_report was still in the code then (I started with a full copy of the other topic and removed/changed till it started working).
[image]

This means then to support this, no changes would be needed to the fromZigbee file, am I reading that correct?

Koenkk, 2020-07-05:
@TheMClaeys so if I understand correctly, with the code from https://github.com/Koenkk/zigbee2mqtt/issues/3849#issuecomment-653771194 it works? Event the "comma is right" of the power values?

TheMClaeys, 2020-07-05:
Hi @Koenkk , it looks like it is, I want to test it with a more heavy device to confirm definitely, hadn't had time for that yet. 
I also ordered another one to see if it gets recognized correctly and works too, if it does I'll look into adding it for native support.

Thanks for the assistance in simplifying it a lot, apparently I was looking at it all wrong.

sjorge, 2020-07-07:
Where did you order from?

TheMClaeys, 2020-07-07:
[Gigatek](https://gigatek.be/nl/niko-geconn-wcd-penaarde-inst--nik170-33505) + ([front](https://gigatek.be/nl/niko-afwerk-set-geconn-wcd--nik101-66630).

And the first one is placed on the washing machine:
[image]

TheMClaeys, 2020-07-08:
The new socket joined without issues. I'll try to add it to native support soon. Thanks for all the help.

closed 2020-07-08 by TheMClaeys

sjorge, 2020-07-08:
> The new socket joined without issues. I'll try to add it to native support soon. Thanks for all the help.

Was this with reporting of on/off state?

TheMClaeys, 2020-07-08:
If you click on the button once it does immediately, if you don't it apparently does also, just takes a small bit longer. The same with power and current, it does not report it when you link it, it does as soon as you plug something in. After that it keeps reporting it. Voltage and link quality it reports from the start.

Sennevds, 2020-09-25:
@TheMClaeys I can't connect the socket to z2m. I recently changed from deconz to z2m.
I've reset the device and get following messages in my log:
```log
info  2020-09-25 12:00:49: Device '0x086bd7fffeedfd4a' joined
info  2020-09-25 12:00:49: Starting interview of '0x086bd7fffeedfd4a'
error 2020-09-25 12:02:50: Failed to interview '0x086bd7fffeedfd4a', device has not successfully been paired
```
Anything I'm doing wrong?

TheMClaeys, 2020-09-25:
Hi Sennevds,

Do you have the latest Zigbee2MQTT update?

I added a 3rd one yesterday that arrived earlier this week. These were the logs:
`info  2020-09-24 19:16:52: Zigbee: allowing new devices to join.
info  2020-09-24 19:16:53: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.14.4","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.14.4"}'
info  2020-09-24 19:17:17: Device '0x086bd7fffeedfc22' joined
info  2020-09-24 19:17:17: Starting interview of '0x086bd7fffeedfc22'
info  2020-09-24 19:17:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x086bd7fffeedfc22"},"type":"device_connected"}'
info  2020-09-24 19:17:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x086bd7fffeedfc22"},"type":"pairing"}'
info  2020-09-24 19:17:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x086bd7fffeedfc22"},"type":"device_announced"}'
info  2020-09-24 19:17:31: MQTT publish: topic 'homeassistant/switch/0x086bd7fffeedfc22/switch/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","command_topic":"zigbee2mqtt/0x086bd7fffeedfc22/set","device":{"identifiers":["zigbee2mqtt_0x086bd7fffeedfc22"],"manufacturer":"Niko","model":"Connected socket outlet (170-33505)","name":"0x086bd7fffeedfc22","sw_version":"Zigbee2MQTT 1.14.4"},"json_attributes_topic":"zigbee2mqtt/0x086bd7fffeedfc22","name":"0x086bd7fffeedfc22_switch","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x086bd7fffeedfc22","unique_id":"0x086bd7fffeedfc22_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
info  2020-09-24 19:17:31: MQTT publish: topic 'homeassistant/sensor/0x086bd7fffeedfc22/voltage/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x086bd7fffeedfc22"],"manufacturer":"Niko","model":"Connected socket outlet (170-33505)","name":"0x086bd7fffeedfc22","sw_version":"Zigbee2MQTT 1.14.4"},"icon":"mdi:alpha-v","json_attributes_topic":"zigbee2mqtt/0x086bd7fffeedfc22","name":"0x086bd7fffeedfc22_voltage","state_topic":"zigbee2mqtt/0x086bd7fffeedfc22","unique_id":"0x086bd7fffeedfc22_voltage_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
info  2020-09-24 19:17:31: MQTT publish: topic 'homeassistant/sensor/0x086bd7fffeedfc22/current/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x086bd7fffeedfc22"],"manufacturer":"Niko","model":"Connected socket outlet (170-33505)","name":"0x086bd7fffeedfc22","sw_version":"Zigbee2MQTT 1.14.4"},"icon":"mdi:current-ac","json_attributes_topic":"zigbee2mqtt/0x086bd7fffeedfc22","name":"0x086bd7fffeedfc22_current","state_topic":"zigbee2mqtt/0x086bd7fffeedfc22","unique_id":"0x086bd7fffeedfc22_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
info  2020-09-24 19:17:31: MQTT publish: topic 'homeassistant/sensor/0x086bd7fffeedfc22/power/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x086bd7fffeedfc22"],"manufacturer":"Niko","model":"Connected socket outlet (170-33505)","name":"0x086bd7fffeedfc22","sw_version":"Zigbee2MQTT 1.14.4"},"icon":"mdi:flash","json_attributes_topic":"zigbee2mqtt/0x086bd7fffeedfc22","name":"0x086bd7fffeedfc22_power","state_topic":"zigbee2mqtt/0x086bd7fffeedfc22","unique_id":"0x086bd7fffeedfc22_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
info  2020-09-24 19:17:31: MQTT publish: topic 'homeassistant/sensor/0x086bd7fffeedfc22/linkquality/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x086bd7fffeedfc22"],"manufacturer":"Niko","model":"Connected socket outlet (170-33505)","name":"0x086bd7fffeedfc22","sw_version":"Zigbee2MQTT 1.14.4"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x086bd7fffeedfc22","name":"0x086bd7fffeedfc22_linkquality","state_topic":"zigbee2mqtt/0x086bd7fffeedfc22","unique_id":"0x086bd7fffeedfc22_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2020-09-24 19:17:32: Successfully interviewed '0x086bd7fffeedfc22', device has successfully been paired
info  2020-09-24 19:17:32: Device '0x086bd7fffeedfc22' is supported, identified as: Niko Connected socket outlet (170-33505)
info  2020-09-24 19:17:32: Configuring '0x086bd7fffeedfc22'
info  2020-09-24 19:17:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Connected socket outlet","friendly_name":"0x086bd7fffeedfc22","model":"170-33505","supported":true,"vendor":"Niko"},"type":"pairing"}'
info  2020-09-24 19:17:32: Successfully configured '0x086bd7fffeedfc22'
info  2020-09-24 19:17:32: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffeedfc22', payload '{"linkquality":134,"voltage":230.3}'
info  2020-09-24 19:17:34: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffeedfc22', payload '{"linkquality":120,"power":0,"voltage":230.3}'`

Sennevds, 2020-09-25:
Just updated to dev and then it works. Think it has something to do with the fact I'm using a conbee adapter.

kennethhaspeel, 2020-11-15:
My apologies for re-opening this but i'm having trouble adding the sockets in Zigbee2mqtt. I have three of them and i'm not able to join any of them.
Is there a special way of adding them. I followed the instructions about resetting them, zigbee2mqtt is allowing pairing. But nothing happens. No error messages in the log except once when it said 'failed interview' with one of the sockets.
I'm about 1m away from the antenna so link quality is no issue. I have a innr smart plugs, led strips, motion detector,... so i assume everything is working like it should.
I'm on version 1.16.1

TheMClaeys, 2020-11-17:
@kennethhaspeel 
What Zigbee2MQTT stick do you use? I could rejoin mine from where they are placed in the socket (about 7 meters from the antenna).
However, I had a similar issue when I had the CC2531. The issue occurred when I joined my 3rd socket. I think that the sockets send out so many that the stick can't handle it. I have now a CC2652RB stick and they don't lose connection with the stick anymore. I still have the issue that some devices don't send info anymore, I think it's the same issue, that they send too many info so the smaller devices go unnoticed. I still have to open a case for this here to see if I can throttle the data in some way but I haven't got the time yet sadly.
Also: be sure to reset them to factory defaults before joining, most Zigbee devices can just be rejoined by putting them in pairing mode, the Niko sockets not, that's by Niko's design and not a Zigbee2MQTT issue.

kennethhaspeel, 2020-11-17:
@TheMClaeys Thanks for your answer. I'm using a CC2531 stick. I was planning to upgrade to a more performing stick because i want to replace my Zwave and Wifi devices to zigbee, as much as possible.

About factory defaults. I was thinking as much because the Niko documentation only mentions reset procedure and no pairing mode.

Maybe starting with another stick wouldn't be a bad start. No point adding them to this stick, just to change it in a couple of weeks

Thanks

fluppie, 2020-12-23:
Hi all, how did you pair the Niko outlet 170-33505 to Z2M? Myself I use a Slaesh Zigbee stick.
In the manual there is only a procedure that shows how to reset the device.
https://www.niko.eu/-/downloadfile/be3a30b8aa924f1ebd22abb900e61c86

Sennevds, 2020-12-23:
> Hi all, how did you pair the Niko outlet 170-33505 to Z2M? Myself I use a Slaesh Zigbee stick.
> In the manual there is only a procedure that shows how to reset the device.
> https://www.niko.eu/-/downloadfile/be3a30b8aa924f1ebd22abb900e61c86

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2984#issuecomment-653553600

fluppie, 2020-12-23:
Jup, just found it... 

_http://guide.niko.eu/display/HGNHC2V2/Manual+use
Als je binnen zes seconden 3x of meer op de ledtoets drukt, gaat de geconnecteerde wandcontactdoos max. 300 sec. in verbindingsmodus (led knippert blauw). Dit proces kan niet onderbroken worden door de gebruiker._

So you need to press the button at least 3 times within 6 seconds. 

But damn, this thing is flooding the network

Sennevds, 2020-12-23:
Yeps you get many updates on power consumption. But it works really nice. (I'm using it for my washing machine)

fluppie, 2020-12-23:
Think we should limit the rate in some way?

> 
> Zigbee2MQTT:info  2020-12-23 17:41:00: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":18.98,"state":"ON","voltage":219.9}'
> Zigbee2MQTT:info  2020-12-23 17:41:04: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":18.98,"state":"ON","voltage":220.3}'
> Zigbee2MQTT:info  2020-12-23 17:41:08: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":30,"power":19.09,"state":"ON","voltage":220.3}'
> Zigbee2MQTT:info  2020-12-23 17:41:08: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":30,"power":19.09,"state":"ON","voltage":220.8}'
> Zigbee2MQTT:info  2020-12-23 17:41:15: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":19.09,"state":"ON","voltage":220.5}'
> Zigbee2MQTT:info  2020-12-23 17:41:16: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":18.64,"state":"ON","voltage":220.5}'
> Zigbee2MQTT:info  2020-12-23 17:41:17: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":30,"power":18.75,"state":"ON","voltage":220.5}'
> Zigbee2MQTT:info  2020-12-23 17:41:19: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":18.75,"state":"ON","voltage":221}'
> Zigbee2MQTT:info  2020-12-23 17:41:21: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":19.2,"state":"ON","voltage":221}'
> Zigbee2MQTT:info  2020-12-23 17:41:24: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":30,"power":19.2,"state":"ON","voltage":221.2}'
> Zigbee2MQTT:info  2020-12-23 17:41:27: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":19.32,"state":"ON","voltage":221.2}'
> Zigbee2MQTT:info  2020-12-23 17:41:29: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":19.32,"state":"ON","voltage":221.2}'
> Zigbee2MQTT:info  2020-12-23 17:41:29: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":19.32,"state":"ON","voltage":221.1}'
> Zigbee2MQTT:info  2020-12-23 17:41:31: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":19.2,"state":"ON","voltage":221.1}'
> Zigbee2MQTT:info  2020-12-23 17:41:33: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":19.2,"state":"ON","voltage":221.2}'
> Zigbee2MQTT:info  2020-12-23 17:41:34: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":19.2,"state":"ON","voltage":221.2}'
> Zigbee2MQTT:info  2020-12-23 17:41:36: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":30,"power":18.98,"state":"ON","voltage":221.2}'
> Zigbee2MQTT:info  2020-12-23 17:41:37: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":18.98,"state":"ON","voltage":221.3}'
> Zigbee2MQTT:info  2020-12-23 17:41:41: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffebd40a4', payload '{"current":0.12,"linkquality":33,"power":18.64,"state":"ON","voltage":221.3}'

Sennevds, 2020-12-23:
Everything works fine for me even with the amount of messages but I don't have that many devices.

fluppie, 2020-12-23:
This is my first socket, but I can imagine if you have multiple this can cause problems.

TheMClaeys, 2020-12-24:
I have currently 3 of them (washing machine, dryer, koffee machine) and they indeed flood the network with a lot of data. As long as it works I don't mind though. After my 3rd I upgraded to a Zigbee2MQTT stick though as I had the impression that the default one I used couldn't handle it anymore.
With the current one I have the impression everything works good again.
Yesterday I linked 6 new Aqara devices and all still seems to work fine. Only 2 sensors who are too far away, so it's tempting to buy another Niko socket now.
I only don't know how many will be too much for the stick, I can't seem to find how you can see the saturation of the network.

kennethhaspeel, 2020-12-28:
I switched to the [Electrolama CC2652R](https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/) and the sockets connected without a problem. But i also noticed right away, the sockets are a bit to active to my taste. At the moment I have two sockets installed and they send updates about once a second on average, more if the socket is in use. I only have 15 devices in my network so it's not a problem at the moment. 
But i'm guessing it is not possible to adjust the frequency it's reporting

TheMClaeys, 2020-12-28:
The Niko Hub supports up to 20 or 25 connected sockets and up to 50 Zigbee devices, so my guess is if your stick supports at least 50 devices that it shouldn't be an issue. Currently I don't have issues with 3, I thought I had issues with some other sensors but it was because I used other names when I connected them to the new stick, so totally user error.
It does cluther the logs though.

fluppie, 2020-12-28:
I also guess the stick can handle it (also using a CC2652RB), but I don't want the MQTT bus flooded with these messages.
In node-red there is a delay node that can drop messages, so it's actually a rate limiter. Maybe Zigbee2MQTT can integrate a rate limiter in the software for "trigger happy" sensors like this Niko wall outlet.
https://github.com/node-red/node-red/blob/master/packages/node_modules/%40node-red/nodes/core/function/89-delay.js

sjorge, 2020-12-28:
You can see debounce for the device. It should merge the messages into just one.

bucofski, 2021-08-19:
hi,

I brook my stick and i replaced it with succes. After this the niko socket is still red but i can operate it and doesn't sent updates. After excluding and including again one socket the red led is gone but i can't see any updates from current power or voltage. If i ask for a state update from current it shows a new power update if I ask a voltage update I get this fault : No converter available for 'voltage' ().  Before I brook my stick it sends regularly messages. Now only when the current changes or when the state (on or off) changes. the firmware that is on the stick is 20201127 an use rpi4b.

xHuubx, 2022-10-10:
@TheMClaeys I really liked your post and I want to give it a try. Would it be possible that you create a step to step instruction set in order to get this working with Home Assistant in combination with Conbee II and zigbee2mqtt?