Koenkk/zigbee2mqtt #3850: unhandled exception for mgmtPermitJoinReq

Kosta-Github, 2020-07-04:
# Bug Report

After an unhandled exception all zigbee devices stopped working.

## What happened

After receiving an unhandled exception (see logs below) all zigbee devices stopped working.

```
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:05: Saving state to file /app/data/state.json
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_wohnzimmer_eat' was last seen '1151.326' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_wohnzimmer_coach' was last seen '129.221' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_floor_0_entrance' was last seen '1180.83' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_bad_eg' was last seen '183.149' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_floor_0_stairs' was last seen '889.198' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_kitchen_window' was last seen '1863.592' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_sleep_2' was last seen '1181.293' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_kitchen_durchgang' was last seen '332.147' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_sleep_1' was last seen '148.011' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_jojo' was last seen '1176.775' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_wohnzimmer_tv' was last seen '1304.975' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_floor_1' was last seen '74.476' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_bad_2_og' was last seen '4754.9' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_fitness' was last seen '2529.748' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'remote_1' was last seen '12700.561' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_bad_1_og' was last seen '1302.189' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:06: Non-pingable device 'motion_guest' was last seen '2033.884' seconds ago.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:10: Failed to ping 'plug_floor_1'
(node:16) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:47:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 8)
(node:16) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:22: Failed to ping 'plug_floor_2'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:28: Failed to ping 'plug_coach'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:45:34: Failed to ping 'plug_floor_0'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:46:16: Failed to ping 'plug_floor_1'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:46:28: Failed to ping 'plug_floor_2'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:46:34: Failed to ping 'plug_coach'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:46:40: Failed to ping 'plug_floor_0'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:47:22: Failed to ping 'plug_floor_1'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:47:34: Failed to ping 'plug_floor_2'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:47:40: Failed to ping 'plug_coach'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:47:46: Failed to ping 'plug_floor_0'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:48:28: Failed to ping 'plug_floor_1'
(node:16) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:47:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 9)
[34mzigbee2mqtt:debug[39m 2020-07-03 23:48:41: Failed to ping 'plug_floor_2'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:48:47: Failed to ping 'plug_coach'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:48:53: Failed to ping 'plug_floor_0'
[34mzigbee2mqtt:debug[39m 2020-07-03 23:49:34: Failed to ping 'plug_floor_1'
...
```

From that time on the logs are just repeating this for >12 hours without recovering into a functional state again. I had to restart the system to recover from this (I haven't tried to just restart the docker container running `z2m`).

Sorry, but I haven't enabled more logging in the herdsman driver (since I don't want to overwhelm my raspi with too much logging)

## What did you expect to happen

That the system recovers automatically from that state.

## How to reproduce it (minimal and precise)

I don't know how to make this reproducible.

## Debug Info

Zigbee2mqtt version: 
```
{
  "version": "1.14.0",
  "commit": "9009de2",
  "coordinator": {
    "type": "zStack12",
    "meta": {
      "transportrev": 2,
      "product": 0,
      "majorrel": 2,
      "minorrel": 6,
      "maintrel": 3,
      "revision": 20190608
    }
  },
  "log_level": "debug",
  "permit_join": true
}
```

Adapter hardware: `CC2531`

Adapter firmware version: `20190608`

MPM1107, 2020-07-04:
Pretty hard to prove anything without knowing how to reproduce this but it almost sounds like the coordinator/adapter crashed or got into a weird state, since it stopped responding to zigbee2mqtt. If it happens again try just restarting z2m like you said.

Koenkk, 2020-07-04:
@MPM1107 the adapter crashed indeed.

@Kosta-Github note that the availability feature can be very heavy for the CC2531 (the more devices the bigger the chances of crashes) However since you are still on the default firmware, the source routing firmware will give more stability (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

fwelvering, 2020-07-05:
I have the same issue, just flashed to the suggested source_routing version, but getting the following issue when joining a device:

zigbee2mqtt:error 2020-07-05 10:46:33: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SRSP - AF - dataRequestExt after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.14.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:47:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

Chamondgl, 2020-07-12:
It seems I have the same problem.

zigbee2mqtt:error 2020-07-12 16:20:42: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.14.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:47:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

closed 2020-08-19 by github-actions[bot]