Koenkk/zigbee2mqtt #3855: Vimar 14594.0 Support

matteopiccioni, 2020-07-05:
Hello,
I'd like to add support for Vimar 14594.0 devices
https://www.vimar.com/en/int/catalog/product/index/code/14594.0

Its a view Wireless connected electronic control device for 1 roller shutter with slat orientation and change-over relay output
I was able to interview it and I add this in devices.js

```
    // Vimar 14594.0
    {
        zigbeeModel: ['Window_Covering_v1.0'],
        model: '14594.0',
        vendor: 'Vimar',
        description: 'Roller shutter with slat orientation and change-over relay',
        supports: 'up, down, rotation, favourite position recall',
        fromZigbee: [],
        toZigbee: [],
    },
```

In my database.db I hava

`{"id":9,"type":"Router","ieeeAddr":"0x0b010e85f075e991","nwkAddr":5939,"manufId":4660,"manufName":"Vimar","powerSource":"Mains (single phase)","modelId":"Window_Covering_v1.0","epList":[10,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":514,"inClusterList":[0,3,258,5,4],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"Window_Covering_v1.0","manufacturerName":"Vimar","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":""}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1593964603904}`

In log nothing is written while play with 14594.0 buttons

Thanks

Koenkk, 2020-07-06:
Can you try with:

```js
{
    zigbeeModel: ['Window_Covering_v1.0'],
    model: '14594.0',
    vendor: 'Vimar',
    description: 'Roller shutter with slat orientation and change-over relay',
    supports: 'open, close, stop, position',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await configureReporting.currentPositionLiftPercentage(endpoint);
    },
},
```

You should be able to control it with `zigbee2mqtt/0x0b010e85f075e991/set` payloads
- `{"state": "OPEN"}` 
- `{"state": "CLOSE"}`
- `{"position": 50}`

Do you also receive updates when controlling manually?

matteopiccioni, 2020-07-07:
Thanks
Still I cant connect my  14594.0 the real blinds
For my tests I connect a light to one of the two output and when I send the command 
`{"state": "CLOSE"}`
the light turn on
 
In log I have this message:
```
info  2020-07-07 22:13:03: Zigbee: allowing new devices to join.
info  2020-07-07 22:13:03: Configuring '0x0b010e85f075e991'
error 2020-07-07 22:13:03: Failed to configure '0x0b010e85f075e991', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:42:24)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:11299:19)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:104:45)
    at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:72:28)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:360:44))
info  2020-07-07 22:13:03: Connecting to MQTT server at mqtt://localhost
info  2020-07-07 22:13:03: Connected to MQTT server

```

Close and open appears to works
```
debug 2020-07-07 22:14:28: Received MQTT message on 'zigbee2mqtt/0x0b010e85f075e991/set' with data '{"state": "CLOSE"}'
debug 2020-07-07 22:14:28: Publishing 'set' 'state' to '0x0b010e85f075e991'
debug 2020-07-07 22:14:37: Received Zigbee message from 'Luce cucina', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 10 with groupID 0
info  2020-07-07 22:14:37: MQTT publish: topic 'zigbee2mqtt/Luce cucina', payload '{"state":"ON","linkquality":31}'
debug 2020-07-07 22:14:41: Received MQTT message on 'zigbee2mqtt/0x0b010e85f075e991/set' with data '{"state": "OPEN"}'
debug 2020-07-07 22:14:41: Publishing 'set' 'state' to '0x0b010e85f075e991'
debug 2020-07-07 22:15:24: Received MQTT message on 'zigbee2mqtt/0x0b010e85f075e991/set' with data '{"position": "50"}'
debug 2020-07-07 22:15:24: Publishing 'set' 'position' to '0x0b010e85f075e991'

```

If I control it manually I dont have feedback in log

I thinks that for extensive tests I have to wait to be able to mount the module and shutters and it may take several days
Thanks

Koenkk, 2020-07-07:
Can you try with:

```js
{
    zigbeeModel: ['Window_Covering_v1.0'],
    model: '14594.0',
    vendor: 'Vimar',
    description: 'Roller shutter with slat orientation and change-over relay',
    supports: 'open, close, stop, position',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(10);
        await bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await configureReporting.currentPositionLiftPercentage(endpoint);
    },
},
```

matteopiccioni, 2020-07-08:
Hello,
error messages on start disappears
```
info  2020-07-08 20:49:30: Successfully configured '0x0b010e85f075e991'
debug 2020-07-08 20:49:30: Received Zigbee message from '0x0b010e85f075e991', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionTiltPercentage":0,"currentPositionLiftPercentage":0}' from endpoint 10 with groupID 0
info  2020-07-08 20:49:31: MQTT publish: topic 'zigbee2mqtt/0x0b010e85f075e991', payload '{"position":100,"tilt":100,"linkquality":0}'
```

If I use it manually now I receive messages
```
debug 2020-07-08 20:59:51: Received Zigbee message from '0x0b010e85f075e991', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionTiltPercentage":0}' from endpoint 10 with groupID 0
info  2020-07-08 20:59:51: MQTT publish: topic 'zigbee2mqtt/0x0b010e85f075e991', payload '{"position":100,"tilt":100,"linkquality":0}'

```
set command works too
```
debug 2020-07-08 21:03:31: Publishing 'set' 'position' to '0x0b010e85f075e991'
debug 2020-07-08 21:03:33: Received Zigbee message from 'Luce cucina', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 10 with groupID 0
info  2020-07-08 21:03:33: MQTT publish: topic 'zigbee2mqtt/Luce cucina', payload '{"state":"ON","linkquality":36}'

```

Unfortunately, for now, I cant test on real blinds
I will give you a feedback as soon as possibiles (it is possible that it will take several days)
Thanks!

Clunku, 2020-08-12:
Hi Matteo, do you have any news? Do you use other vimar products? I need some information. Thanks

matteopiccioni, 2020-08-13:
Hi, still I cant use it in a real blinds, but I think it will works well with you configuration.
I am able to send OPEN, CLOSE and position messages without error and I am able to control it manually
In my opinion the configuration is right

Joseraj, 2020-09-10:
Hi,
I've just installed this device.
I've used this in devices.js:
```javascript
{
    zigbeeModel: ['Window_Covering_v1.0'],
    model: '14594.0',
    vendor: 'Vimar',
    description: 'Roller shutter with slat orientation and change-over relay',
    supports: 'open, close, stop, position, tilt',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    meta: {configureKey: 1, coverInverted: true},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(10);
        await bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await configureReporting.currentPositionLiftPercentage(endpoint);
    },
},
```
It seems to work.
"coverInverted: true" is necessary for OpenHab as it understands 0 = full open and 100% = full closed.
I use it with rollershutters, so I can't test "tilt" command or state.

It sends periodically at about every 15 seconds:
` zigbee2mqtt/[FRIENDLY_NAME]`
```javascript
{
   "device":{
      "applicationVersion":1,
      ...
   },
   "last_seen":15997.....
   "linkquality":34,
   "position":84,
   "tilt":0
}
```

My question regards the handle of this reports, I don't know if the source is the device or Z2M.

When not moving it reports the info avove every 15 seconds.

If I send a command
`zigbee2mqtt/[FRIENDLY_NAME]/set/state : "OPEN"   ( or "CLOSE" )`    
I receive the msg with the previous "position" value. (84)
Only some seconds after the end of the course the field "position" has a new value.

If I send a command
`zigbee2mqtt/[FRIENDLY_NAME]/set/position : "10"`
I see a msg, but "position" field has the value of the command, (10) as if it where some kind of ACK.
After this msg, while moving, the msg repeats with the previous "position" value. (84)
Only some seconds after the end of the course the field "position" has a new value. (10)

Can this reports be configurded in Z2M or depends on Vimar fw?
For example:
   Reporting every 15 sec while not moving seems unnecesary, maybe 30sec, 1min or even more would be enough.
   Reporting whidel moving frequently, maybe every second, or every 10% of ther course would be useful.
   Reporting actual state: "Opening, Closing, Stopped"
   Differentiate that "ACK" from actual "position"

PS:
I've seen something strange since paired this device, in the networkmap.
There are many lines between other routers and this one with arrow pointing it, but most with signal 0.
[image]

Probably not related with this issue, maybe for a future one.

Thanks !

closed 2020-10-18 by github-actions[bot]

Enne2, 2021-01-18:
> Can you try with:
>
> ```js
> {
>     zigbeeModel: ['Window_Covering_v1.0'],
>     model: '14594.0',
>     vendor: 'Vimar',
>     description: 'Roller shutter with slat orientation and change-over relay', [...]

Hi Koenkk!
I'm trying to configure this device with my Z2M + node-red setup. The Zigbee model name is slighty different (Window_Cov_v1.0) and I used your code editing the new zigbee id. However I'm encountering this error on start:

Jan 18 01:36:35 rpi-master npm[1657]: Zigbee2MQTT:info  2021-01-18 01:36:35: 0xf4ce3627ec26346a (0xf4ce3627ec26346a): 14594.0 - Vimar Roller shutter with slat orientation and change-over relay (Router)
Jan 18 01:36:35 rpi-master npm[1657]: Zigbee2MQTT:info  2021-01-18 01:36:35: Configuring '0xf4ce3627ec26346a'
Jan 18 01:36:35 rpi-master npm[1657]: Zigbee2MQTT:error 2021-01-18 01:36:35: Failed to configure '0xf4ce3627ec26346a', attempt 1 (ReferenceError: bind is not defined
Jan 18 01:36:35 rpi-master npm[1657]:     at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:10358:9)
Jan 18 01:36:35 rpi-master npm[1657]:     at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:131:45)
Jan 18 01:36:35 rpi-master npm[1657]:     at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:99:28)
Jan 18 01:36:35 rpi-master npm[1657]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:381:44))

sending manually mqtt state payload works as intended
can you help me? I don't know where to start.
Thank in advance

luci-11, 2021-02-24:
Hi @Koenkk ,
I have this item too to test :)
I'm using hassio and I successfully managed at least to change the device name and description with custom js.
I just have a problem in filling "exposes" field - In the doc there is written that for cover we have state, position and tilt.
I tried with e.state() but doesn't work.
Here is my vimar.js in external converters:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const device = {
    zigbeeModel: ['Window_Cov_v1.0'],
    model: '14594',
    vendor: 'Vimar',
    description: 'Roller shutter with slat orientation and change-over relay',
    supports: 'open, close, stop, position',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    meta: {configureKey: 1},
    exposes: []
};

module.exports = device;
```
 
Could you please help?
Thanks in advance!

Koenkk, 2021-02-24:
Should work with: `exposes: [e.cover_position()],`

luci-11, 2021-02-24:
> Should work with: `exposes: [e.cover_position()],`

It works!
Thanks you!

Koenkk, 2021-02-24:
Added the device, will be supported out of the box with zigbee2mqtt 1.18.0

luci-11, 2021-02-26:
> Added the device, will be supported out of the box with zigbee2mqtt 1.18.0

Awesome.
After a couple of days I still don't see any devices connected to the router.
So I was trying to add the configure part That I initially left out but i get errors:

This is what I'm trying to add:
`	configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(10);
        await bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await configureReporting.currentPositionLiftPercentage(endpoint);
    },`

This is what I get:
`evalmachine.<anonymous>:16
	configure: async (device, coordinatorEndpoint) => {
	^^^^^^^^^
SyntaxError: Unexpected identifier
    at new Script (vm.js:101:7)
    at createScript (vm.js:262:10)
    at Object.runInNewContext (vm.js:303:10)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.js:159:16)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.js:10:20)
    at new Controller (/app/lib/controller.js:77:34)
    at Object.<anonymous> (/app/index.js:25:20)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)` 

Looking at this [https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/externalConvertersExample/freepad_ext.js](https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/externalConvertersExample/freepad_ext.js) I don't see differences instructure..
Could this missing part be an issue?
Thanks a lot.

Lucia

luci-11, 2021-03-01:
> Added the device, will be supported out of the box with zigbee2mqtt 1.18.0

Hi, 
I just saw your commit,that's great.

I think we can also add 

```
	configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(10);
		await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
		await reporting.currentPositionLiftPercentage(endpoint);
    },
```
These lines don't give errors and I suppose are needed. 
Also @Joseraj is using them but for example in my case the bind.xx is giving errors. 

Up and down timings with cover arrows are ok, there is something strange in the percentage.
i.e. if I open it partially at 14%, then 100% then 0% it returns to 14% .   It's like it's not calculating remaining percentage from current position but just summing up.
If you have any ideas I can test

About `coverInverted` i'm unsure if we need it or not, because position = 100 is open so I think we should put it to true, but HA is recognizing directions and icons correctly.

Rendeik, 2021-07-15:
I tried to pair this device. The first attempts were unsuccessful, while now I can see and command it, but it does not update the position and a red triangle remains at the top with this value at the bottom: Interview completed -- False.

sginestrini, 2023-11-30:
Hi, I'm looking to get a few for controlling roller shutter with slat orientation. The things I noticed is that cover (tilt) is not exposed, am I right? From the manual https://faidate.vimar.com/irj/go/km/docs/z_catalogo/DOCUMENT/ZIS_30804-20594EN_FI.106648.pdf I can see that it can be set and calibrated, but can I do it from zigbee? Thank you

kartone, 2024-10-02:
I'm trying to integrate these controls within my house and I use zigbee2mqtt. I'm can open and close the curtains from HA but the position is wrong, and I don't know how to tell the device that the actual state is, for example, 100% OPEN or 0% CLOSED. Can you please help? Many thanks.