Koenkk/zigbee2mqtt #3856: Osram Plug no state update on the mqtt interface if the plug switch manual with the button on the plug

Maddin2020, 2020-07-06:
# Bug Report
## What happened
Osram Plug no state update on the mqtt interface if the plug switch manual with the button on the plug

## What did you expect to happen
That the state on the mqtt interface change too, if i manual switch the plug

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:1.14.0
Adapter hardware: CC2538
Adapter firmware version: MODKAMRU_V3_USB

lolorc, 2020-07-06:
If I recall correctly, that happened to me one time, and I re-paired the plug to make it work again.

IVI053, 2020-07-16:
Could also be a connection issue. These devices have very bad antenna design.

angryfish404, 2020-07-31:
I have the same issue, i guess since the last 2 versions. I can turn the osram plug on and off via mqtt, but the status is not updated, when I use the button on the plug

Koenkk, 2020-08-01:
Try to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

angryfish404, 2020-08-01:
wow, that worked like a charm. Thanks a lot!

Koenkk, 2020-08-01:
Good to hear, assuming this fixes it for OP, so can be closed.

closed 2020-08-01 by Koenkk