Koenkk/zigbee2mqtt #3858: EchoStar SAGE Doorbell Sensor (SAGE206612) implementation loses connection after awhile

unfurled, 2020-07-06:
# Bug Report
After a few hours from pairing, no events is received by Zigbee2MQTT
## What happened
This seems to be the case : [https://community.hubitat.com/t/porting-sage-doorbell-sensor-driver-help-needed/20653/14](url) where a different driver had to be written to get reliable events.
## What did you expect to happen
Doorbell press events should still be sent.
## How to reproduce it (minimal and precise)
1. Pair the Sage Doorbell Sensor
2. Wait overnight.
3. Sage Doorbell Sensor press events not longer sent.
## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531
Adapter firmware version: 20190608

unfurled, 2020-07-06:
Working code for Hubitat found here : [https://github.com/trunzoc/Hubitat/blob/master/Drivers/Sage_Doorbell/Sage_Doorbell.groovy](url)

Koenkk, 2020-07-06:
@mattia480 does it work for you? I've looked at the hubitat code but that looks similar to what zigbee2mqtt does.

mattia480, 2020-07-09:
In my enviroment it works perfectly also after few days.
[image]
My implementation was inspired by the work done by https://darwinsden.com/sage-doorbell-sensor-review-smartthings-device-handler/

closed 2020-08-16 by github-actions[bot]