Koenkk/zigbee2mqtt #3862: problem with deconz over TCP Port

salopette, 2020-07-06:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

sjorge, 2020-07-06:
Are you trying to connect to the device when zigbee2mqtt is running? I don’t think that will work.

salopette, 2020-07-06:
That doesn't work either, it works with a CC2531, but not with the conbeeII-

Koenkk, 2020-07-06:
Conbee over TCP port is not supported yet, you can request this feature here: https://github.com/Koenkk/zigbee-herdsman/issues/72 (Deconz team)

closed 2020-07-06 by Koenkk

zombielinux, 2020-10-26:
Crossposting this here. 

https://github.com/cminyard/ser2net/issues/35

ConbeeII cannot be piped over tcp at this moment. Vote for the feature. 

Also, ser2net's syntax has changed.

cminyard, 2020-10-26:
I debugged this a bit with zombielinux, and I'm a bit confused by the issue.  Is this something that ser2net is doing wrong?  It should be able to open and read/write the device, I would think.

ChrisHae, 2020-10-27:
Hi, I'm starting to look into this.

ChrisHae, 2020-10-28:
TCP support for ConBeeII is now ready. Use it like descriped here: https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

(Wait until it is merged in zigbee2mqtt-dev or hassio addon)

@Koenkk Could you please add ConBeeII support to the readme(s)