Koenkk/zigbee2mqtt #3865: Disable processing for TRADFRI wireless dimmer (ICTC-G-1)

vdkeybus, 2020-07-07:
# Help for debugging
## What happened

The Ikea ICTC-G dimmers seem to have processing built in to Z2M. There must be some bug that causes a command to be repeated to the backend (e.g. HA) indefinitely and bringing the whole system down.

At least the Zigbee sniffer indicates that the dimmers aren't sending any commands at the time Z2M is.

## Question

I would like to debug this, but I'm unfamiliar with the Z2M code. Can someone point me to the right directory / source file ? I also see that there's some settings possible, but I'd like to use the auto-detect if possible.

Thanks

Koenkk, 2020-07-07:
Please try disabling the legacy integration, the new integration only provides the actual commands being send (https://www.zigbee2mqtt.io/devices/ICTC-G-1.html#legacy-integration)

vdkeybus, 2020-07-07:
Thanks for the quick response.

I saw the option, but I wanted to see if I could fix the bizarre error; it
has brought my system down countless times, and even for 'legacy' users
it's a bug. Apart from that, I want to avoid relying on config files with
exceptions as much as possible.

I'm wondering if this option shouldn't be turned off by default. Or even
removed.

It looks like the integration tries to complete missing messages from the
remote, to either smoothen the transition or terminate every sequence with
'rotate_stop' when that last command isn't received. When the bug happens,
'rotate_stop' is never issued and 'rotate_left' / 'rotate_right' is
repeated ad infinitum. (For some reason, that completely bogs down the HA
server, which is a bug in itself too.)

For a z2m client, there is no difference between passing the original
message sequence, which could possibly not contain the 'stop_' message, and
the infinite sequence that doesn't contain the 'stop_' either. So it
wouldn't matter to them if legacy would be turned off by default ?

Koenkk, 2020-07-07:
> I'm wondering if this option shouldn't be turned off by default. Or even
removed.

It will be removed in zigbee2mqtt 2.0.0, but as this is a breaking change it is kept in and enabled by default.

vdkeybus, 2020-07-07:
Ok, in that case I will work with the exceptions in configuration.yaml.

This can be closed.

Thanks.

closed 2020-07-07 by vdkeybus