Koenkk/zigbee2mqtt #3866: No interaction with the devices possible

dschanoeh, 2020-07-07:
# Bug Report
## What happened
No interaction with the devices possible anymore without config change or update.

Issues started while using 1.12.0 (Initially, the version worked fine but issues started without an obvious change a few days ago). An update to 1.14.1 didn't help.
It seems like the CC2531 firmware crashes since zigbee2mqtt cannot even connect to the adapter when it is restarted without also restarting the adapter.

I'm running zigbee2mqtt in an LXC container on Ubuntu 20.04 with the serial device connected to the container.
```test -w /dev/ttyACM0 && echo success || echo failure```
returns success.

## What did you expect to happen
Normal operation as before.

## How to reproduce it (minimal and precise)
1. Unplug and plug the adapter
2. Start zigbee2mqtt
3. Trigger any command to modify device state
4. Observe error messages in the log:
```error 2020-07-07 15:11:21: Publish 'set' 'state' to '0x000b57fffebe26fd' failed: 'Error: Command 0x000b57fffebe26fd/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":50}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: SRSP - AF - dataRequest after 6000ms)'
debug 2020-07-07 15:11:21: Error: Command 0x000b57fffebe26fd/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":50}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: SRSP - AF - dataRequest after 6000ms)
    at Endpoint.<anonymous> (/pool/homeassistant/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:365:23)
    at Generator.throw (<anonymous>)
    at rejected (/pool/homeassistant/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
```
[full log](https://github.com/Koenkk/zigbee2mqtt/files/4885742/log.txt)

## Debug Info
Zigbee2mqtt version: 1.14.1 (commit #7e46435)
Adapter hardware: CC2531
Adapter firmware version: {"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}

Koenkk, 2020-07-07:
Probably your adapter crashed, I would recommend switching to the source routing firmware which gives more stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

dschanoeh, 2020-07-11:
Closing this. I cannot even re-flash the adapter. Replaced it with a ConBee II and it's working fine again.

closed 2020-07-11 by dschanoeh