Koenkk/zigbee2mqtt #3867: Support for channel masks

alekslyse, 2020-07-07:
I see in Home Assistant and Deconz they have support for multiple channel masks for end devices,

example in HA: channels: [15, 20, 25]  # Channel mask to distribute to end devices.

Does z2m support channel masks too?

Koenkk, 2020-07-07:
Channel mask are not supported, I've read about it but cannot understand what practical advantage it brings.

alekslyse, 2020-07-07:
From what I understand it makes it easier for endpoints to expect a channel change if the current channel is too noisy. I never noticed it before, but the other implementations implemented it so it seems like it has some benefits?

alekslyse, 2020-08-08:
Unstale. Should be implemented to follow functions of the other importations that all support channel masks

closed 2020-09-17 by github-actions[bot]