Koenkk/zigbee2mqtt #3868: IKEA LED1624G9 missing color attribute

mvtslot, 2020-07-07:
#Bug Report
My IKEA bulb LED1624G9 is paired with the zigbee2mqtt stick successfully. It can be switched on / off and brightness can be changed. But the color attribute is missing when I look at the topic via mosquitto_sub -d -t zigbee2mqtt/#. The only attributes are: 
- update_available
- state
- brightness
- linkquality

Any suggestions what could be the cause of this? Thanks!

 
## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: 
Adapter firmware version:

Koenkk, 2020-07-07:
In order to have the color value you first have to [set](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset) or [get](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameget) the color value.

Try e.g. sending to `zigbee2mqtt/DEVICE_FRIENDLY_NAME/get` with payload `{"color": ""}`

mvtslot, 2020-07-11:
Great suggestion. The color value is presented now. I will continue searching for the right configuration to control x and y from Openhab. Thanks.

Koenkk, 2020-07-11:
Good, I assume this can be closed now.

closed 2020-07-11 by Koenkk