Koenkk/zigbee2mqtt #3871: Device not finding closer path (router). Weak link to coordinator.

blarrkan, 2020-07-08:
Hi,
I have a batterypowered temp+hum sensor Xiaomi (the round). It have very weak link quality to the coordinator 1/255. But in the same room I have IKEA bulb connected to mains that have link quality 159/255. And just outside there is another IKEA bulb with link quality 159/255.

Why is my Xiaomi sensor not finding a better path via the two IKEA bulbs instead of trying to reach the coordinator? The two IKEA bulbs are different models.

The sensor and bulbs has been online for several days.

Versions
Version of Zigbee2Mqtt: 1.14.1
Coordinator version: 20190425

Koenkk, 2020-07-08:
Xiaomi battery powered devices are known to stick to their parent and therefore don't switch to the most optimal one. Something that can help is repair them close to the desired parent. Note that Zigbee2mqtt has no control over the meshing as the devices itself are responsible for this.

blarrkan, 2020-07-13:
Thanks for the quick answer @Koenkk 
I will try this and hope it will get better reception.

closed 2020-07-13 by blarrkan