Koenkk/zigbee2mqtt #3872: Setting the color on IKEA LED1624G9

filiptypjeu, 2020-07-08:
# Bug Report
I'm having some trouble setting the color on my Ikea Trådfri RGB light bulb. Sending `x` and `y` values work fine, and sending `rgb` values also work. However, when I try sending colors in any other format, for example `hsv`, the bulb does not respond accordingly.

When I'm setting the bulb with `{ "color": { "hsv": "20,100,100" } }`, I get the response `{"color":{"h":20,"s":100,"v":100},"brightness":254}`, just like I would expect. Or maybe I would expect to receive the `x` and `y` values instead, like I do when sending `rgb` colors? Anyway, the bulb does not change to the corresponding color, but instead turns to this bleak blue/white-ish color, as if I would have sent `"x": 0` and `"y": 0`.

So, am I just doing something wrong, or is setting `hsv` colors somehow not supported?

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531

Koenkk, 2020-07-09:
The Tradfri RGB bulb doesn't support setting hue/saturation (only x/y).

filiptypjeu, 2020-07-09:
Okay, then everything works like expected! I already solved my problems by setting the `xy` values directly, but an automatic `hsv` to `xy` conversion would be much appreciated anyway (just like the `rgb` to `xy` conversion)!

Koenkk, 2020-07-09:
The goal of hsv is to set hue saturation values (because it works for some bulbs where x/y cant be set), so converting this to xy makes no sense in this case

filiptypjeu, 2020-07-09:
Okay, the other possibility that would have helped me in my quest is if the bulb would respond with a `"rgb": "R,G,B"` property if it was set with an `rgb` color, instead of only responding with `x` and `y`. I've seen this feature have been requested here before.

Koenkk, 2020-07-10:
@filiptypjeu I also remember this being requested indeed, can you refer to the original issue and add a thumbs up to the first OP? (so I can see how much time this feature has been requested)

closed 2020-08-17 by github-actions[bot]