Koenkk/zigbee2mqtt #3874: Connection to Aqara temperatur sensor breaks after a couple of days.

anonimno1, 2020-07-08:
# Bug Report
## What happened
I have 2 Aqara temperature sensors which are connected to Domoticz using the Zigbee2MQTT Plugin.
The sensors are placed in 2 bathrooms. One of them works great. The other one works for a couple of days and then the connection breaks. If I take the sensor close to the coordinator (Zigbee2MQTT stick) and wake it up, then it starts sending again. After a couple of days the connection brakes again.
I tried cross-changing with the sensor which works good. When this one is moved to the other bathroom, the connection also breaks after a couple of days. So it is not an issue of a specific sensor.
Is the distance between the sensor and coordinator a problem? If so, why does it work correctly a couple of days?

Any suggestions how I can debug this?

## What did you expect to happen
Stabile connection to both sensors.

## How to reproduce it (minimal and precise)
Keep the system running a couple of days. I didn't find a way to provoke it.

## Debug Info
Zigbee2mqtt version: v.1.13.1 (zStack12 20190608)
Adapter hardware: CC2531
Adapter firmware version: 08.06.2019

Koenkk, 2020-07-09:
I've seen some Xiaomi sensors drop from the network when the network connection is to weak, try putting a router close to it.

mr2c12, 2020-07-11:
I used to suffer from the same problem when my coordinator was a Cc2530+cc2591.
That problem went away when I switched to a better coordinator (Texas instrument one).

closed 2020-08-19 by github-actions[bot]