Koenkk/zigbee2mqtt #3877: PTVO CC2530 does not work after update

PetrP2B, 2020-07-08:
After the upgrade z2m from 1.10 to 1.14, the cc2530 modules with the PTVO firmware stopped working. When reverting a version of z2m to an older one, everything works fine. I also tried the latest version of PTVO, but without success.

Log: zigbee2mqtt: error 2020-07-08 22:48:18: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'bottom_right' is given but device has no such endpoint
    at Zigbee.resolveEntity (/app/lib/zigbee.js:202:25)
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:59:44)
    at Controller.callExtensionMethod (/app/lib/controller.js:360:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:254:14)
    at MQTT.emit (events.js: 315: 20)
    at MQTT.onMessage (/app/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js: 315: 20)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1269:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:409:12)
    at work (/app/node_modules/mqtt/lib/client.js:320:12))

Koenkk, 2020-07-17:
@ptvoinfo can you check this?

ptvoinfo, 2020-07-17:
@Koenkk @PetrP2B I've renamed 'old-style' topic names like bottom_right to l1..l8. Also, I've updated HA integration. The end-user should change the old configuration manually.

jvrrr, 2020-07-18:
I have te same Issue (I use Domoticz) .
How can I fix this problem?

jvrrr, 2020-07-21:
Thanks to 9zigen I fix it....

The problem is in the zigbee2mqtt plugin:
Edit this file: /domoticz/plugins/zigbee2mqtt/adapters/diy/ptvo_switch.py
Then change PTVOID part by this:

PTVOID = {'input1' : 'l1',
		  'input2' : 'l2',
		  'input3' : 'l3',
		  'input4' : 'l4',
          'input5' : 'l5'}

Then restart Domoticz.

Now I'm able to open and close my garage door ;-)

InvertedMinor, 2020-07-21:
> @Koenkk @PetrP2B I've renamed 'old-style' topic names like bottom_right to l1..l8. Also, I've updated HA integration. The end-user should change the old configuration manually. But the documentation (https://www.zigbee2mqtt.io/devices/ptvo.switch.html) is not updated yet.

How can I fix this in Home Assistant?

jvrrr, 2020-07-21:
At te bottom of this page, you will find the config: https://www.zigbee2mqtt.io/devices/ptvo.switch.html

I think you need to replace bottom-right etc by l1, l2, etc.
I never run HA, sk I don't know where you can find this config

InvertedMinor, 2020-07-26:
> How can I fix this in Home Assistant?

There has been a breaking change in the PTVO firmware and zigbee2mqtt.

So after your zigbee2mqtt has been upgraded to > 1.14.1  the switch.0x00000_switch_bottom_left no longer functions, just replace this with switch.0x00000_switch_l1.

switch.0x00000_switch_bottom_left => switch.0x00000_switch_l1
switch.0x00000_switch_bottom_right=>switch.0x00000_switch_l2
switch.0x00000_switch_top_left=>switch.0x00000_switch_l3
switch.0x00000_switch_top_right=>switch.0x00000_switch_l4

PetrP2B, 2020-08-04:
After the update, everything works. Thank you

closed 2020-08-04 by PetrP2B

zamenator, 2020-09-01:
> the switch.0x00000_switch_bottom_left no longer functions, just replace this with switch.0x00000_switch_l1.

where it should be replaced ?

InvertedMinor, 2020-09-01:
In all the  files where it has been used like : Automation, scripts and lovelace

zamenator, 2020-09-01:
In HA  switch panel, pvto switch entities are not functional, when I try to switch them on/off it causes an exception in zigbee2mqtt :
```
error 2020-09-01 13:50:35: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'bottom_left' is given but device has no such endpoint

```
I can change entity ID from, say, "switch.relay_one_switch_bottom_left" to "switch.relay_one_switch_l1", but, mqtt event is still "zigbee2mqtt/relay_one/bottom_left/set ON" .

>  Automation, scripts and lovelace

no automation / scripts are used at this moment, an attempt to create one leads to the same problem -- I am adding "Action" with Device, relay_one, toggle the switch entity, and get the same zigbee2mqtt error.

HA and Z2M are running in docker containers, images were refreshed yesterday,  MQTT persistent messages are cleared. What else I can do?

zamenator, 2020-09-01:
ok, i see what happened, lib/extension/homeassistant.js survived docker image update and it still contained old definition of ptvo.switch .