Koenkk/zigbee2mqtt #3878: HS2WD-E - no message ?

eRudy, 2020-07-08:
# Bug Report
## What happened

I added a HEIMAN Smart siren in my network, it paired successfully. I can trigger the alarm/light, but I receive no message from the device.

## What did you expect to happen
I expected regular messages with battery status, and link quality.
Nothing in the log, even in debug...

## How to reproduce it (minimal and precise)
Waiting for hours for a message with : mosquitto_sub -v -h server.lan -t /zigbee2mqtt/device_alias

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531
Adapter firmware version: zstack 1.2 (20190608)

eRudy, 2020-07-08:
Well... hmmm it did finally arrive:

{"battery":85,"linkquality":39}

I expected a message when unpluging the device from the mains as well, but maybe there is not ?

Koenkk, 2020-07-09:
> I expected a message when unpluging the device from the mains as well, but maybe there is not ?
 
I'm not aware of such, it is know that battery take some time to report.

Are you able to trigger the siren?

closed 2020-08-16 by github-actions[bot]