Koenkk/zigbee2mqtt #3879: Bug Report - Stop reading from serial

mailart, 2020-07-09:
[image]

I have my pi based zigbee2mqtt hangs all the time. It runs for mins then stop reading from serial, I need to kill the process, reconnect the usb dongle to get readings for next mins.

I noticed when it stops reading after Ctrl + C I can always see this line 

umpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,2,0,30,6,0,25,3,2,11,0,0,20]

Koenkk, 2020-07-09:
What firmware are you running?

mailart, 2020-07-09:
> What firmware are you running?

Thanks Koenkk, I am using 20190608 with latest zigbee2mqtt. I have more than 40 sensors around, it seems the USB dongle does not response every 20,30 mins for me. and get retry. reload the cdc_acm USB driver does not help only reconnect and dongle can make it back.

Do you think it is a firmware issue? if the latest 3.0 one might fix that?

[image]

mailart, 2020-07-10:
[image]
 
another hang shown SYS - osalTimerExpired

Koenkk, 2020-07-10:
It seems that your adapter crashes, with such a large network I would recommend switching to the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

closed 2020-08-17 by github-actions[bot]