Koenkk/zigbee2mqtt #3883: Zigbee2mqtt slow and stops working after a few days

Martinnnl, 2020-07-09:
Hello,

My zigbee2mqtt is up-to-date but after a few days the stick stops working. It sends no commands from domoticz and the green light is also off.

And even after few days the delay between commands get more and more. 

Running on pi with the stick in a USB hub.

I have 15 devices (lamps and sensors).

Many thanks

Koenkk, 2020-07-10:
Probably your adapter crashed, in case you are not yet on the source routing firmware please try with that: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Martinnnl, 2020-07-10:
Thank you Koen. I tried but I can figure out how I can install this on pi by Putty.

Koenkk, 2020-07-10:
You have to follow https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html

Martinnnl, 2020-07-10:
I will buy the debugger. But after resetting and unplugging the sticks works again. But after a few days it again will be very slow and finally stops working.

Is this also because the firmware?

Koenkk, 2020-07-11:
Yes, the stick is probably running into memory issues, the source routing has more memory available so this should improve it.

Martinnnl, 2020-07-12:
Thank you. I now have to wait on the debugger.

closed 2020-08-19 by github-actions[bot]