Koenkk/zigbee2mqtt #3884: Ikea 1527 Floalt LED Panel reports false off state (still on, even if "off")

monotonus, 2020-07-10:
# Bug Report
## What happened

I have an Ikea 1527 Floalt LED Panel in our kitchen. It's usually triggered by a Xiaomi RTCGQ11LM motion sensor if it's dark enough. After occupany_timeout of 60 seconds the light will be turned off.  This worked really good for a long time. But now I'm facing the issue from time to time, that the Floalt Panel is still on after turning off. Home Assistant also shows the light as off and so it could be that the light shines unnecessarily for a longer time period. Manually toggling the light after that will result that it goes off.

The day before yesterday, I deleted the float from z2m and joined it again. Yesterday I also installed an OTA update to the Floalt LED panel, hoping this would change the behaviour, but this morning it happened again.

I now activated the debug log level for z2m. Maybe I can see something additional info there.

## What did you expect to happen

The light goes off as it's intended, reporting it's correct status to home assistant via MQTT.

## How to reproduce it (minimal and precise)

Maybe somebody has a similar scenario at home?

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-07-17:
Can you provide debug log of the moment you expect it to turn off and it doesn't?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

monotonus, 2020-07-17:
Yes as I said, I already have enabled debug logging. "Unfortunaltely" the error didn't show up in the last days, so I can't provide any insights yet.

sjorge, 2020-07-17:
Are they sometimes powered of completely? I’ve seen setup of reporting fail on Mine once or twice

monotonus, 2020-07-22:
No it's usually not powered of completely. Can't remember the last time I used to switches :-D

Right at this moment it was happening again. Home Assistant shows state off since 21:21:35, but the light is still on in "real life". The z2m logs are looking good so far:

```
zigbee2mqtt:debug 2020-07-22 21:19:32: Received MQTT message on 'zigbee2mqtt/kuechenlampe/set' with data '{"state": "ON"}',
zigbee2mqtt:debug 2020-07-22 21:19:32: Publishing 'set' 'state' to 'kuechenlampe',
zigbee2mqtt:info  2020-07-22 21:19:33: MQTT publish: topic 'zigbee2mqtt/kuechenlampe', payload '{"update_available":false,"state":"ON","brightness":0,"color_temp":316,"linkquality":68}',
zigbee2mqtt:debug 2020-07-22 21:19:33: Received MQTT message on 'zigbee2mqtt/kuechenlampe/set' with data '{"state": "ON", "transition": 1.0, "brightness": 253, "color_temp": 387}',
zigbee2mqtt:debug 2020-07-22 21:19:33: Publishing 'set' 'brightness' to 'kuechenlampe',
zigbee2mqtt:info  2020-07-22 21:19:33: MQTT publish: topic 'zigbee2mqtt/kuechenlampe', payload '{"update_available":false,"state":"ON","brightness":253,"color_temp":316,"linkquality":68}',
zigbee2mqtt:debug 2020-07-22 21:19:33: Publishing 'set' 'transition' to 'kuechenlampe',
zigbee2mqtt:debug 2020-07-22 21:19:33: Publishing 'set' 'color_temp' to 'kuechenlampe',
zigbee2mqtt:info  2020-07-22 21:19:33: MQTT publish: topic 'zigbee2mqtt/kuechenlampe', payload '{"update_available":false,"state":"ON","brightness":253,"color_temp":387,"linkquality":68}',
zigbee2mqtt:debug 2020-07-22 21:20:37: Received MQTT message on 'zigbee2mqtt/kuechenlampe/set' with data '{"state": "ON", "transition": 60.0, "brightness": 253, "color_temp": 398}',
zigbee2mqtt:debug 2020-07-22 21:20:37: Publishing 'set' 'brightness' to 'kuechenlampe',
zigbee2mqtt:info  2020-07-22 21:20:37: MQTT publish: topic 'zigbee2mqtt/kuechenlampe', payload '{"update_available":false,"state":"ON","brightness":253,"color_temp":387,"linkquality":68}',
zigbee2mqtt:debug 2020-07-22 21:20:37: Publishing 'set' 'transition' to 'kuechenlampe',
zigbee2mqtt:debug 2020-07-22 21:20:37: Publishing 'set' 'color_temp' to 'kuechenlampe',
zigbee2mqtt:info  2020-07-22 21:20:37: MQTT publish: topic 'zigbee2mqtt/kuechenlampe', payload '{"update_available":false,"state":"ON","brightness":253,"color_temp":398,"linkquality":68}',
zigbee2mqtt:debug 2020-07-22 21:21:35: Received MQTT message on 'zigbee2mqtt/kuechenlampe/set' with data '{"state": "OFF"}',
zigbee2mqtt:debug 2020-07-22 21:21:35: Publishing 'set' 'state' to 'kuechenlampe',
zigbee2mqtt:info  2020-07-22 21:21:35: MQTT publish: topic 'zigbee2mqtt/kuechenlampe', payload '{"update_available":false,"state":"OFF","brightness":0,"color_temp":398,"linkquality":68}'
```

Next step is to disable the circadian lighting switch which is automatically dimming the light according to daytime. Maybe this is causing this behaviour in some way.

closed 2020-08-30 by github-actions[bot]