Koenkk/zigbee2mqtt #3885: modelid is undefined for CC1352p-2 examples

manju-rn, 2020-07-10:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)
I am trying to connect the TI CC1352p-2 launchpad board to zigbee2mqtt (as a add-on in HA).

    I am running code composer studio from TI and loaded up the sample examples zed_light and zed_sw (Switch)
    When I load one of the program to the board and select commission (through the usb). It connects fine to Z2M (I had ensured the correct PanID and the network key.
    However, upon successfully announcing an joining the network, I still see zigbee2mqtt:warn 2020-07-10 13:52:27: Device '0xxxx' with Zigbee model 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
    I know I need to edit the devices.js, but even after editing it the above error persists, where do I provide teh modelid 2020-07-10 13:52:27: Skipping message, modelID is undefined and still interviewing ?

My objective is to control the launchpad led light through HA by using the example program zed_light

## Debug Info
Zigbee2mqtt version:
Adapter hardware:  CC1352P-2
Adapter firmware version:  NA. Loading from the examples from the SDK from Code Composer Studio of TI

Koenkk, 2020-07-10:
I don't have any experience with the zed_light example, I guess the modelID is not defined by default (as it's optional in the ZCL). Anyway I think it would be better to ask your question on the TI forum (https://e2e.ti.com/) as there are likely more people which can help.

manju-rn, 2020-07-10:
Thanks. So is my understanding correct that without the modelid i will not be able to go forward? Is there a way to still recognize the using the actual device 0xxxx?

Koenkk, 2020-07-11:
It may be possible to identify it through a fingerprint (e.g. digi router also doesn't have a modelID: https://github.com/Koenkk/zigbee-herdsman-converters/blob/121a7eaaa93ffdf1f2dea72eebdf751bcc99800a/devices.js#L3813). You can retrieve the properties from the `data/database.db` entry of this device.

sdhuaren, 2020-08-01:
I am having the same problem, while identifying model ID of my Samsung DP728 lock.
After pairing for many times, Z2M still can't get its model ID

closed 2020-09-09 by github-actions[bot]

manju-rn, 2020-09-13:
Solution:
1. In zcl_samplelight_data.c,  define the model ID array
const uint8_t modelId[] = { 5, 'M','A','N','J','U'};
2. In the same file, within the attributes definitions, create the model id cluster attribute
CONST zclAttrRec_t zclSampleLight_Attrs[] = 
{
    // Other existing attributes
    {
      ZCL_CLUSTER_ID_GEN_BASIC,
      { // Attribute record
        ATTRID_BASIC_MODEL_ID,
        ZCL_DATATYPE_CHAR_STR,
        ACCESS_CONTROL_READ,
        (void *)modelId
      }
    },
};

So this will add the modelid in ZCL and should be reflected in the Z2M.   You now need to follow the sequence of activities for adding the unsupported devices

hosentraeger, 2021-11-13:
Z-Stack 3.6.0+ to Z-Stack 4.3.0:
"Several ZCL files inside C:\ti\simplelink_cc13x2_26x2_sdk_x_xx_xx_xx\source\ti\zstack\stack\zcl have renamed definitions for cluster IDs, attributes, and commands."

ZCL_CLUSTER_ID_GEN_BASIC ->    ZCL_CLUSTER_ID_GENERAL_BASIC
ATTRID_BASIC_MODEL_ID -> ATTRID_BASIC_MODEL_IDENTIFIER