Koenkk/zigbee2mqtt #3887: Osram Flex RGBW stops working after pairing

networkingcat, 2020-07-10:
# Bug Report
## What happened

Osram Flex RGBW Lightstrip is not working correctly.
Pairing works but Light is not controllable (like turning off/on, adjusting stuff)

zigbee2mqtt debug log: https://pastebin.com/3x5p44YC
It is this light: https://www.zigbee2mqtt.io/devices/4052899926110.html

I am running zigbee2mqtt on a docker container on a qnap and all other devices work just fine.
The Osram Flex is not the most distant device so i suspect not a distance problem.

## What did you expect to happen
Controling of light

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: zigbee2mqtt@1.14.1
Adapter hardware: , CC26X2R1
Adapter firmware version: Newest Z-Stack_3.x.0

P.S. Thanks and Keep up the impressive work with zigbee2mqtt

Koenkk, 2020-07-10:
Can you provide the herdsman debug log when controlling fails?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

networkingcat, 2020-07-11:
Hi,

Many thanks.

Here is the output of the debug log:
https://pastebin.com/mUgbBQDR

The Device is: zigbee2mqtt/Flex Kitchen

Also i am wondering about the db entry for this light:

{"id":35,"type":"Router","ieeeAddr":"0x8418260000ca976d","nwkAddr":38832,"manufId":48042,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1594482197383}

does not seems correct for me (because of interviewCompleted":false and other missing stuff)

I also tried deleting this device from db and repairing, but that does no help.

Regards

networkingcat, 2020-07-11:
Update:

Just tried another repairing, but not with the Power on/power off multiple times method but with a hue remote. Now it looks like this in the database:

{"id":35,"type":"Router","ieeeAddr":"0x8418260000ca976d","nwkAddr":38832,"manufId":48042,"manufName":"OSRAM","modelId":"Flex RGBW","epList":[3],"endpoints":{"3":{"profId":49246,"epId":3,"devId":528,"inClusterList":[4096,0,3,4,5,6,8,768,64516,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Flex RGBW","manufacturerName":"OSRAM","powerSource":42,"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20140331CNWT****","swBuildId":"V1.03.07"}}},"binds":[]}},"appVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20140331CNWT****","swBuildId":"V1.03.07","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1594484471743}

Controlling worked for about 10 seconds, then it is not controllable again.

Here is a new debug log:https://pastebin.com/Rjucss4W

It says "No network route' (205)" but pairing at the same location works :/

Thanks and regards

Koenkk, 2020-07-12:
I've checked the log but cannot find the problem yet, could you sniff the traffic of the pairing + controlling for 10 seconds? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

networkingcat, 2020-07-12:
The No network route stuff triggered me to repair again multiple times.

Now the error changed and sometimes (but rarely) i can even control the light.

debug log with when working: https://pastebin.com/Rjucss4W

debu log when failing: https://pastebin.com/5w6tAgri
It is MAC no ack

I am thinking it is some weird stuff from the light not using a router but trying anything to talking directly to the coordinator, which can be problematic because of poor link quality. There are a lot of routers in between the coordinator and this light but it only shows a linkquality of 3.

I can not sniff the traffic because i dont have a CC2531. I got rid of all that stuff when i got my CC26X2R1 (too big network and and not so good range with CC2531)

Koenkk, 2020-07-14:
> I am thinking it is some weird stuff from the light not using a router but trying anything to talking directly to the coordinator, which can be problematic because of poor link quality.

When having the osram close to the coordinator, does it work reliable?

networkingcat, 2020-07-29:
I cant get it very close without a lot of work so at first i tried by getting them much closer together but not very close.

If i try this it does indeed works very good, even for a day.
After putting everything back in its place it does stop working.

I think it tries then to use a router in the middle but that fails somehow. (But again pairing at this place works without problem)

Koenkk, 2020-07-29:
In that case it's the same problem as #3910 , let's continue there as we got some sniffs

closed 2020-07-29 by Koenkk