Koenkk/zigbee2mqtt #3888: router

cichy63, 2020-07-10:
# Bug Report
## What happened
My network is made out of two routers (CC2530 + CC2591). I have them on two floors. One day I've moved the upstair downstairs and the other way around. I needed to go around the house and repair devices (mainly xiaomi door sensors). Since that time i get constantly some messages from sensors that are not in range. Example below:

zigbee2mqtt:debug 2020-07-10 19:57:52: Received Zigbee message from '0x00124b00199dd3d5', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"29402/0x00158D0003F88085","inactiveText":"CHILD_RFD","presentValue":170,"relinquishDefault":2,"minimumOffTime":0}' from endpoint 8 with groupID 0
zigbee2mqtt:info  2020-07-10 19:57:52: MQTT publish: topic 'zigbee2mqtt/0x00124b00199dd3d5', payload '{"state":true,"led_state":true,"linkquality":84,"description":"29402/0x00158D0003F88085","type":"CHILD_RFD","rssi":170,"led":true}'
0x00124b00199dd3d5 - upstairs router
0x00158D0003F88085 - downstairs sensor (definately not in range)

"presentValue":170 is always the same. From that i know that even when i have sensor on the zigbee map showed with linkquality of 170 it's not active and the trafic is routed by other router where the value is different. Those messages are constantly spamming my log. Is there a way to 'clear it' so i got only the messages that occur in reality without those 'ghosts'?

## What did you expect to happen

Reduction of false log messages. Understand why for all those situations value of 170 is presented. 

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531, 
Adapter firmware version: latest, CC2531_DEFAULT_20190608.zip

Koenkk, 2020-07-10:
@ptvoinfo do you know what these messages mean?

ptvoinfo, 2020-07-11:
@cichy63 @Koenkk It is a message from a diagnostics version of the router firmware. The router reports about child or neighborhood devices.

cichy63, 2020-07-11:
@ptvoinfo that's true this is diag version of router installed. But why only some sensors are reported. I see a lot of messages from this one. And as i said this router is not in direct range with a sensor. On top value is 170 always and it is not matching linquality of this sensor at all.

ptvoinfo, 2020-07-13:
The routers reports about sensors in the field of view. I cannot explain that data. The firmware gets it from SDK and wraps it to a report.

closed 2020-07-13 by cichy63