Koenkk/zigbee2mqtt #3890: Zibgbee2mqtt stopped working with no apparent reason

pavel-vrba1, 2020-07-11:
Hi,

I have been running Home assistant + Zigbee2mqtt + Mosquitto in docker on Odroid xu4 (Ubuntu Mate OS) for several months without an issues. Few days ago it stopped working without any reason - I neither performed any updates on the machine nor did any change to docker or Home assistant configuration. 

I have a TI CC2531 stick and its LED used to be always GREEN. But now when I unplug it from Odroid and plug again the LED goes green but after a while it turns off (the same behavior is on different USB ports and even when plugged to desktop PC with windows). I also see several "disconnect" messages when running `dmesg | grep -i usb`:

``[    5.691079] usb 3-1.2: Product: TI CC2531 USB CDC
[    5.695803] usb 3-1.2: Manufacturer: Texas Instruments
[    5.700869] usb 3-1.2: SerialNumber: __0X00124B0018E28037
[    5.700882] usb 6-1: New USB device found, idVendor=0bda, idProduct=8153
[    5.712931] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[    5.720029] usb 6-1: Product: USB 10/100/1000 LAN
[    5.724701] usb 6-1: Manufacturer: Realtek
[    5.728778] usb 6-1: SerialNumber: 000001000000
[    5.901025] usb 6-1: reset SuperSpeed USB device number 2 using xhci-hcd
[   29.445160] cdc_acm 3-1.2:1.0: ttyACM0: USB ACM device
[   29.452030] usbcore: registered new interface driver cdc_acm
[   29.452039] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[ 3557.099506] usb 3-1.2: USB disconnect, device number 4
``

Finally I am getting errors in the log when starting Zigbee2mqtt that might mean the stick is not available over USB:
`zigbee2mqtt:info  2020-07-11 15:54:40: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-07-11 15:54:53: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-07-11 15:54:53: Failed to start zigbee
zigbee2mqtt:error 2020-07-11 15:54:53: Exiting...
zigbee2mqtt:error 2020-07-11 15:54:53: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)`

Is it possible that the CC2531 stick is broken. Is there a way to check it is working (something like ping command). Might there be an issue with some power saving that powers off the USB? I do not have the CC debugger so I cannot test flashing latest firmware.

Thanks.

Koenkk, 2020-07-12:
It is likely that the stick crashed (which is unfortunately common for the CC2531 as it's not very powerful). You have to reflash your stick to get it working again, I would recommend the [source routing](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing) firmware as it provides increased stability.

pavel-vrba1, 2020-07-14:
Hi Koen, thanks for info, I will try to reflash the stick, hopefully I will be able to do so without soldering :)

pavel-vrba1, 2020-08-17:
Hi Koenkk, sorry for not replying earlier, I was waiting for the CC debugger delivery from Ebay. I flashed the source routing firmware and the CC2531 stick returned to normal working state - I even did not need to repair the devices nor doing an other reconfiguration. So everthing is working again. Thanks, Pavel

Koenkk, 2020-08-17:
Good to hear! Assuming this can be closed now.

closed 2020-08-17 by Koenkk