Koenkk/zigbee2mqtt #3891: CC2530 + CC2591 needs hardware cycle for every pairing

SaturnusDJ, 2020-07-11:
Using this as coordinator with Z-Stack_Home_1.2 (source_routing).

First off, https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator here it says it can handle 5 direct children. Why do I see it having 6 direct children?

But the real issue: pairing is a mess. It seems for every pairing the USB connections needs to be remade and Zigbee2MQTT restarted. Anyone has a clue why?

Koenkk, 2020-07-12:
Can you share some more details about your network (how many routers/end devices)?

I think this issue occurs because the child device table is full, after power cycling, this table is cleared and populated again as devices reconnect, so until those devices reconnected some can still directly join the coordinator.

SaturnusDJ, 2020-07-12:
Thanks.

At that moment (after starting from scratch with the new adapter) I was trying to connect the first end devices via routers. At first from meters distance, then moving closer and closer until direct contact. At one point I even removed the hardware antenna from the coordinator to prevent end devices and new routers from directly connecting with it, if that is how it works? It did not help. In the end it somehow came together, I wish I knew exactly how but it was with lots of re-pairing. What was especially weird was how the end devices (all HS3SA as I started with those) did pair, but their smoke reporting feature did not come through even though the devices were reported as successfully configured sending the configure command also did not work. It took those re-pairs to finally get them to report if smoke was detected (test button to trigger). After that, I joined more and more routers which still connected with the coordinator (I thought 5 was max...) and also a bit more end devices which connected via routers.

This is the network now:
[image]

Considering 16 more light bulbs and 20 Xiaomi sensors end-devices.

Koenkk, 2020-07-14:
To get further with this we need to have a sniff of the problem, would you be able to provide such? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

SaturnusDJ, 2020-07-15:
I would do it, however the problem is gone. After those hours of re-paring in the end it worked. No hints to what the solution was. I kept removing those HS3SA devices until they configured and worked correctly. Maybe they are picky and/or what you said about the coordinator table.

But even now, when adding new routers they always also show to have a link with the coordinator. Is that really the case? As the max. should be 5 according to the firmware page.

The network seems good, so much better than CC2531.

SaturnusDJ, 2020-07-23:
@Koenkk 
Apparently the problem is not gone.
While MCCGQ11LM, WXKG11LM, WXKG12LM, I think joined without problems, HS3SA (earlier), RTCGQ11LM, WSDCGQ11LM, MFKZQ01LM have a lot of trouble. Maybe the ones willing to join was luck. The MFKZQ01LM all of a sudden did join... I can't find logic behind it.

Flashed the old CC2531 for sniffing. **What specifically should I look for now?**

A lot of data is coming in. For example `Many-to-One Discovery: With Source Routing`
Or:
[image]

This is the network size now:

[image]

SaturnusDJ, 2020-07-23:
From trial and error by re-pairing the MFKZQ01LM it seems:
`Beacon` kind of messages mean a new device is near.
`Device Announcement` means a new device is going to join.

After finding this out I saw the WSDCGQ11LM device not wanting to join also not even triggering that `beacon` messages. Replaced battery and it paired immediately. 🤦‍♂️ Seems like pairing requires quite a bit more of energy compared to just being in a network and sending out updates once in a while.

Now this makes sense for some other devices which are running on 1-1,5 year old batteries. But...I think this is not *the fix* for this problem *if* it is still present because the HS3SA devices had a near full battery.

Will report again if new trouble occurs.

SaturnusDJ, 2020-07-23:
Trying more sensors of type WSDCGQ11LM.
3 fails, 1 join.

Let me know how to proceed with sniffing.

Koenkk, 2020-07-23:
A typical joining procedure looks the following:

[image]

- Device does a Beacon request
- Router or coordinator responds with Beacon
- Device does an association request (request to join the network)
- Router or coordinator responds with association response
- Router or coordinator sends network key via a Transport key command
- At this point the device joined the network, Zigbee2mqtt will now start interviewing (gathering information) of the device and will e.g. do node descriptor requests.

SaturnusDJ, 2020-07-23:
Thanks, I will report back with logs once new Zigbee devices have trouble joining. 👌

SaturnusDJ, 2020-07-23:
Well, a WSDCGQ11LM became half alive, back from dead, so results are in already:
* Beacon request
* Beacon, src... (4x)
* Association request
* Association response
.....end?

[image]

Koenkk, 2020-07-24:
> .....end?

Now the transport key should be send by the coordinator or router (I forgot who is responsible for sending this when the device joins via a router). Two options here:
- Hue Lightstrip Plus is responsible: not sure why it doesn't send it, maybe it doesn't like Xiaomi devices, probably not fixable from the Zigbee2MQTT/firmware side
- Coordinator is responsible: can probably be fixed in the coordinator firmware (similar to https://github.com/Koenkk/zigbee2mqtt/issues/3177). I've compiled a CC2530_CC2591_source_routing firmware with extra debug information to see where it gets stuck. Run zigbee2mqtt with `DEBUG=zigbee-herdsman:adapter:zStack:znp:AREQ npm start` and you should see the following in your log when a device joins:

[image]

Firmware:  [CC2530_CC2591_source_routing_join_debug.zip](https://github.com/Koenkk/zigbee2mqtt/files/4974297/CC2530_CC2591_source_routing_join_debug.zip)

> Am I correct it tries to join via device ending with af:52?

yes

SaturnusDJ, 2020-07-28:
More proof of the problem being the Philips Hue devices.

There was trouble joining IKEA LED lights. After some tries, 4 of the 6 joined. The other 2 stayed problematic until the Hue devices were unplugged from power. Also I saw at least one failure when the light tried to join via the Hue light, same like before.

Can that debug-firmware be flashed without having to re-pair/re-setup the devices/network?

For the next paring, if not that firmware, I will pair next to those Philips Hue devices, them being on at first, off afterwards.

Koenkk, 2020-07-28:
> Can that debug-firmware be flashed without having to re-pair/re-setup the devices/network?

Yes, no repairing is required

SaturnusDJ, 2020-07-29:
Would it be possible the Philips Hue devices are not integrated fully because I did not reset them (https://www.zigbee2mqtt.io/information/touchlink.html)? Instead I removed them from within the Hue app and they automatically joined the Z2M network.

Koenkk, 2020-07-29:
Don't expect to, both should give the same result.

closed 2020-09-05 by github-actions[bot]

Morphy99, 2021-01-04:
I have this issue with pairing not working until I physically power off the CC2530. Restarting or rebooting my RPi didn't resolve it. I had it while running the default 20190608 firmware and now with source routing 20190619 fw on my new instance. Both were running connected to a RPi4 GPIO.
When this happens I don't see any other issues with the network. I've had it when I've been joining multiple devices and it fails after 1 or 2 or when I enable joining and try to add 1 it can also do it.

SaturnusDJ, 2021-01-05:
Today I switched to Zzh. Pairing routers went fine, pairing end-devices after a bunch of ten turned out a hell again. I did not check what the problem was.
Nevertheless I can report that I see three end-devices connected to a Philips Hue light. I am not 100% sure if this happened before but this was extremely rare here. Maybe Zigbee 3.0 is the reason for it happening.